libretro/Lakka-LibreELEC #20: nestopia does not compile for Generic/x86_64

noromamai, 2017-01-16:
checked out git a few days ago (0170116001250-r25105-g461b7fb) and compiled Generic/x86_64. borked while compiling nestopia with message (last few lines):

`../source/core/NstStream.hpp: In member function 'void Nes::Core::Stream::Out::Write(const byte (&)[N])':
../source/core/NstStream.hpp:123:19: warning: typedef 'Nestopia_assertion_at_line_123' locally defined but not used [-Wunused-local-typedefs]
      NST_COMPILE_ASSERT( N > 0 );
                   ^~~~~~~~~~~~~~~~              
make[1]: Leaving directory '/home/kdun/Projects/Lakka-LibreELEC/build.Lakka-Generic.x86_64-8.0-devel/nestopia-eeda3a8/libretro'
Makefile:12: recipe for target 'image' failed`

removing nestopia from compile would let compile continue and complete.

Kivutar, 2017-01-16:
What is your host OS? Because nestopia for x86_64 compiles well on 2 of our ubuntu build servers.
I'm checking on debian now.

noromamai, 2017-01-18:
i was compiling on Korora 25 (upgraded from 24), 64-bit. Same machine (C64x, Asrock J4205 Mini-ITX) Lakka is supposed to run on. I just downloaded nestopia sources from https://github.com/rdanbrook/nestopia and put that in the build directory of Lakka. Using those sources it did compile.

Kivutar, 2017-01-18:
Yesterday @Ntemis pushed another fix for nestopia. Let try it.

Kivutar, 2017-01-21:
We now have binary images there: http://le.builds.lakka.tv/

nestopia compiles well for us.

noromamai, 2017-01-22:
checked out latest sources and compiles now! so everything is ok now.

is there a way to force first run status? so it will create all the filestructure on the storage file? i deleted some hoping to reset configuration, now i can't enable ssh anymore.

closed 2017-01-22 by noromamai