libretro/Lakka-LibreELEC #31: Dualshock 3 auto pairing is broken

Kivutar, 2017-01-21:
Certainly by the bluez update.
All images are affected.
LibreELEC does have our bluez patches for auto pairing.

Gerporgl, 2017-01-22:
I am also having issues pairing my Dualshock 3, tried the command line through ssh but bluetooth does not appear to work. Please note this is the first time I try Lakka and I'm running on an ODROID C2. I was also not sure if my bluetooth dongle would work (it is an Asus something...) but it was working on my other LibreElec Kodi image build for Odroid C2

Gerporgl, 2017-01-29:
Bluetooth pairing seems to work fine on my side now using the latest build on my ODROID C2 Lakka-Odroid_C2.arm-8.0-devel-20170125104540-r25568-gc7d8e37.img.gz
I haven't tried with a Dualshock however this time (I gave up on that one), I'm now using a 8bitdo controller.
Not sure if this was actually fixed or if this was just me not doing it right the first time.

Kivutar, 2017-02-06:
Pairing works, but what about auto pairing? I mean you plug the DS3 as USB, and it automatically gets paired.

Gerporgl, 2017-02-06:
Good point. Originally (when I tested the first build), it wasn't working I had to pair manually, since then I switched to 8bitdo controllers, so I didn't try again the auto pairing with dual shock. I'll give it another try with the latest build I have and let you know.
I think overall it would be nice if auto pairing was also working with other controllers (like 8bitdo), or if at least there was a graphical bluetooth menu, so that we don't have to use the console.

Gerporgl, 2017-02-06:
Just tried, and auto pairing does not work. I think it could be related to USB mode not working as well.

Here is what I got in dmesg:
[  150.687683@0] usb 1-1.2: new full-speed USB device number 5 using dwc_otg
[  150.691496@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  150.691496@0] 
[  150.694838@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  150.694838@0] 
[  150.698170@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  150.698170@0] 
[  150.777668@0] usb 1-1.2: device descriptor read/64, error -71
[  150.997515@0] usb 1-1.2: New USB device found, idVendor=054c, idProduct=0268
[  150.997530@0] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  150.997535@0] usb 1-1.2: Product: PLAYSTATION(R)3 Controller
[  150.997540@0] usb 1-1.2: Manufacturer: Sony
[  151.002004@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  151.002004@0] 
[  151.005666@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  151.005666@0] 
[  151.008560@0] sony 0003:054C:0268.0001: Fixing up Sony Sixaxis report descriptor
[  151.012929@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  151.012929@0] 
[  151.016234@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  151.016234@0] 
[  151.019564@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  151.019564@0] 
[  151.023396@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  151.023396@0] 
[  151.023713@0] input: Sony PLAYSTATION(R)3 Controller as /devices/dwc2_b/usb1/1-1/1-1.2/1-1.2:1.0/0003:054C:0268.0001/input/input5
[  151.024120@0] sony 0003:054C:0268.0001: input,hiddev0,hidraw0: USB HID v1.11 Joystick [Sony PLAYSTATION(R)3 Controller] on usb-dwc2_b-1.2/input0
[  151.027538@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  151.027538@0] 
[  151.027578@0] sony 0003:054C:0268.0001: can't set operational mode
[  151.127779@0] sony: probe of 0003:054C:0268.0001 failed with error -71

I got the same messages on all 3 different builds I tried: the web site img for Odroid C2 #1962d23, as well as my own images build from git sources using commit 7357ecf for both arm and aarch64).

On my Linux Mint Laptop (Core i7 4700HQ x86_64) I get different message and the DS3 controller led #1 turns (and stay) on normally (with Lakka all LEDs flash red rapidly):
[31209.150116] usb 3-2: new full-speed USB device number 127 using xhci_hcd
[31209.347566] usb 3-2: New USB device found, idVendor=054c, idProduct=0268
[31209.347570] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[31209.347572] usb 3-2: Product: PLAYSTATION(R)3 Controller
[31209.347573] usb 3-2: Manufacturer: Sony
[31209.390202] input: Sony PLAYSTATION(R)3 Controller as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/0003:054C:0268.000B/input/input100
[31209.390565] sony 0003:054C:0268.000B: input,hiddev0,hidraw0: USB HID v1.11 Joystick [Sony PLAYSTATION(R)3 Controller] on usb-0000:00:14.0-2/input0

Kivutar, 2017-02-09:
OK this is fixed for me, with the commit of @Ntemis 
@Gerporgl can you confirm that auto pairing also works for you? (don't forget to press the reset button of the DS3 with a needle or toothpick).

Kivutar, 2017-02-09:
Fixed on Generic too.

closed 2017-02-09 by Kivutar

Gerporgl, 2017-02-10:
I just tried on Odroid C2 and in my case it still does not work. As I reported above, I am getting some USB error in dmesg and I'm not even able to use the DS3 in USB mode (as I understand it should work at least as a connected joystick). I got this DS3 controller when I bought my Playstation 3 from BestBuy about 3 years ago (so I assume it is a genuine one) but it is possible that it is defective somehow.
Anyway, that is fine if you are closing this issue, maybe my USB issue is something else, and I'm not going to use the DS3 much on my side in the future (just for personal reference)