libretro/Lakka-LibreELEC #35: Takes long time (2 minutes) to start emulator on Asrock J4205 Motherboard

noromamai, 2017-01-22:
Asrock J4205+ Mini-ITX board (Apollo Lake based), Generic, x86_64

Tested this on own compiles and the new precompiled images. It takes around 1.5-2 minutes after choosing to run a game for the emulator to start and show something on screen. Once it runs everything seems normal. Exiting from emulator back to retroarch also takes a long time.
if ssh'ing into lakka and stopping retroarch service an restarting from command line (just retroarch or retroarch -v) this problem does not exist. perhaps some startup options are being used causing this.
I can not confirm this on the precompiled images because i can no longer ssh into the machine (i stupidly deleted some directories on the storage partition hoping that would trigger lakka to recreate them and give me a fresh install. i did this because i thought the keyboard config was borked, but it turned out that when using the dualshock3 (via usb) it would break the keyboard and would no longer let the keyboard go left or up). any way to trigger lakka to recreate the defaut dirs?

Kivutar, 2017-01-22:
Ok this is certainly due to a texture format. When we boot with the menu, RetroArch chooses a texture format, then has to switch to another one to launch the game.

While when you launch the game from command line, it chooses the right texture format directly.

What is you exact graphic card model?

noromamai, 2017-01-22:
i am not launching from command line. what i did was ssh in, stop retroarch service, then restart the retroarch service (without any command line options). then when i start a game from the menu, it is instant. the processor on the mobo has an integrated graphics: Integrated Intel® HD Graphics 505: 18 EUs inside (Up to 800MHz)

Brunnis, 2017-01-25:
I can confirm this issue on my J4205-ITX with Intel HD Graphics 505 as well. Took more than 2 minutes, though. Maybe 5-10 minutes.

Brunnis, 2017-01-25:
I have identified what triggers this issue (and a work-around):

If the Ethernet cable is plugged in when you fire up the game, it takes a really long time to start. If the Ethernet cable is unplugged, starting a game takes only a few seconds.

I currently have no internet access on my network (due to the fiber line having been severed nearby). Don't know if internet access or no internet access affects the issue.

You don't need to restart Lakka/RetroArch between plugging/unplugging the network cable. Just unplug it right before launching a game and start-up time will be normal.

Kivutar, 2017-01-25:
It may be linked to the achievements or netplay code then. @Twinaphex, any idea?

hackbird001, 2017-01-25:
Hello,
after a long time and failure from my site lakka works on my asrock j3455-itx...
When i play now PS1 or SNES Games, EMUS start after i click on the game (1-2 seconds)....
LAN InterneCable is even in my HTPC....
Only thing with keyboard, when i go back with ESC lakka makes a new boot. On my Xbox360 wirelles controller nothing. But thats no problem. Just want write that the emus will load after i click the game (1-2 seconds)

noromamai, 2017-01-25:
i just tested this and can confirm. that if there is no network cable connected and wifi has been disabled (i did this in the bios, because i could not find a setting in retroarch), starting up games is instant.

gouchi, 2017-02-10:
Maybe related to [this issue](https://github.com/libretro/RetroArch/issues/4578) but you have no crash just delay.

fr500, 2017-03-11:
it might be achievements... although 9 minutes seems too long...
can you try disabling achievements?

Kivutar, 2017-03-11:
I got the same issue here https://github.com/lakkatv/Lakka-LibreELEC/issues/77 with disabled achievements. I was connected to a wifi network where dhcp gave ma a wrong IP. Disabling that connection fixed it.

I wonder what part of RA is requiring network on game launch.

Kivutar, 2017-04-02:
I'm not able to reproduce this bug anymore. Even with the same version of RA.

Kivutar, 2017-05-22:
No replies, I close this too.

closed 2017-05-22 by Kivutar