libretro/Lakka-LibreELEC #36: Problems with asrock j3455-ITX

hackbird001, 2017-01-22:
Hey, 
i installed on USB the latest lakka libreelec version.
My graphics Card is: Integrated Intel® HD Graphics 500: 12 EUs inside (Up to 750MHz)

Boot was ok and lakka opens.
Problems:
1. Cores wont be show, button with "update cores" missing
2. Sound wont go  (Realtek ALC892)

noromamai, 2017-01-22:
i am running on Asrock J4205+ which is basically the same board with a different cpu. I can confirm the sound is not working, when looking up "audio device" in retroarch interface it shows nothing.

have not used retroarch before, so can't check about "cores not shown" because i am not sure where it normally is supposed to be.

Kivutar, 2017-01-22:
Hello, when you go in Load Core, you see nothing?

hackbird001, 2017-01-22:
Nope, no core will be shown. 
The update core button which normally shown isnt there too....

Kivutar, 2017-01-22:
"Update Cores" has been removed from Lakka for more than one year now.

What do you see in /tmp/cores?

hackbird001, 2017-01-23:
Hey Kivutar,

when i looged in over ssh (\\lakka) i see the folders from cheats till update. Only folder (Configfiles) git files the others all empty. I used (Live) because i intalled on my ssd libreelec beta.

/tmp/cores isnt there

Kivutar, 2017-01-23:
Ok, now you need to use "Run" which is the same as "Live" on the old images.

hackbird001, 2017-01-23:
Hey,

so i download again and install on USB.
When i now choose Run it will install and i am like before on the main screen of lakka.
But no button with "Load Core". Same on "Online Updater"... 
On "Online Updater" I got only  "Update Lakka", Thumbnails Updater, Content Downloader.....

When i lookin over ssh same like i write in the post before....

noromamai, 2017-01-23:
sound does not get detected correctly. but can be fixed. i can't change the audio device setting in retroarch (won't let me type). but ssh into the device, stop retroarch ("systemctl stop retroarch.service"). run "aplay -l" to list audio devices. check what is listed as HDMI 0 (in my case it was card 0, device 3. it should be the same for your asrock j3455). cd /storage/.config/retroarch/. then edit retroarch.cfg ("vi retroarch.cfg") and find (towards the end of the file) audio_device = "" and change that to 'audio_device = "hw:0,3"'. this should give you sound over your hdmi connection.

hackbird001, 2017-01-23:
My sound at my HTPC goes over SPDIF, because i use surround system. I dont use TV sound. 

Before sound work at mine, should work first cores and other things because my folders are all empty...

Did you have cores in your folders noromamai??

noromamai, 2017-01-23:
you can try hw:0,1
that's non-hdmi digital output. 
i am having problems ssh'ing into the box, so i can't check about the cores.

Kivutar, 2017-01-24:
> so i download again and install on USB.
> When i now choose Run it will install and i am like before on the main screen of lakka.
> But no button with "Load Core". Same on "Online Updater"...
> On "Online Updater" I got only "Update Lakka", Thumbnails Updater, Content Downloader.....

This is how lakka works, all the cores are contained in the firmware. You don't need to download cores, theyr are already there.

hackbird001, 2017-01-24:
Yes i know Kivutar,
but there is no Core when i Choose RUN...

For a time on my old zbox lakka run without any problems. Zbox was only to slow for gaming...

But now with my new board and install USB lakka/libreelec boot from RUN or LIVE no Core and so on in the folders.

When i use the lakka version from lakka.tv  lakka stops booting on the flower screen.
I did too an isue on  github lakka/openelec

Brunnis, 2017-01-25:
I believe this issue can be closed, since the issues described seem to be due to user error.

**1. Cores wont be show, button with "update cores" missing**

The cores are there. When I choose "Load Core", they're all listed. When I enable the advanced menu options and go into "Playlists", I can choose default core for each playlist.

2. Sound wont go (Realtek ALC892)

I have an Asrock J4205-ITX, which is identical to J3455-ITX, save from the slightly beefier CPU/GPU. Sound works fine, both using HDMI and analog. When you go to "Audio Device" in the menu, you shall press left/right on the controller to choose between the different output devices, i.e.:

For analog output: "surround51:CARD=PCH,DEV=0"
For HDMI audio (via the board's DVI connector): "hdmi:CARD=PCH,DEV=1"

If you're using the actual HDMI port on the board (which is 4k 60Hz capable), you need to choose one of the other options available. I believe it's "hdmi:CARD=PCH,DEV=0".

hackbird001, 2017-01-25:
Hey Brunnis,

which version of lakka do you use?
Do you use lakka directly from USB? 
I know that your board is identical with mine and its not the first time i use lakka, but its first time on this board. 
So if you can tell will be nice...

If its not working i make some pictures and will post

Brunnis, 2017-01-25:
I'm using the following image:

Lakka-Generic.x86_64-8.0-devel-20170121181654-r25552-g0c703fc.img.gz

Downloaded from http://le.builds.lakka.tv/Generic.x86_64/

I'm running Lakka directly from USB, i.e. just booted it up and started configuring and testing.

hackbird001, 2017-01-25:
Ok, i got exactly same.
Do you choose RUN or LIVE?

hackbird001, 2017-01-25:
OK sorry, i found my fault....

U write: When I enable the advanced menu options and go into "Playlists", I can choose default core for each playlist.

Didnt do this. Now i enable and i can see all.
Sorry for this....

hackbird001, 2017-01-25:
ISSUE Can be Closed

Brunnis, 2017-01-25:
That's good news! :-)

Did you get sound working as well then?

hackbird001, 2017-01-25:
Thx Brunnis for your help.
Sound will be testing now.

Brunnis, 2017-01-25:
Nice! Enjoy. :-)

closed 2017-01-25 by hackbird001