libretro/Lakka-LibreELEC #40: Option to control CPU scaling governor

Brunnis, 2017-01-25:
In testing with Lakka on my J4205-ITX, I noticed that CPU frequency scaling can cause performance issues in games. It's easy to test with Yoshi's Island in combination with using the Frame Delay setting. The symptom is that performance starts to deteriorate quickly when using frame delay, even though the system "should" be capable of handling it. It's caused by the CPU down clocking due to thinking it's idle.

Completely disabling C-states and SpeedStep in UEFI is not a good enough option, since this appears to disable CPU turbo as well. On some CPUs, such as the Pentium J4205, base clock and turbo clock are far apart, which means this would leave a lot of performance on the table (the J4205 actually stays close to max turbo even for continuous loads).

_[Side note: I believe it's specifically SpeedStep that affects turbo, so C-states can (and probably should) be disabled for maximum performance.]_

When I have setup RetroArch on Ubuntu previously, I've manually forced the operating system to apply the 'performance' CPU governor. This way, turbo works as it should and the CPU doesn't down clock unless due to thermal/power limits being reached.

So, is it possible to actually add this setting into Lakka and automatically make it change CPU governor to 'performance' when launching a game and restore it to normal/powersave when exiting the game? RetroPie has this feature, but maybe it's more complicated to implement in Lakka due to the variety of architectures supported?

Kivutar, 2017-05-12:
@gouchi @Ntemis do you know if we can do this?
I'd like to implement this.
And is it problematic to always enable the performance scheduler?

gouchi, 2017-05-13:
We need to check how to do it and provide testing image.

larskj, 2017-06-05:
Just want to add that I also see significantly worse performance after upgrading to Lakka 2.0 coming from an older build that had lots of performance related adjustments activated (CPU Governor=Performance, 1000hz tickrate, preemptible kernel etc.). Don't know if it is still relevant, but perhaps consider revisiting these changes if they have since been discarded again: https://github.com/libretro/Lakka/pull/401 

The main topics covered are:

- Setting cpu governor to performance if not already set. (Probably low risk)
- Setting tick rate to 1000Hz. (Probably low risk)
- Setting preemption model to 'Fully Preemptible Kernel (RT) (Probably high risk)

Kivutar, 2017-06-05:
You're right. I do this now.

Kivutar, 2017-06-05:
Thanks for pointing it @larskj 

What are the dangers of Fully Preemptible Kernel (RT) ?

Should I enable these options on ARM boards?

larskj, 2017-06-05:
If we didn't see any problems with it the last time we used the RT patch and enabled the Fully Preemtible Kernel then I think we should add the changes again.

There should be nothing wrong with enabling these options for ARM boards as well.

Kivutar, 2017-06-05:
I noticed no issues with the fully preemptible kernel last time. Except that we could not compile Xorg anymore. But I have no plans to support Xorg anyway.
A small downside was the maintainance of the patch on kernels like the sunxi one where the patch may not apply.

Note: RT patches are there https://www.kernel.org/pub/linux/kernel/projects/rt/

larskj, 2017-06-05:
Ok great, then I think we should just go ahead and apply it again.

I'm looking at the changes made in the current build I am running, which runs very well.

Besides the RT patch and preemtible kernel I also ended up changed the priority in retroarch to SCHED_FIFO with max priority, see https://github.com/larskj/RetroArch/commit/ce597fe8575a27e9a78143f2b92c410f3e0c0992

Kivutar, 2017-06-05:
Ok, I enabled the RT patch for the generic images. I will add it on RPi and Gamegirl tonight.

What about submitting a PR for this RetroArch patch? Is it safe?

larskj, 2017-06-05:
Ok great I will try the image for generic x64 when it is available. For Lakka who owns the entire system it makes sense to set max priority. I will test the new image first and then consider making a PR for this other patch.

Kivutar, 2017-06-05:
Here is an image http://static.kivutar.me/Lakka-Generic.x86_64-2.1-rc2.img.gz

It is named rc2, but it's a developer version.

larskj, 2017-06-09:
I tried the image in your link and it works much better than the Lakka 2.0 image. I now get a stable frame rate again similarly as the older builds.

Kivutar, 2017-06-09:
Thanks for your test. I will release soon then.

@Brunnis what do you think about this change, is it enough to close this issue?

Brunnis, 2017-06-09:
Yep, I believe so. As I understand it you're now forcing the performance governor at all times, even when in the menu right? Although the optimal solution would be to use the performance governor only when running a game, this is fine and should only have a pretty marginal negative effect on power consumption when in the menu.

closed 2017-06-09 by Kivutar