libretro/Lakka-LibreELEC #41: onscreen keyboard does not respond to keyboard

noromamai, 2017-01-25:
when i get the onscreen keyboard popup (when asked to enter wifi passphrase, username, etc.), i can't use the physical keyboard (arrow keys) to navigate and type. there is nothing i can do, i can't exit the onscreen keyboard with any keypress. the only thing that works is to plug in a joypad (in my case a dualshock3 connected via usb) and use that to type and/or exit the onscreen keyboard.
even without a joypad connected the onscreen keyboard pops up. onscreen keyboard should either let the user use the physical keyboard (arrow keys, enter) to navigate and type or not show when no joypad is connected. then again, no joypad connected does not mean a physical keyboard is present....

Kivutar, 2017-02-16:
Yes, an important bug to fix.
I'm not able to reproduce this bug on my dev env, but it affects every Lakka images. Even the OpenELEC based ones.

kryptalivian, 2017-03-04:
Ya, I'm having it happen on my Mac Book Pro 2015 :-(

WIll give a controller a try

closed 2017-03-13 by Kivutar

keilmillerjr, 2017-03-15:
@Kivutar unfortunately, the issue is not yet resolved.

I updated to rc3, which includes the fix for this issue. However, the issue is not resolved. After a fresh update to rc3 on my sd card, now get a virtual keyboard that pops up - unlike before with rc2. I can not navigate the virtual keyboard with my physical keyboard, and I can not type a passphrase with my physical keyboard. In fact, I can not even escape from this point.

keilmillerjr, 2017-03-16:
So I went out and bought a ps4 controller. Hooked it up with a usb cable and could now enter a wifi password with the virtual keyboard. I followed the directions in link below to have it connect wirelessly, and it worked well. The issue only stands with a physical keyboard. Seems to work well with a controller though.

https://github.com/libretro/Lakka/wiki/Accessing-Lakka-command-line-interface
http://www.lakka.tv/doc/Wireless-Dualshock/

Kivutar, 2017-03-16:
I don't understand. I tested it myself. And you should have had a virtual keyboard appearing too with RC2, it was already there. Also the keyboard should be unable to navigate the virtual keyboard.
Check that you don't have checked "integrated inputs" in "input settings".

ryanfb, 2017-10-30:
@keilmillerjr @Kivutar I'm also seeing this behavior when testing 2.1-rc5 on a RasPi 3. The WiFi passphrase screen seems completely unresponsive to my physical keyboard (in fact, once I go into the WiFi passphrase screen it seems the only way I can get out is by rebooting). With 2.0 it works fine.

I don't see an "integrated inputs" option, but I'm guessing this is "unified menu controls"? If so, it's not checked when I see this behavior.

ryanfb, 2017-11-07:
See now: #238

discobanco, 2018-07-01:
still happening on Lakka-RPi2.arm-2.1.img.gz

ToKe79, 2018-07-01:
This is fixed in the [devel builds](http://nightly.builds.lakka.tv) and also in the soon to be released 2.1.1

discobanco, 2018-07-01:
Thanks for the quick response!

commit 30cb8c56 (2017-03-13): Update RetroArch and add libxkbcommon. Fixes #41