libretro/Lakka-LibreELEC #42: XU4 does not boot

externo6, 2017-01-26:
Hey,

As the title says, XU4 does not boot with LE builds. Tried 2 SD cards and still no anvil.

Thanks

Kivutar, 2017-02-06:
My XU3 boots well with the LE builds, and they are supposed to be fully software compatible.

externo6, 2017-02-07:
Still having the issue, flashed the SD card the same as with the the main Lakka project.

cudencuden, 2017-02-08:
seems like libreelec xu3 do not boot on xu4.  https://forum.libreelec.tv/thread-1627.html

Kivutar, 2017-02-08:
There are no official LibreELEC XU3 iirc. It's just the Lakka port I adapted to this Lakka-LibreELEC repo.

Do you see the boot logo @externo6 ?

Are the LEDs blinking or something?

Kivutar, 2017-02-08:
I will try to compile it with the old u-boot I was using in Lakka before.

externo6, 2017-02-08:
No LED's blinking, just solid RED LED as if no SD card is inserted.

Kivutar, 2017-02-09:
I was not able to compile the old u-boot, too much incompatibilities with gcc6 :(

Kivutar, 2017-02-09:
@zalaare when you use u-boot 2015.10 with the two patches, are you able to boot your LibreELEC images on XU4?

Kivutar, 2017-02-10:
Another user confirmed that it doesn"t boot on XU4

Kivutar, 2017-02-12:
Can you try a pure LibreELEC image from here please? http://freeweb.mine.nu/zalaare/LibreELEC/
A 8.0 for example.

Kivutar, 2017-02-12:
Looks like this guy can boot it https://forum.libreelec.tv/thread-1627.html

Kivutar, 2017-02-13:
OK I got something new this morning. It looks like the results are different if we're using eMMC or SD.

Kivutar, 2017-02-14:
I got it booting on SD, it should fix XU4 too http://le.builds.lakka.tv/OdroidXU3.arm/Lakka-OdroidXU3.arm-8.0-devel-20170214194014-r25646-g92bb6e7.img.gz

closed 2017-02-14 by Kivutar