libretro/Lakka-LibreELEC #45: Artifacts at startup of Lakka, with ATI X1650 Pro graphics card

iancilos, 2017-02-06:
With the ATI X1650 PRO video card I have video artifacts trouble starting LAKKA (as in the photo attached) stable version and also in the under development based on LibreELEC.

It works perfectly with 32bit & 64bit:

- Windows 7 + Retroarch v1.4.1
- Ubuntu 10.16 + Retroarch v1.4.1
- Android-x86 6.0-r2 + Retroarch v1.4.1

It does not work lakka 32bit & 64bit:

- Lakka-Generic.x86_64-devel-20161221183017-r21729-g274b40f
- Lakka-Generic.x86_64-8.0-devel-20170206002232-r25604-g32d2949

Tried LibreElec "LibreELEC-Generic.x86_64-7.95.2" and it works perfectly KODI interface

I include photos dell'artifatto interface just started and the Logs LAKKA
I hope for a resolution of your problem, thanks to the work you do!

[image]
[Lakka system logs.txt](https://github.com/lakkatv/Lakka-LibreELEC/files/754676/Lakka.system.logs.txt)
[Graphic card logs.txt](https://github.com/lakkatv/Lakka-LibreELEC/files/754678/Graphic.card.logs.txt)
[Audio device logs.txt](https://github.com/lakkatv/Lakka-LibreELEC/files/754680/Audio.device.logs.txt)
[Input device logs.txt](https://github.com/lakkatv/Lakka-LibreELEC/files/754681/Input.device.logs.txt)

gouchi, 2017-02-06:
Thank you for your report.

Is it possible to get also dmesg output ?

Can you try also to lower your resolution if you reproduce your issue ? First you need to get the [connectors](http://www.lakka.tv/doc/PC) than change [cmdline](http://www.lakka.tv/doc/The-bootloader/#on-generic-pc) with [video parameters](https://wiki.archlinux.org/index.php/ATI#Force_TV-out_in_KMS). Something like, if you use DVI output 
```
video=DVI-I-1:1280x720@60
video=DVI-I-1:1024x768@75
video=DVI-I-1:1024x768@72
video=DVI-I-1:1024x768@70
video=DVI-I-1:1024x768@60
```

Thank you.

iancilos, 2017-02-06:
Thank you for the reply!
I tried to insert each of these parameters to the startup line:

`video=DVI-I-1:1280x720@60
video=DVI-I-1:1024x768@75
video=DVI-I-1:1024x768@72
video=DVI-I-1:1024x768@70
video=DVI-I-1:1024x768@60`

but nothing changes.

there it is attached dmesg:
[dmesg.txt](https://github.com/lakkatv/Lakka-LibreELEC/files/756003/dmesg.txt)

gouchi, 2017-02-06:
We are using DRM/KMS not Xorg. 

You need to add one argurments at a time. First append video=DVI-I-1:1280x720@60 if it is not working try with video=DVI-I-1:1024x768@75 ...

iancilos, 2017-02-06:
yes, I tried them one at a time. if you want I can try again :)

gouchi, 2017-02-06:
And what is the output of ?
`for p in /sys/class/drm/*/status; do con=${p%/status}; echo -n "${con#*/card?-}: "; cat $p; done`

To get more log you can try append and get the output of dmesg.
`drm.debug=0x1e log_buf_len=1M`

iancilos, 2017-02-06:
> Lakka:~ # for p in /sys/class/drm/*/status; do con=${p%/status}; echo -n "${con#
> */card?-}: "; cat $p; done
> DVI-I-1: connected
> DVI-I-2: disconnected
> SVIDEO-1: disconnected

[dmesg2.txt](https://github.com/lakkatv/Lakka-LibreELEC/files/756135/dmesg2.txt)

gouchi, 2017-02-07:
The artifact happened directly when you boot or when the interface is started ?

iancilos, 2017-02-07:
Hello, when you start the interface, after bootscreen. With Windows, Ubuntu, LibreELEC, android-x86 everything works fine.
Now that I remember well a similar thing happened to me a long time ago with a version of ubuntu (?) Very old and with the next version everything was solved. Maybe they updated the drivers ...

The video card works perfectly and is not damaged.
I tried the parameter: video=DVI-I-1:1024x768@60
and it is ignored
I tried the parameter: video=DVI-I-1:1024x768@60e
And 'the correct form of the parameter. The screen remains black and does not go forward loading.

gouchi, 2017-02-07:
Can you try to append boot arguments `textmode` to the cmdline then start RA manually with 
LIBGL_DEBUG=verbose retroarch --menu --verbose >> log.txt 2>&1

It is not mentioned with lspci but you don't have integrated graphic card ? If you have one please try to disable it in the BIOS.

Thank you for your tests !

iancilos, 2017-02-09:
Hi! I run the command that you sent me and extrapolated the LOG file.
I am available for other tests, see you soon.

PS. The motherboard has an integrated video card but disabled by the BIOS to use the PCI Express 2.0, ATI X1650 PRO

[log.txt](https://github.com/lakkatv/Lakka-LibreELEC/files/762309/log.txt)

iancilos, 2017-02-12:
There are other tests that I can do?

gouchi, 2017-02-12:
Is it possible to get the output of `lsmod` ? I want to check if you nouveau module is loaded or not.

Thank you.

iancilos, 2017-02-13:
```
Lakka:~ # lsmod
Module                  Size  Used by
snd_hda_codec_realtek    56515  1
snd_hda_codec_generic    53514  1 snd_hda_codec_realtek
k8temp                  3514  0
8021q                  17727  0
snd_hda_intel          22061  3
snd_hda_codec          81447  3 snd_hda_intel,snd_hda_codec_generic,snd_hda_codec_realtek
snd_hda_core           42945  4 snd_hda_intel,snd_hda_codec,snd_hda_codec_generic,snd_hda_codec_realtek
snd_hwdep               5354  1 snd_hda_codec
overlay                32616  4
```

This both before and after the blacklist nouveau.
It should not appear something like radeon?

gouchi, 2017-02-14:
Yes you should have radeon module loaded. I don't understand why it is not loaded as we didn't disable it. When you disable it you still get the issue ? Did you try to modprobe radeon module then launch retroarch ?

iancilos, 2017-02-14:
tell me step by step what I have to do for you.
Do not load the radeon module either from live

gouchi, 2017-02-14:
```
systemctl stop retroarch
modprobe radeon
dmesg | tail
systemctl start retroarch
```

gouchi, 2017-02-14:
modprobe -v radeon ? It should display something if there is error.

iancilos, 2017-02-14:
no, no error shown in the terminal. no changes.

gouchi, 2017-02-14:
Maybe try 
`modprobe drm drm_kms_helper ttm i2c-algo-bit radeon`

As provided by 
`modinfo  radeon | grep depends`

iancilos, 2017-02-14:
```
Lakka:~ # systemctl stop retroarch
Lakka:~ # modprobe drm drm_kms_helper ttm i2c-algo-bit radeon
Lakka:~ # modinfo radeon | grep depends
modinfo: ERROR: Module radeon not found.
Lakka:~ # systemctl start retroarch
```

no changes. I do not understand what the problem is.
I think it's a driver issue. there is a way to try others here?

gouchi, 2017-02-15:
Is it possible to get the output of `dmesg` ?

Thank you.

gouchi, 2017-02-15:
We don't see it as it has not been enabled as a [module](https://github.com/lakkatv/Lakka-LibreELEC/blob/master/projects/Generic/linux/linux.x86_64.conf#L3232) ;-)

iancilos, 2017-02-15:
```
Lakka:~ # dmesg | grep drm
[    1.082306] [drm] Initialized
[    1.082326] [drm] radeon kernel modesetting enabled.
[    1.083025] [drm] initializing kernel modesetting (RV530 0x1002:0x71C1 0x174B:0x0840 0x9E).
[    1.083045] [drm] register mmio base: 0xDFFF0000
[    1.083046] [drm] register mmio size: 65536
[    1.083949] [drm] Generation 2 PCI interface, using max accessible memory
[    1.084012] [drm] Detected VRAM RAM=512M, BAR=256M
[    1.084014] [drm] RAM width 128bits DDR
[    1.084165] [drm] radeon: 512M of VRAM memory ready
[    1.084167] [drm] radeon: 512M of GTT memory ready.
[    1.084188] [drm] GART: num cpu pages 131072, num gpu pages 131072
[    1.086577] [drm] Possible lm63 thermal controller at 0x4c
[    1.090712] [drm] radeon: 1 quad pipes, 2 z pipes initialized.
[    1.091768] [drm] PCIE GART of 512M enabled (table at 0x0000000000040000).
[    1.091814] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[    1.091815] [drm] Driver supports precise vblank timestamp query.
[    1.091902] [drm] radeon: irq initialized.
[    1.091925] [drm] Loading R500 Microcode
[    1.092061] [drm] radeon: ring at 0x0000000020001000
[    1.092091] [drm] ring test succeeded in 7 usecs
[    1.092253] [drm] ib test succeeded in 0 usecs
[    1.092748] [drm] Radeon Display Connectors
[    1.092750] [drm] Connector 0:
[    1.092750] [drm]   DVI-I-1
[    1.092751] [drm]   HPD1
[    1.092753] [drm]   DDC: 0x7e40 0x7e40 0x7e44 0x7e44 0x7e48 0x7e48 0x7e4c 0x7e4c
[    1.092754] [drm]   Encoders:
[    1.092754] [drm]     CRT1: INTERNAL_KLDSCP_DAC1
[    1.092756] [drm]     DFP1: INTERNAL_KLDSCP_TMDS1
[    1.092756] [drm] Connector 1:
[    1.092757] [drm]   SVIDEO-1
[    1.092757] [drm]   Encoders:
[    1.092758] [drm]     TV1: INTERNAL_KLDSCP_DAC2
[    1.092759] [drm] Connector 2:
[    1.092759] [drm]   DVI-I-2
[    1.092760] [drm]   HPD2
[    1.092761] [drm]   DDC: 0x7e50 0x7e50 0x7e54 0x7e54 0x7e58 0x7e58 0x7e5c 0x7e5c
[    1.092762] [drm]   Encoders:
[    1.092762] [drm]     CRT2: INTERNAL_KLDSCP_DAC2
[    1.092763] [drm]     DFP3: INTERNAL_LVTM1
[    1.180088] [drm] fb mappable at 0xC00C0000
[    1.180091] [drm] vram apper at 0xC0000000
[    1.180092] [drm] size 4325376
[    1.180092] [drm] fb depth is 24
[    1.180093] [drm]    pitch is 5632
[    1.180202] fbcon: radeondrmfb (fb0) is primary device
[    1.233298] radeon 0000:02:00.0: fb0: radeondrmfb frame buffer device
[    1.233313] [drm] Initialized radeon 2.48.0 20080528 for 0000:02:00.0 on minor 0
[    1.236762] [drm] amdgpu kernel modesetting enabled.
```
```
Lakka:~ # ls  /sys/class/drm
card0           card0-DVI-I-2   controlD64      ttm
card0-DVI-I-1   card0-SVIDEO-1  renderD128      version
```

gouchi, 2017-02-15:
I am not sure it will help and if the option is still available in kernel 4.9 but can you try to boot with `radeon.new_pll=0`
You need to append this option to the [cmdline](http://www.lakka.tv/doc/The-bootloader/#on-generic-pc).

iancilos, 2017-02-15:
I modified cmdline boot to boot with `radeon.new_pll=0`
No changes.

What do you think it could be?

gouchi, 2017-02-16:
I tried to ask on [#radeon](http://webchat.freenode.net/?channels=%23radeon) but no answer.

iancilos, 2017-02-16:
I wish I could give more information about the problem. tell me what to do and do the testing for you.

I see if I can recreate the same problem with an old version of ubuntu. I seem to remember having a simi problem with a very old version of ubuntu. I will let you know

gouchi, 2017-02-18:
Can you check also in [retroarch.cfg](http://www.lakka.tv/doc/Configuring-Lakka) that 
`video_threaded = "false"`

iancilos, 2017-02-18:
no improvement. the menu movements are fast and the background animation smoothly moves, then it appears to have hardware acceleration. just that everything you see distorted as if not working properly in the frame buffer

P.S: I could not replicate this problem in older versions of Ubuntu. everything happens only in lakka

gouchi, 2017-02-18:
Remember that we are using DRM/KMS not Xorg. So if you want to test in older versions of Ubuntu, you need to access virtual console with (CTRL+ALT+F1 and launch RA). You will need to enable some [compilation options ](https://github.com/libretro/RetroArch/wiki/KMS-mode)and do some [configuration](https://github.com/libretro/RetroArch/wiki/Input-drivers-in-Linux-without-Xorg).

iancilos, 2017-02-18:
It's not something that's right for me :-)
At this point what is left to do?

gouchi, 2017-02-18:
You tried also with the second DVI output ?

iancilos, 2017-02-18:
yes, no change.
Tested well with dvi-vga adapter and no change
Then everything would be a driver regression DRM / KMS since being removed x11?

gouchi, 2017-02-19:
Do you reproduce the issue when you test in virtual console (CTRL+ALT+F1) in other distro ?

iancilos, 2017-02-19:
1. Started Lubuntu 10.16 64bit Live USB
2. Installed Retroarch 1.4.1 From PPA Official
3. Started Retroarch from the desktop (X11) and working properly.
4. CTRL + ALT + F1 and started Retroarch ... **distorted screen**! same problem as reflected in Lakka!

Thank you for guiding me, now I was able to replicate the problem in other distal.
So it should be an issue of open DRM/KMS driver in the Linux kernlel?

gouchi, 2017-02-19:
Yes should try to open an [issue](https://bugs.freedesktop.org) (Product : DRI, Component : DRM/Radeon) with cmdline options drm.debug=0x1e log_buf_len=1M and dmesg log, libdrm version, kernel version, lspci -nnk | grep -A4 VGA... and step to reproduce your issue. Also tell them it is not happening with Xorg.

But first, can you try to reproduce with this [distro](https://nouveau.pmoreau.org) because it is using latest libdrm, mesa...?

gouchi, 2017-08-06:
Did you have the opportunity to make some [test](https://nouveau.pmoreau.org) ?

Also you may try [Lakka 2.1 rc4](http://le.builds.lakka.tv/Generic.x86_64/Lakka-Generic.x86_64-2.1-rc4.img.gz).

Thank you.

ToKe79, 2018-01-31:
Issue closed as no response received. If the issue still exists with latest [release](http://www.lakka.tv/get) and/or [nightly](http://nightly.builds.lakka.tv), please feel free to re-open this issue. Thank you.

closed 2018-01-31 by ToKe79