libretro/Lakka-LibreELEC #46: No sound with Raspberry Pi 3

rejurime, 2017-02-07:
Hi! 
I just try "Lakka-RPi2.arm-8.0-devel-20170206005138-r25604-g32d2949-noobs" with my Raspberry Pi 3 and I have no sound, any ideas? works fine with "Lakka-RPi2.arm-devel-20161221184910-r21729-g274b40f-noobs".
Regards.

Kivutar, 2017-02-07:
Uncomment hdmi_drive=2 in config.txt?

rejurime, 2017-02-07:
Still the same 😞.

Kivutar, 2017-02-08:
Well I don't get it then, the same build runs fine on my RPi3.

rejurime, 2017-02-09:
It's rare because I have the problem with the noobs version only.

Kivutar, 2017-02-10:
How did you install it in noobs?

rejurime, 2017-02-10:
I follow the steps of http://www.lakka.tv/doc/Alternative-installation-methods/ as I do with the previous version (based on OpenELEC) and that works with sound.

Kivutar, 2017-02-16:
I could reproduce this bug. It looks like we're using a different config.txt for NOOBS. I'm going to fix it.

Firaenix, 2017-02-17:
I have also noticed this issue when placing the .tar file in the ~/.updates/ folder and letting Lakka install the update via a reboot, I've downgraded to latest stable OpenELEC in the mean time

Firaenix, 2017-02-17:
Just fixed it by editing the config.txt.

I added dtparam=audio=on to the bottom and now all is good.

Kivutar, 2017-02-22:
I fixed it in the last builds.

closed 2017-02-22 by Kivutar