libretro/Lakka-LibreELEC #54: Official Dualshock 3 not working on Odroid C2

Kivutar, 2017-02-16:
However, my Shanwan is working well.

dmesg:

```
[  244.427141@1] usb 1-1.1: new full-speed USB device number 17 using dwc_otg
[  244.430775@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  244.430775@0] 
[  244.434375@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  244.434375@0] 
[  244.437734@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  244.437734@0] 
[  244.516967@1] usb 1-1.1: device descriptor read/64, error -71
[  244.630176@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  244.630176@0] 
[  244.633882@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  244.633882@0] 
[  244.637276@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  244.637276@0] 
[  244.717026@1] usb 1-1.1: device descriptor read/64, error -71
[  244.906998@1] usb 1-1.1: new full-speed USB device number 18 using dwc_otg
[  244.910335@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  244.910335@0] 
[  244.913758@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  244.913758@0] 
[  244.917196@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  244.917196@0] 
[  244.997149@1] usb 1-1.1: device descriptor read/64, error -71
[  245.109948@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  245.109948@0] 
[  245.113383@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  245.113383@0] 
[  245.117184@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  245.117184@0] 
[  245.197178@1] usb 1-1.1: device descriptor read/64, error -71
[  245.386957@1] usb 1-1.1: new full-speed USB device number 19 using dwc_otg
[  245.409943@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  245.409943@0] 
[  245.413470@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  245.413470@0] 
[  245.417066@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  245.417066@0] 
[  245.417104@1] usb 1-1.1: device descriptor read/8, error -71
[  245.550778@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  245.550778@0] 
[  245.554149@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  245.554149@0] 
[  245.557613@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  245.557613@0] 
[  245.557643@3] usb 1-1.1: device descriptor read/8, error -71
[  245.746988@3] usb 1-1.1: new full-speed USB device number 20 using dwc_otg
[  245.770163@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  245.770163@0] 
[  245.773552@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  245.773552@0] 
[  245.776897@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  245.776897@0] 
[  245.776923@3] usb 1-1.1: device descriptor read/8, error -71
[  245.909919@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  245.909919@0] 
[  245.913330@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  245.913330@0] 
[  245.917004@0] ERROR::handle_hc_nak_intr:1306: Can not read device info from hub.We take it error
[  245.917004@0] 
[  245.917036@3] usb 1-1.1: device descriptor read/8, error -71
[  246.027161@3] hub 1-1:1.0: unable to enumerate USB device on port 1
```

Kivutar, 2017-04-07:
@Ntemis will test if the official DS3 is working on S905 which has a similar kernel.

Kivutar, 2017-05-25:
j0nnymoe confirm that it works on WeTek Hub

Kivutar, 2017-09-28:
Fixed on C2 after the 8.2 merge

Kivutar, 2017-09-28:
Oups, it was a DS4

Raybuntu, 2017-10-17:
Mentioning it here cause I just stumbled across this issue:
https://github.com/LibreELEC/linux-amlogic/pull/62

Kivutar, 2017-10-17:
Perfect. Thank you for this precious information!

Kivutar, 2017-10-20:
His PR got merged in 8.2. We should merge the latest 8.2 changes.

Raybuntu, 2017-10-20:
@Kivutar linux patch is not merged yet but it will be merged soon.

Kivutar, 2017-10-20:
Ah OK, I was thinking that it was https://github.com/LibreELEC/LibreELEC.tv/pull/2130

Raybuntu, 2017-10-20:
That's required too but the fix for the C2 is: https://github.com/LibreELEC/linux-amlogic/pull/62

closed 2017-10-24 by Kivutar

steff75, 2018-07-10:
I have the latest Lakka nightly installed on Odroid C2 and coneccted my Dualshock 3.
Buttons are working. What not works:

- Rumble/Vibration
- LED connection status

commit aa68b6e5 (2017-10-24): Fix DS3 support on C2. Fixes #54