libretro/Lakka-LibreELEC #58: Request: Lakka to accept language translations via Transifex services?

Hedda, 2017-02-22:
Suggest that developers setup a new project for Lakka on Transifex translation services (just as suggested seperate projects on Transifex for RetroArch and libretro-database) to get crowdsourced translations from your community.

https://en.wikipedia.org/wiki/Transifex

Many popular open source software projects use Transifex workflows to get free string language translations via crowdsourcing from their communities.

Example of an open source project using Transifex for many years is Kodi as you can read here:
https://kodi.tv/new-translation-service/
and here below (and it now have been translation to 70+ langauges):
http://kodi.wiki/view/Translation_System

For end-users wanting to help out with the project using Transifex to translate strings is as easy as using a wiki.

Users just need to register an account for free on Transifex then join the project for the languages they know and begin to translate strings they understand.

All you need to do is point to files on GitHub you want translating and then copy/pull them from Transifex on a regular bases (usually before each release).

Kivutar, 2017-02-22:
You need to move this request to libretro/RetroArch issue tracker. RetroArch is the frontend used in Lakka, this is what needs to be translated.

closed 2017-02-22 by Kivutar

comradekingu, 2021-12-20:
@Hedda Transifex is neither free, nor used by https://kodi.weblate.cloud/ now.
I know this request is old, but for the record :)