libretro/Lakka-LibreELEC #60: LAKKA as AccessPoint - enchancement

duduke, 2017-02-27:
`[Unit]
Description=Local AP via Connman
ConditionKernelCommandLine=|localap
ConditionPathExists=|/storage/.cache/services/localap.conf

[Service]
EnvironmentFile=-/storage/.cache/services/localap.conf
#ExecStart=/usr/bin/sshd -D $SSH_ARGS
ExecStart=/usr/bin/connmanctl tether wifi on $APNAME $PASSWORD
ExecStop=/usr/bin/connmanctl tether wifi off
TimeoutStopSec=1s
StartLimitInterval=0

[Install]
WantedBy=multi-user.target`

The above service triggers when **localap** is defined as a kernel parameter (similar to how SSH works).
It will then look in /storage/.cache/services/localap.conf which should contain:
APNAME=LAKKA
PASSWORD=BLAHBLAH
Connman will then tether the WIFI AP (192.168.1.0/24) to the onboard ethernet (useful for hotels as well).

markwkidd, 2017-05-03:
I created a bounty to this issue. Seems like it would be a neat feature.

https://www.bountysource.com/issues/42484309-lakka-as-accesspoint-enchancement

Kivutar, 2017-05-11:
I added it here https://github.com/libretro/Lakka-LibreELEC/commit/8bd547482a6e4576e002b44efc95bd9bcccf41df

You can try this image please? http://static.kivutar.me/Lakka-RPi2.arm-8.0-devel-20170511070237-r26126-gf275cda.img.gz

markwkidd, 2017-05-26:
@Kivutar I spend much of May, June, and July traveling for my work. Thank you for taking on this project but I'm unfortunately not able to test that image in the immediate future.

Kivutar, 2017-06-01:
@duduke I get a

    connmanctl> tether wifi on LAKKA TEST
    Wifi SSID set
    Error setting wifi passphrase: Passphrase required

duduke, 2017-06-01:
I think the password should be at least 8 chars length. It is WPA2.

Kivutar, 2017-06-01:
OK, it works now, but I can't connect SSH:

    root@192.168.1.2's password: 
    Permission denied, please try again.

Kivutar, 2017-06-01:
Documented here: https://github.com/libretro/Lakka-LibreELEC/wiki/Lakka-as-AccessPoint

duduke, 2017-06-01:
Possibly a restart of SSHd and SMBd is required to pickup the new IP address.

closed 2017-06-03 by Kivutar