libretro/Lakka-LibreELEC #62: Automount local network shares

Kivutar, 2017-02-27:
First, we need to find a bugfix to allow retroarch to unzip content on network shares.

Then, we need to find a service that detects network shares and automatically mounts them.

markwkidd, 2017-05-03:
I started a bounty for this issue. Seems like it would be a neat feature.

https://www.bountysource.com/issues/42484309-lakka-as-accesspoint-enchancement

RobLoach, 2017-06-26:
http://wiki.openelec.tv/index.php/Mounting_network_shares

I find using a `/storage/.config/system.d/storage-retroarch.mount` file the easiest way to automount drives. Not sure if Lakka will wait until the network is connected before making the mount though.

Kivutar, 2017-06-26:
By automounting I mean automatic discovery of the shares on the network. Without manual files editing.

RobLoach, 2017-06-26:
Ah, that would be cool.

closed 2020-08-18 by Ntemis