libretro/Lakka-LibreELEC #64: Lakka not working on GTX 1050ti

loadrunner, 2017-02-28:
`
RetroArch [INFO] :: This is RetroArch version 1.4.1 (Git 7358ac5)
RetroArch [INFO] :: === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 SSE4 SSE4.2 AES 
Built: Feb 25 2017
RetroArch [INFO] :: Version: 1.4.1
RetroArch [INFO] :: Git: 7358ac5
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Loading default config.
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Redirecting save file to "/storage/savefiles/.srm".
RetroArch [INFO] :: Redirecting savestate to "/storage/savestates/.state".
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29999.99 Hz.
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/bilinear.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/nearest.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/stock.glsl"
RetroArch [INFO] :: Video @ fullscreen
RetroArch [INFO] :: Starting threaded video driver ...
RetroArch [ERROR] :: [KMS]: Couldn't find a suitable DRM device.
RetroArch [INFO] :: Found GL context: null
RetroArch [INFO] :: Detecting screen resolution 320x240.
RetroArch [INFO] :: [GL]: Vendor: (null), Renderer: (null).
RetroArch [INFO] :: [GL]: Version: (null).
Segmentation fault (core dumped)
`
I guess it's a nouveau issue, but is there way to get the nvidia proprietary driver working with Lakka?

Thanks.

gouchi, 2017-02-28:
You are using the latest Pascal NVidia graphic card and NVidia didn't provide yet signed redistributable firmware blobs so 2D/3D acceleration is not possible for now.

You can try to use NVidia proprietary driver if you use [this patch](https://github.com/libretro/Lakka/pull/391) or [this one](https://github.com/libretro/Lakka/pull/640). You need to adapt it to Lakka LE.

gouchi, 2017-03-01:
We will need to wait at least kernel 4.12.

Kivutar, 2017-05-22:
So I close this for now.

closed 2017-05-22 by Kivutar

mlnlbrt, 2018-12-10:
@loadrunner FYI, I have created a fork of Lakka-LibreELEC with X11 enabled some time ago:
https://github.com/mlnlbrt/Lakka-LibreELEC/tree/master-x11

I use it on a daily basis with GT 1030 card. 

You can try to build yourself, just compile with `PROJECT=Generic-X11`.

loadrunner, 2018-12-10:
I will try it. Thanks!

gouchi, 2019-03-08:
@loadrunner Can you try this [testing image](https://drive.google.com/drive/folders/128MaH09bjVcB2pR4Kj2XxdBabR-oDP75) ?

Thank you.

loadrunner, 2019-03-09:
It doesn't work.
nouveau doesn't seem to be installed.

```
Lakka:~ # lsmod
Module                  Size  Used by
snd_hda_codec_hdmi     49152  1
8021q                  24576  0
snd_hda_intel          36864  0
xpad                   36864  0
snd_hda_codec         106496  2 snd_hda_codec_hdmi,snd_hda_intel
snd_hda_core           65536  3 snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec
snd_hwdep              16384  1 snd_hda_codec
overlay                98304  5
Lakka:~ # modprobe nouveau
modprobe: FATAL: Module nouveau not found in directory /lib/modules/4.19.12
Lakka:~ # modprobe nvidia
modprobe: FATAL: Module nvidia not found in directory /lib/modules/4.19.12
Lakka:~ # lspci -nnk | grep -A 3 VGA
00:03.0 VGA compatible controller [0300]: NVIDIA Corporation GP104 [GeForce GTX 1080] [10de:1b80] (rev a1)
	Subsystem: Gigabyte Technology Co., Ltd GP104 [GeForce GTX 1080] [1458:3717]
00:03.1 Audio device [0403]: NVIDIA Corporation GP104 High Definition Audio Controller [10de:10f0] (rev a1)
	Subsystem: Gigabyte Technology Co., Ltd GP104 High Definition Audio Controller [1458:3717]
```

mlnlbrt, 2019-03-09:
@loadrunner try one of these builds:
http://le.builds.lakka.tv/exp/

one of them is Vulkan-only build and another one is X11 + OpenGL. More details here:
http://www.lakka.tv/articles/2019/01/16/lakka-generic-nvidia-experimental-builds/

loadrunner, 2019-03-09:
I did not try any games yet, but I booted the OpenGL one and everything looked fine.

gouchi, 2019-03-09:
@loadrunner Indeed it seems nouveau is not enabled. I will enable it again.

ToKe79, 2019-07-16:
@loadrunner we started working on LE92 based builds, which should now support your GPU. You can test the image from here - http://nightly.builds.lakka.tv/special_builds/vudiq/Generic-LE92/

loadrunner, 2019-07-19:
Yes, it's working.
Thanks!