libretro/Lakka-LibreELEC #66: Lakka 2.0 Virtual Box ,error xorg server

batserra, 2017-03-01:
Lakka 2.0 2017022010016-r25287.img.ova
VirtualBox 5.1.14

Just a new installed virtualbox and imported Lakka Image, after imported to virtualbox i recive.
 Failed to start xorg server. 

See systemctl status xorg.service for details.

Thanks....

This issue tracker is NOT for help or support. Please use the Lakka forum (https://forums.libretro.com/c/libretro/lakka-tv-general) or IRC @ #lakka on freenode.net for those kinds of posts.

BEFORE you report a bug make sure you have tried the latest nightly version of Lakka at http://sources.lakka.tv/nightly/. Your bug might be already fixed.

If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
- What system hardware are you using?
- What did you do?
- What did you expect to happen?
- What happened instead?

It is important to provide logs for debugging as part of your report. Instructions for generating logs can be found in the Troubleshooting Lakka documentation at http://www.lakka.tv/doc/Troubleshooting-Lakka/

Please delete this text before posting.

Kivutar, 2017-03-01:
Hello, only VMWare is supported for now. And you need to use the Virtual image, not the Generic image.

closed 2017-03-01 by Kivutar

WHK102, 2021-06-06:
Now is 2021, any update for virtualbox supports?

Ntemis, 2021-06-06:
Never VB support was dropped