libretro/Lakka-LibreELEC #68: Lakka not working with nvidia ion gpu

bigboo3000, 2017-03-02:
Hi

I tested lakka 2.0 build x64 from 28/02/2017 on my Acer Revo R2600 with nvidia ion GPU with hdmi.

The installation boots fine in text mode but after few seconds it doesn't display anything, I see the usb key working (led blicking) so the installation process seems to work.

I think it's a video driver problem.

I tested Libreelec 8.0 stable and the video driver works correctly there. There are using nvidia-legacy 340.102 for old nvidia gpu:
https://github.com/LibreELEC/LibreELEC.tv/tree/master/packages/x11/driver/xf86-video-nvidia-legacy

Could you add this driver to lakka and make it the default with old nvidia gpu?

Kivutar, 2017-03-02:
In Lakka we use nouveau instead of the proprietary drivers, to avoid using X11.

@gouchi Do you know if nouveau can support ION in drm/kms?

gouchi, 2017-03-02:
It should be [supported](https://nouveau.freedesktop.org/wiki/CodeNames). It is possible to get [Lakka system and graphic card logs](http://www.lakka.tv/doc/Troubleshooting-Lakka) ?

Thank you.

bigboo3000, 2017-03-02:
I can't see anything and SSH access doesn't seems to work, is SSH activated by default in lakka?

gouchi, 2017-03-02:
Please try to append to the [cmdline](http://www.lakka.tv/doc/The-bootloader/#on-generic-pc) `ssh`.

bigboo3000, 2017-03-02:
Lakka:~ # systemctl stop retroarch.service ; retroarch -v
RetroArch [INFO] :: This is RetroArch version 1.4.1 (Git 7358ac5)
RetroArch [INFO] :: === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3
Built: Feb 25 2017
RetroArch [INFO] :: Version: 1.4.1
RetroArch [INFO] :: Git: 7358ac5
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Loading default config.
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Redirecting save file to "/storage/savefiles/.srm".
RetroArch [INFO] :: Redirecting savestate to "/storage/savestates/.state".
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29970.03 Hz.
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/bilinear.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/nearest.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/stock.glsl"
RetroArch [INFO] :: Video @ fullscreen
RetroArch [INFO] :: Starting threaded video driver ...
RetroArch [INFO] :: [DRM]: Found 2 connectors.
RetroArch [INFO] :: [DRM]: Connector 0 connected: yes
RetroArch [INFO] :: [DRM]: Connector 0 has 38 modes.
RetroArch [INFO] :: [DRM]: Connector 0 assigned to monitor index: #1.
RetroArch [INFO] :: [DRM]: Connector 1 connected: no
RetroArch [INFO] :: [DRM]: Connector 1 has 0 modes.
RetroArch [WARN] :: [DRM]: Couldn't find DRM encoder.
RetroArch [WARN] :: [DRM]: Couldn't get device resources.
RetroArch [INFO] :: [DRM]: Found 2 connectors.
RetroArch [INFO] :: [DRM]: Connector 0 connected: yes
RetroArch [INFO] :: [DRM]: Connector 0 has 38 modes.
RetroArch [INFO] :: [DRM]: Connector 0 assigned to monitor index: #1.
RetroArch [INFO] :: [DRM]: Connector 1 connected: no
RetroArch [INFO] :: [DRM]: Connector 1 has 0 modes.
RetroArch [WARN] :: [DRM]: Couldn't find DRM encoder.
RetroArch [ERROR] :: [KMS]: Couldn't find a suitable DRM device.
RetroArch [INFO] :: Found GL context: null
RetroArch [INFO] :: Detecting screen resolution 320x240.
RetroArch [INFO] :: [GL]: Vendor: (null), Renderer: (null).
RetroArch [INFO] :: [GL]: Version: (null).
Segmentation fault (core dumped)

bigboo3000, 2017-03-02:
Lakka:~ # lspci -nnk | grep -A 3 VGA
03:00.0 VGA compatible controller [0300]: NVIDIA Corporation ION VGA [10de:087d] (rev b1)
        Subsystem: Acer Incorporated [ALI] ION VGA [1025:0222]
        Kernel driver in use: nouveau
05:00.0 Network controller [0280]: Qualcomm Atheros AR93xx Wireless Network Adapter [168c:0030] (rev 01)
        Subsystem: Apple Inc. AR93xx Wireless Network Adapter (AirPort Extreme) [106b:009a]

bigboo3000, 2017-03-02:
I used another pc to set ssh on in retroarch ui.
And FYI my pc with ion has 2 video output, one hdmi (that I use) and one VGA, maybe that's the problem?

Kivutar, 2017-03-02:
Note, to enable SSH, go in Setting->Services and switch it on.

bigboo3000, 2017-03-02:
Yep that's what i have done with my usb key on another pc

gouchi, 2017-03-03:
Can you try to append to the cmdline `drm.debug=0x1e log_buf_len=1M` to get more log and get the output of `dmesg` ?

Thank you.

bigboo3000, 2017-03-03:
When you say "append to the cmdline" it means that I must add this to syslinux.cfg after the line starting with "APPEND" in the "LIVE" section ?

gouchi, 2017-03-03:
It depends if you have installed Lakka (edit syslinux.cfg) or if you are using live mode (press tab in the boot menu) as it written in the [documentation](http://www.lakka.tv/doc/The-bootloader/#on-generic-pc).

bigboo3000, 2017-03-03:
`Lakka:~ # dmesg
[    0.000000] Linux version 4.9.8 (kivutar@Server01) (gcc version 6.2.0 (GCC) )  #1 SMP Sat Feb 25 11:11:28 CST 2017
[    0.000000] Command line: BOOT_IMAGE=/KERNEL boot=UUID=2602-2725 disk=UUID=f2 1c9cab-d5bc-4da9-bdc5-1bc955113fdd portable quiet drm.debug=0x1e log_buf_len=1M
[    0.000000] Disabled fast string operations
[    0.000000] x86/fpu: Legacy x87 FPU detected.
[    0.000000] x86/fpu: Using 'eager' FPU context switches.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ebff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009ec00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000006ff8ffff] usable
[    0.000000] BIOS-e820: [mem 0x000000006ff90000-0x000000006ff9dfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000006ff9e000-0x000000006ffdffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000006ffe0000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff700000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.6 present.
[    0.000000] DMI: Acer Aspire R3610/FMCP7A-ION-LE, BIOS P01-A4 11/03/2009
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x6ff90 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-CFFFF write-protect
[    0.000000]   D0000-DFFFF uncachable
[    0.000000]   E0000-EFFFF write-through
[    0.000000]   F0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 000000000 mask 080000000 write-back
[    0.000000]   1 disabled
[    0.000000]   2 disabled
[    0.000000]   3 disabled
[    0.000000]   4 disabled
[    0.000000]   5 disabled
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WC  UC- WT
[    0.000000] found SMP MP-table at [mem 0x000ff780-0x000ff78f] mapped at [ffff 8800000ff780]
[    0.000000] Scanning 1 areas for low memory corruption
[    0.000000] Base memory trampoline at [ffff880000098000] 98000 size 24576
[    0.000000] BRK [0x03767000, 0x03767fff] PGTABLE
[    0.000000] BRK [0x03768000, 0x03768fff] PGTABLE
[    0.000000] BRK [0x03769000, 0x03769fff] PGTABLE
[    0.000000] BRK [0x0376a000, 0x0376afff] PGTABLE
[    0.000000] BRK [0x0376b000, 0x0376bfff] PGTABLE
[    0.000000] log_buf_len: 1048576 bytes
[    0.000000] early log buf free: 62640(95%)
[    0.000000] ACPI: Early table checksum verification disabled
[    0.000000] ACPI: RSDP 0x00000000000FB770 000024 (v02 ACPIAM)
[    0.000000] ACPI: XSDT 0x000000006FF90100 00006C (v01 ACRSYS ACRPRDCT 2009110 3 MSFT 00000097)
[    0.000000] ACPI: FACP 0x000000006FF90290 0000F4 (v03 ACRSYS FACP0757 2009110 3 MSFT 00000097)
[    0.000000] ACPI: DSDT 0x000000006FF90620 006CAA (v01 963A1  963A1P06 0000000 4 INTL 20051117)
[    0.000000] ACPI: FACS 0x000000006FF9E000 000040
[    0.000000] ACPI: FACS 0x000000006FF9E000 000040
[    0.000000] ACPI: APIC 0x000000006FF90390 000080 (v01 ACRSYS APIC0757 2009110 3 MSFT 00000097)
[    0.000000] ACPI: MCFG 0x000000006FF90410 00003C (v01 ACRSYS OEMMCFG  2009110 3 MSFT 00000097)
[    0.000000] ACPI: SLIC 0x000000006FF90450 000176 (v01 ACRSYS ACRPRDCT 2009110 3 MSFT 00000097)
[    0.000000] ACPI: WDRT 0x000000006FF905D0 000047 (v01 ACRSYS NV-WDRT  2009110 3 MSFT 00000097)
[    0.000000] ACPI: OEMB 0x000000006FF9E040 000079 (v01 ACRSYS OEMB0757 2009110 3 MSFT 00000097)
[    0.000000] ACPI: HPET 0x000000006FF9A620 000038 (v01 ACRSYS OEMHPET0 2009110 3 MSFT 00000097)
[    0.000000] ACPI: NVHD 0x000000006FF9E0C0 000284 (v01 ACRSYS NVHDCP   2009110 3 MSFT 00000097)
[    0.000000] ACPI: AWMI 0x000000006FF9E350 00004E (v01 ACRSYS OEMB0757 2009110 3 MSFT 00000097)
[    0.000000] ACPI: Local APIC address 0xfee00000
[    0.000000] No NUMA configuration found
[    0.000000] Faking a node at [mem 0x0000000000000000-0x000000006ff8ffff]
[    0.000000] NODE_DATA(0) allocated [mem 0x6fe8c000-0x6fe8ffff]
[... 1290 lines cut ...]
[   48.352813] [drm:drm_mode_debug_printmodeline] Modeline 50:"640x480" 75 31500  640 656 720 840 480 481 484 500 0x40 0xa
[   48.352820] [drm:drm_mode_debug_printmodeline] Modeline 51:"640x480" 73 31500  640 664 704 832 480 489 492 520 0x40 0xa
[   48.352828] [drm:drm_mode_debug_printmodeline] Modeline 114:"640x480" 60 2520 0 640 656 752 800 480 490 492 525 0x40 0xa
[   48.352835] [drm:drm_mode_debug_printmodeline] Modeline 52:"640x480" 60 25175  640 656 752 800 480 490 492 525 0x40 0xa
[   49.317826] [drm:drm_mode_addfb2] [FB:73]
[   49.317865] [drm:drm_mode_setcrtc] [CRTC:34:crtc-0]
[   49.317876] [drm:drm_mode_setcrtc] [CONNECTOR:37:HDMI-A-1]
[   49.317884] [drm:drm_crtc_helper_set_config]
[   49.317890] [drm:drm_crtc_helper_set_config] [CRTC:34:crtc-0] [FB:73] #connec tors=1 (x y) (0 0)
[   49.317898] [drm:drm_crtc_helper_set_config] modes are different, full mode s et
[... cut ...]
[   53.817561] [drm:drm_detect_monitor_audio] Monitor has basic audio support
[   53.817567] [drm:drm_edid_to_eld] ELD monitor TOSHIBA-TV
[   53.817575] [drm:drm_edid_to_eld] HDMI: latency present 1 1, video latency 11 1 121, audio latency 111 121
[   53.817581] [drm:nv50_audio_mode_set.isra.12] ELD size 36, SAD count 2
[   56.068412] [drm:drm_crtc_helper_set_config] Setting connector DPMS state to  on
[   56.068421] [drm:drm_crtc_helper_set_config]         [CONNECTOR:37:HDMI-A-1]  set DPMS on
[   59.541870] [drm:drm_mode_addfb2] [FB:74]
`

gouchi, 2017-03-03:
Thank you for the log.

Did you try to make a test also with HDMI monitor ? As it is plugged now on TV.

bigboo3000, 2017-03-03:
this is with hdmi > dvi cable to a monitor:

Lakka:~ # dmesg
[    0.000000] Linux version 4.9.8 (kivutar@Server01) (gcc version 6.2.0 (GCC) ) #1 SMP Sat Feb 25 11:11:28 CST 2017
[    0.000000] Command line: BOOT_IMAGE=/KERNEL boot=UUID=2602-2725 disk=UUID=f21c9cab-d5bc-4da9-bdc5-1bc955113fdd portable quiet drm.debug=0x1e log_buf_len=1M
[    0.000000] Disabled fast string operations
[    0.000000] x86/fpu: Legacy x87 FPU detected.
[    0.000000] x86/fpu: Using 'eager' FPU context switches.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ebff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009ec00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000006ff8ffff] usable
[    0.000000] BIOS-e820: [mem 0x000000006ff90000-0x000000006ff9dfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000006ff9e000-0x000000006ffdffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000006ffe0000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff700000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.6 present.
[    0.000000] DMI: Acer Aspire R3610/FMCP7A-ION-LE, BIOS P01-A4 11/03/2009
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x6ff90 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-CFFFF write-protect
[    0.000000]   D0000-DFFFF uncachable
[    0.000000]   E0000-EFFFF write-through
[    0.000000]   F0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 000000000 mask 080000000 write-back
[    0.000000]   1 disabled
[    0.000000]   2 disabled
[    0.000000]   3 disabled
[    0.000000]   4 disabled
[    0.000000]   5 disabled
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WC  UC- WT
[    0.000000] found SMP MP-table at [mem 0x000ff780-0x000ff78f] mapped at [ffff8800000ff780]
[    0.000000] Scanning 1 areas for low memory corruption
[    0.000000] Base memory trampoline at [ffff880000098000] 98000 size 24576
[    0.000000] BRK [0x03767000, 0x03767fff] PGTABLE
[    0.000000] BRK [0x03768000, 0x03768fff] PGTABLE
[    0.000000] BRK [0x03769000, 0x03769fff] PGTABLE
[    0.000000] BRK [0x0376a000, 0x0376afff] PGTABLE
[    0.000000] BRK [0x0376b000, 0x0376bfff] PGTABLE
[    0.000000] log_buf_len: 1048576 bytes
[    0.000000] early log buf free: 62640(95%)
[    0.000000] ACPI: Early table checksum verification disabled
[    0.000000] ACPI: RSDP 0x00000000000FB770 000024 (v02 ACPIAM)
[    0.000000] ACPI: XSDT 0x000000006FF90100 00006C (v01 ACRSYS ACRPRDCT 20091103 MSFT 00000097)
[    0.000000] ACPI: FACP 0x000000006FF90290 0000F4 (v03 ACRSYS FACP0757 20091103 MSFT 00000097)
[    0.000000] ACPI: DSDT 0x000000006FF90620 006CAA (v01 963A1  963A1P06 00000004 INTL 20051117)
[    0.000000] ACPI: FACS 0x000000006FF9E000 000040
[    0.000000] ACPI: FACS 0x000000006FF9E000 000040
[    0.000000] ACPI: APIC 0x000000006FF90390 000080 (v01 ACRSYS APIC0757 20091103 MSFT 00000097)
[    0.000000] ACPI: MCFG 0x000000006FF90410 00003C (v01 ACRSYS OEMMCFG  20091103 MSFT 00000097)
[    0.000000] ACPI: SLIC 0x000000006FF90450 000176 (v01 ACRSYS ACRPRDCT 20091103 MSFT 00000097)
[    0.000000] ACPI: WDRT 0x000000006FF905D0 000047 (v01 ACRSYS NV-WDRT  20091103 MSFT 00000097)
[    0.000000] ACPI: OEMB 0x000000006FF9E040 000079 (v01 ACRSYS OEMB0757 20091103 MSFT 00000097)
[    0.000000] ACPI: HPET 0x000000006FF9A620 000038 (v01 ACRSYS OEMHPET0 20091103 MSFT 00000097)
[    0.000000] ACPI: NVHD 0x000000006FF9E0C0 000284 (v01 ACRSYS NVHDCP   20091103 MSFT 00000097)
[    0.000000] ACPI: AWMI 0x000000006FF9E350 00004E (v01 ACRSYS OEMB0757 20091103 MSFT 00000097)
[    0.000000] ACPI: Local APIC address 0xfee00000
[    0.000000] No NUMA configuration found
[... 767 lines cut ...]
[   36.725181] [drm:drm_mode_debug_printmodeline] Modeline 55:"1280x1024" 75 135000 1280 1296 1440 1688 1024 1025 1028 1066 0x40 0x5
[   36.725189] [drm:drm_mode_debug_printmodeline] Modeline 45:"1280x1024" 60 108000 1280 1328 1440 1688 1024 1025 1028 1066 0x40 0x5
[   36.725197] [drm:drm_mode_debug_printmodeline] Modeline 44:"1440x900" 60 88750 1440 1488 1520 1600 900 903 909 926 0x40 0x9
[   36.725204] [drm:drm_mode_debug_printmodeline] Modeline 46:"1280x960" 60 108000 1280 1376 1488 1800 960 961 964 1000 0x40 0x5
[   36.725212] [drm:drm_mode_debug_printmodeline] Modeline 47:"1280x720" 60 74250 1280 1390 1430 1650 720 725 730 750 0x40 0x5
[   36.725219] [drm:drm_mode_debug_printmodeline] Modeline 56:"1024x768" 75 78750 1024 1040 1136 1312 768 769 772 800 0x40 0x5
[   36.725227] [drm:drm_mode_debug_printmodeline] Modeline 57:"1024x768" 70 75000 1024 1048 1184 1328 768 771 777 806 0x40 0xa
[   36.725234] [drm:drm_mode_debug_printmodeline] Modeline 58:"1024x768" 60 65000 1024 1048 1184 1344 768 771 777 806 0x40 0xa
[... cut ...]
[   36.725308] [drm:drm_mode_debug_printmodeline] Modeline 54:"720x400" 70 28320 720 738 846 900 400 412 414 449 0x40 0x6
[   37.671903] [drm:drm_mode_addfb2] [FB:63]
[   37.671942] [drm:drm_mode_setcrtc] [CRTC:34:crtc-0]
[   37.671953] [drm:drm_mode_setcrtc] [CONNECTOR:37:HDMI-A-1]
[   37.671961] [drm:drm_crtc_helper_set_config]
[   37.671967] [drm:drm_crtc_helper_set_config] [CRTC:34:crtc-0] [FB:63] #connectors=1 (x y) (0 0)
[   37.671977] [drm:drm_crtc_helper_set_config] [CONNECTOR:37:HDMI-A-1] to [CRTC:34:crtc-0]
[   44.230769] [drm:drm_mode_addfb2] [FB:65]

gouchi, 2017-03-03:
I have some confirmation from nouveau dev that it should work in drm/kms.

So we need to find why RA doesn't find drm [encoders](https://github.com/libretro/RetroArch/blob/master/gfx/common/drm_common.c#L125) and [devices](https://github.com/libretro/RetroArch/blob/master/gfx/drivers_context/drm_ctx.c#L337) ?

What is the output of ?
`ls -l  /sys/class/drm`

Thank you.

bigboo3000, 2017-03-03:
FYI I have the exact same problem with Recalbox X64 4.1 unstable, it also use nouveau I think.

Lakka:~ # ls -l /sys/class/drm
total 0
lrwxrwxrwx    1 root     root             0 Mar  3 20:37 card0 -> ../../devices/pci0000:00/0000:00:10.0/0000:03:00.0/drm/card0
lrwxrwxrwx    1 root     root             0 Mar  3 20:37 card0-HDMI-A-1 -> ../../devices/pci0000:00/0000:00:10.0/0000:03:00.0/drm/card0/card0-HDMI-A-1
lrwxrwxrwx    1 root     root             0 Mar  3 20:37 card0-VGA-1 -> ../../devices/pci0000:00/0000:00:10.0/0000:03:00.0/drm/card0/card0-VGA-1
lrwxrwxrwx    1 root     root             0 Mar  3 20:37 controlD64 -> ../../devices/pci0000:00/0000:00:10.0/0000:03:00.0/drm/controlD64
lrwxrwxrwx    1 root     root             0 Mar  3 20:37 renderD128 -> ../../devices/pci0000:00/0000:00:10.0/0000:03:00.0/drm/renderD128
lrwxrwxrwx    1 root     root             0 Mar  3 20:37 ttm -> ../../devices/virtual/drm/ttm
-r--r--r--    1 root     root          4096 Mar  3 20:37 version

gouchi, 2017-03-04:
Do you reproduce the issue with another Linux distribution if you compile RA ([enable KMS](https://github.com/libretro/RetroArch/wiki/KMS-mode)) and run it from virtual console (CTRL+ALT+F1) ?

Thank you.

bigboo3000, 2017-03-04:
I don't know Linux very much so it become too difficult to me to install another distibution and compile RA.

But Recalbox starts EmulationStation when booting (and not RA) but I don't see anything either.

gouchi, 2017-03-06:
Can you try to get log from Recalbox to see which error you have ?
It seems you need to change the RA command line in `/recalbox/scripts/emulatorlauncher.sh` to enable the [log](http://www.lakka.tv/doc/Troubleshooting-Lakka/#option-b-creating-a-log-file).

Thank you.

bigboo3000, 2017-03-06:
ok this is the log from batocera linux 5.3 x64 (a fork of recalbox, because the download site of recalbox doesn't work currently so I can't download an image)

bigboo3000, 2017-03-06:
# retroarch -v
RetroArch [INFO] :: === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3
Built: Feb 19 2017
RetroArch [INFO] :: Version: 1.4.0
RetroArch [INFO] :: Git: a1b30f0
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Loading default config.
RetroArch [INFO] :: Looking for config in: "/recalbox/share/system/.config/retroarch/retroarch.cfg".
RetroArch [WARN] :: screenshot_directory is not an existing directory, ignoring ...
RetroArch [INFO] :: Incompatible shader for backend gl, clearing...
RetroArch [INFO] :: Resetting undo buffers.
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29970.03 Hz.
RetroArch [INFO] :: Video @ 960x720
RetroArch [INFO] :: [DRM]: Found 2 connectors.
RetroArch [INFO] :: [DRM]: Connector 0 connected: yes
RetroArch [INFO] :: [DRM]: Connector 0 has 38 modes.
RetroArch [INFO] :: [DRM]: Connector 0 assigned to monitor index: #1.
RetroArch [INFO] :: [DRM]: Connector 1 connected: no
RetroArch [INFO] :: [DRM]: Connector 1 has 0 modes.
RetroArch [INFO] :: [DRM]: Mode 0: (1920x1080) 1920 x 1080, 50 Hz
RetroArch [INFO] :: [DRM]: Mode 1: (1920x1080) 1920 x 1080, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 2: (1920x1080) 1920 x 1080, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 3: (1920x1080i) 1920 x 1080, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 4: (1920x1080i) 1920 x 1080, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 5: (1920x1080i) 1920 x 1080, 50 Hz
RetroArch [INFO] :: [DRM]: Mode 6: (1920x1080) 1920 x 1080, 30 Hz
RetroArch [INFO] :: [DRM]: Mode 7: (1920x1080) 1920 x 1080, 30 Hz
RetroArch [INFO] :: [DRM]: Mode 8: (1920x1080) 1920 x 1080, 25 Hz
RetroArch [INFO] :: [DRM]: Mode 9: (1920x1080) 1920 x 1080, 24 Hz
RetroArch [INFO] :: [DRM]: Mode 10: (1920x1080) 1920 x 1080, 24 Hz
RetroArch [INFO] :: [DRM]: Mode 11: (1280x1024) 1280 x 1024, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 12: (1280x1024) 1280 x 1024, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 13: (1360x768) 1360 x 768, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 14: (1280x768) 1280 x 768, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 15: (1280x768) 1280 x 768, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 16: (1280x768) 1280 x 768, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 17: (1280x720) 1280 x 720, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 18: (1280x720) 1280 x 720, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 19: (1280x720) 1280 x 720, 50 Hz
RetroArch [INFO] :: [DRM]: Mode 20: (1280x720) 1280 x 720, 30 Hz
RetroArch [INFO] :: [DRM]: Mode 21: (1280x720) 1280 x 720, 30 Hz
RetroArch [INFO] :: [DRM]: Mode 22: (1280x720) 1280 x 720, 24 Hz
RetroArch [INFO] :: [DRM]: Mode 23: (1280x720) 1280 x 720, 24 Hz
RetroArch [INFO] :: [DRM]: Mode 24: (1024x768) 1024 x 768, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 25: (1024x768) 1024 x 768, 70 Hz
RetroArch [INFO] :: [DRM]: Mode 26: (1024x768) 1024 x 768, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 27: (800x600) 800 x 600, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 28: (800x600) 800 x 600, 72 Hz
RetroArch [INFO] :: [DRM]: Mode 29: (800x600) 800 x 600, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 30: (800x600) 800 x 600, 56 Hz
RetroArch [INFO] :: [DRM]: Mode 31: (720x576) 720 x 576, 50 Hz
RetroArch [INFO] :: [DRM]: Mode 32: (720x480) 720 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 33: (720x480) 720 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 34: (640x480) 640 x 480, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 35: (640x480) 640 x 480, 73 Hz
RetroArch [INFO] :: [DRM]: Mode 36: (640x480) 640 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 37: (640x480) 640 x 480, 60 Hz
RetroArch [INFO] :: Found GL context: kms
RetroArch [INFO] :: Detecting screen resolution 1920x1080.
RetroArch [INFO] :: [EGL]: EGL version: 1.4
RetroArch [INFO] :: [EGL]: Current context: 0x10f0a20.
RetroArch [INFO] :: [KMS]: New FB: 1920x1080 (stride: 7680).
RetroArch [ERROR] :: Cannot open video driver ... Exiting ...
RetroArch [ERROR] :: Fatal error received in: "init_video()"

bigboo3000, 2017-03-06:
# lspci -nnk | grep -A 3 VGA
#

bigboo3000, 2017-03-06:
[    0.000000] Linux version 4.9.0 (nicolas@setsouko) (gcc version 4.9.4 (Buildroot 2017.02-git-ga1b30f0) ) #1 SMP Sun Feb 19 04:59:44 CET 2017
[    0.000000] Command line: BOOT_IMAGE=/boot/linux label=RECALBOX console=tty3 quiet loglevel=0 vt.global_cursor_default=0
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Centaur CentaurHauls
[    0.000000] Disabled fast string operations
[    0.000000] x86/fpu: Legacy x87 FPU detected.
[    0.000000] x86/fpu: Using 'eager' FPU context switches.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ebff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009ec00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000006ff8ffff] usable
[    0.000000] BIOS-e820: [mem 0x000000006ff90000-0x000000006ff9dfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000006ff9e000-0x000000006ffdffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000006ffe0000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff700000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.6 present.
[    0.000000] DMI: Acer Aspire R3610/FMCP7A-ION-LE, BIOS P01-A4 11/03/2009
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x6ff90 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-CFFFF write-protect
[    0.000000]   D0000-DFFFF uncachable
[    0.000000]   E0000-EFFFF write-through
[    0.000000]   F0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 000000000 mask 080000000 write-back
[    0.000000]   1 disabled
[    0.000000]   2 disabled
[    0.000000]   3 disabled
[    0.000000]   4 disabled
[    0.000000]   5 disabled
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WC  UC- WT
[    0.000000] found SMP MP-table at [mem 0x000ff780-0x000ff78f] mapped at [ffff8dc0400ff780]
[    0.000000] Scanning 1 areas for low memory corruption
[    0.000000] Base memory trampoline at [ffff8dc040098000] 98000 size 24576
[    0.000000] BRK [0x5947b000, 0x5947bfff] PGTABLE
[    0.000000] BRK [0x5947c000, 0x5947cfff] PGTABLE
[    0.000000] BRK [0x5947d000, 0x5947dfff] PGTABLE
[    0.000000] BRK [0x5947e000, 0x5947efff] PGTABLE
[    0.000000] BRK [0x5947f000, 0x5947ffff] PGTABLE
[    0.000000] BRK [0x59480000, 0x59480fff] PGTABLE
[    0.000000] RAMDISK: [mem 0x37ee4000-0x37f69fff]
[    0.000000] ACPI: Early table checksum verification disabled
[    0.000000] ACPI: RSDP 0x00000000000FB770 000024 (v02 ACPIAM)
[    0.000000] ACPI: XSDT 0x000000006FF90100 00006C (v01 ACRSYS ACRPRDCT 20091103 MSFT 00000097)
[    0.000000] ACPI: FACP 0x000000006FF90290 0000F4 (v03 ACRSYS FACP0757 20091103 MSFT 00000097)
[    0.000000] ACPI: DSDT 0x000000006FF90620 006CAA (v01 963A1  963A1P06 00000004 INTL 20051117)
[    0.000000] ACPI: FACS 0x000000006FF9E000 000040
[    0.000000] ACPI: FACS 0x000000006FF9E000 000040
[    0.000000] ACPI: APIC 0x000000006FF90390 000080 (v01 ACRSYS APIC0757 20091103 MSFT 00000097)
[    0.000000] ACPI: MCFG 0x000000006FF90410 00003C (v01 ACRSYS OEMMCFG  20091103 MSFT 00000097)
[    0.000000] ACPI: SLIC 0x000000006FF90450 000176 (v01 ACRSYS ACRPRDCT 20091103 MSFT 00000097)
[    0.000000] ACPI: WDRT 0x000000006FF905D0 000047 (v01 ACRSYS NV-WDRT  20091103 MSFT 00000097)
[    0.000000] ACPI: OEMB 0x000000006FF9E040 000079 (v01 ACRSYS OEMB0757 20091103 MSFT 00000097)
[    0.000000] ACPI: HPET 0x000000006FF9A620 000038 (v01 ACRSYS OEMHPET0 20091103 MSFT 00000097)
[    0.000000] ACPI: NVHD 0x000000006FF9E0C0 000284 (v01 ACRSYS NVHDCP   20091103 MSFT 00000097)
[    0.000000] ACPI: AWMI 0x000000006FF9E350 00004E (v01 ACRSYS OEMB0757 20091103 MSFT 00000097)
[    0.000000] ACPI: Local APIC address 0xfee00000
[... 647 lines cut ...]
[    7.632115] snd_hda_codec_realtek hdaudioC0D0:      Mic=0x18
[    8.994060] input: HDA NVidia Mic as /devices/pci0000:00/0000:00:08.0/sound/card0/input7
[    8.994565] input: HDA NVidia Headphone as /devices/pci0000:00/0000:00:08.0/sound/card0/input8
[   13.419436] Console: switching to colour frame buffer device 240x67
[   13.422217] nouveau 0000:03:00.0: fb0: nouveaufb frame buffer device
[   13.440092] [drm] Initialized nouveau 1.3.1 20120801 for 0000:03:00.0 on minor 0
[   13.884408] EXT4-fs (sdb2): couldn't mount as ext3 due to feature incompatibilities
[   13.894527] EXT4-fs (sdb2): couldn't mount as ext2 due to feature incompatibilities
[   14.374852] EXT4-fs (sdb2): recovery complete
[   14.387680] EXT4-fs (sdb2): mounted filesystem with ordered data mode. Opts: (null)
[   14.468646] nf_conntrack version 0.5.0 (16384 buckets, 65536 max)
[   14.587787] EXT4-fs (sda1): couldn't mount as ext3 due to feature incompatibilities
[   14.588299] EXT4-fs (sda1): couldn't mount as ext2 due to feature incompatibilities
[... cut ...]
[   14.875831] forcedeth 0000:00:0a.0 eth0: no link during initialization
[   14.876257] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   14.893378] EXT4-fs (sda2): mounted filesystem with ordered data mode. Opts: (null)
[   19.136326] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   19.136331] Bluetooth: BNEP filters: protocol multicast
[   19.136349] Bluetooth: BNEP socket layer initialized
[   22.737672] forcedeth 0000:00:0a.0 eth0: link up
[   22.738543] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   23.241688] 8021q: 802.1Q VLAN Support v1.8

bigboo3000, 2017-03-06:
# ls -l /sys/class/drm
total 0
lrwxrwxrwx    1 root     root             0 Jan  1  2009 card0 -> ../../devices/pci0000:00/0000:00:10.0/0000:03:00.0/drm/card0
lrwxrwxrwx    1 root     root             0 Jan  1  2009 card0-HDMI-A-1 -> ../../devices/pci0000:00/0000:00:10.0/0000:03:00.0/drm/card0/card0-HDMI-A-1
lrwxrwxrwx    1 root     root             0 Jan  1  2009 card0-VGA-1 -> ../../devices/pci0000:00/0000:00:10.0/0000:03:00.0/drm/card0/card0-VGA-1
lrwxrwxrwx    1 root     root             0 Jan  1  2009 controlD64 -> ../../devices/pci0000:00/0000:00:10.0/0000:03:00.0/drm/controlD64
lrwxrwxrwx    1 root     root             0 Jan  1  2009 renderD128 -> ../../devices/pci0000:00/0000:00:10.0/0000:03:00.0/drm/renderD128
lrwxrwxrwx    1 root     root             0 Jan  1  2009 ttm -> ../../devices/virtual/drm/ttm
-r--r--r--    1 root     root          4096 Jan  1  2009 version

gouchi, 2017-04-09:
Can you try to boot to with 

`live video=VGA-1:d video=HDMI-A-1:e`

It will force the  output to HDMI.

Thank you.

bigboo3000, 2017-04-12:
it still doesn't display anything with that.

minipretzel, 2017-05-01:
Same situation.  

[3610_aplay.txt](https://github.com/libretro/Lakka-LibreELEC/files/968518/3610_aplay.txt)
[3610_dmesg.txt](https://github.com/libretro/Lakka-LibreELEC/files/968521/3610_dmesg.txt)
[3610_drm.txt](https://github.com/libretro/Lakka-LibreELEC/files/968519/3610_drm.txt)
[3610_drm_status.txt](https://github.com/libretro/Lakka-LibreELEC/files/968520/3610_drm_status.txt)
[3610_lspci.txt](https://github.com/libretro/Lakka-LibreELEC/files/968523/3610_lspci.txt)
[3610_lspci_vga.txt](https://github.com/libretro/Lakka-LibreELEC/files/968524/3610_lspci_vga.txt)
[3610_lsusb.txt](https://github.com/libretro/Lakka-LibreELEC/files/968522/3610_lsusb.txt)
[3610_retroarch.txt](https://github.com/libretro/Lakka-LibreELEC/files/968525/3610_retroarch.txt)

nighoa, 2017-05-02:
Same thing here with an ASROCK ION 330.

I can install with option : $ install nomodeset

Everything run smooth, but after the first reboot... black screen and voila...

If I can give you more detail, or try some options, don't hesitate.

bigboo3000, 2017-07-14:
This issue must be this:
https://bugs.freedesktop.org/show_bug.cgi?id=101792
https://bugs.freedesktop.org/show_bug.cgi?id=99858

It's a bug with kernel 4.9 fixed in 4.10.

Could you update lakka pc kernel to 4.10?

Kivutar, 2017-07-14:
If LibreELEC does it yes, if not no

bigboo3000, 2017-07-14:
it seems libreelec 8.2 branch is using kernel 4.11, so maybe in lakka 2.2?

Kivutar, 2017-07-16:
Yes, do you know when they started this 8.2 branch? I wonder if we can merge it safely.

bigboo3000, 2017-07-16:
I'm not an expert of github but it seems 8.2 branch have been created 15 june, one month ago. On their site they say they plan to release it in july or august.

gouchi, 2017-07-30:
I confirm that Lakka 2.1 RC1 is working on Nvidia Ion gpu. For the hdmi output it is working with kernel >= 4.10.

bigboo3000, 2017-08-30:
I tested 2.1 RC3 and it's fixed.

closed 2017-08-30 by bigboo3000