libretro/Lakka-LibreELEC #69: Issues with Boxart

kryptalivian, 2017-03-04:
I'm using a pi 3 with Lakka-RPi2.arm-8.0-devel-20170226163235-r25805-g900da08.img . 

I've found that when I point Lakka to my box art directory(**downloaded from pc and put on flash drive in own directory**) Wonderswan's, and Atari wont display box art. 

nes, snes, sega, n64, etc will grab the art work and display them.

Kivutar, 2017-03-11:
Certainly a name mismatch for these two systems. We need to compare the names here https://github.com/libretro/libretro-database/tree/master/rdb and here http://thumbnailpacks.libretro.com/

Kivutar, 2017-04-01:
I just tested Wonderswan Color, and I was able to see the boxart.

What Atari version?

kryptalivian, 2017-04-10:
Loaned out flash drive that has the boxart. Closing for now.

closed 2017-04-10 by kryptalivian