libretro/Lakka-LibreELEC #76: Freezing when scanning PSX games

kryptalivian, 2017-03-07:
Running Pi2 image on Pi 3. Running fresh copy of RC2. Still having issues when scanning psx games. It will go 1/8 2/8 then stop and then system becomes unresponsive. Let it set for 15 minutes, and nothing changed.

Kivutar, 2017-03-13:
It can be due to the bin scanning, when using bin+cue. The cue will force a serial scanning on the bin, which is fast, but the bin itself will also be scanned by crc (because megadrive games can use .bin extension too). And that could cause the freeze.

ToKe79, 2018-01-31:
Issue closed as no response received. If the issue still exists with latest [release](http://www.lakka.tv/get) and/or [nightly](http://nightly.builds.lakka.tv), please feel free to re-open this issue. Thank you.

closed 2018-01-31 by ToKe79