libretro/Lakka-LibreELEC #79: Tp-link tl wn823n v2 not working

juri74, 2017-03-12:
i have that wifi dongle and it isn't recognized by Lakka
Gouchi made a test image that worked flawlessly
here the reference thread:
https://forums.libretro.com/t/tp-link-tl-wn823n-v2-support/9294/9

thanks

gouchi, 2017-03-26:
It has been enabled by this [commit](https://github.com/lakkatv/Lakka-LibreELEC/commit/9810d3a0e3f75f1c493f97b00f94c28b801d9e03). Should be available in rc4 release.

Please report if it is working.

Thank you.

Kind regards,

juri74, 2017-03-29:
thanks, i will test as soon as rc4 is released

gouchi, 2017-03-29:
It has been [released](http://www.lakka.tv/articles/2017/03/27/lakka-20-rc4-is-out).

juri74, 2017-03-31:
yes, my fault, i replied here BEFORE checking :D

Kivutar, 2017-04-02:
@juri74 any news on this?

juri74, 2017-04-03:
hello i think i have somehow damaged my odroid xu4, so i tested rc4 on a intel x64 machine with the same wi-fi dongle and it worked out of the box. so i imagine it work on odroid xu4 too

Kivutar, 2017-04-03:
OK, thanks!

closed 2017-04-03 by Kivutar