libretro/Lakka-LibreELEC #82: Doesn't compile for S905

shantigilbert, 2017-03-14:
Just a heads up when trying to compile from scratch you get this error:

APPLY PATCH (common)   /Lakka/projects/S905/packages/linux-drivers/qca9377-aml/patches/qca9377-aml-001-Modify-SDIO-and-firmware-handling-to-support-chips-u.patch
patching file CORE/HDD/src/wlan_hdd_cfg.c
patching file CORE/HDD/src/wlan_hdd_main.c
patching file CORE/SERVICES/BMI/ol_fw.c
patching file CORE/SERVICES/HIF/sdio/linux/if_ath_sdio.c
patching file CORE/SERVICES/HIF/sdio/linux/native_sdio/include/hif_internal.h
patching file CORE/SERVICES/HIF/sdio/linux/native_sdio/src/hif.c
Hunk #10 FAILED at 813.
Hunk #11 FAILED at 830.
Hunk #14 FAILED at 1424.
3 out of 18 hunks FAILED -- saving rejects to file CORE/SERVICES/HIF/sdio/linux/native_sdio/src/hif.c.rej
patching file Kbuild
Hunk #1 succeeded at 1070 (offset 2 lines).
Makefile:12: recipe for target 'image' failed
make: *** [image] Error 1

shantigilbert, 2017-03-24:
It seems it still doesn't compile for s905, but its now a different error. with a clean clone from github I get this error 

`
DISTRO=Lakka PROJECT=S905 ARCH=arm make image -j8

BUILD    zlib (host)
Executing (host): cmake -DCMAKE_TOOLCHAIN_FILE=/home/sg/sx05re/Lakka-LibreELEC/build.Lakka-S905.arm-8.0-devel/toolchain/etc/cmake-x86_64-linux-gnu.conf -DCMAKE_INSTALL_PREFIX=/home/sg/sx05re/Lakka-LibreELEC/build.Lakka-S905.arm-8.0-devel/toolchain -DCMAKE_BUILD_TYPE=Release /home/sg/sx05re/Lakka-LibreELEC/build.Lakka-S905.arm-8.0-devel/zlib-1.2.8
-- The C compiler identification is unknown
CMake Error at CMakeLists.txt:4 (project):
  The CMAKE_C_COMPILER:

    /home/Lakka/build.Lakka-S905.arm-8.0-devel/toolchain/bin/host-gcc

  is not a full path to an existing compiler tool.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also "/home/Lakka-LibreELEC/build.Lakka-S905.arm-8.0-devel/zlib-1.2.8/.x86_64-linux-gnu/CMakeFiles/CMakeOutput.log".
See also "/home/Lakka-LibreELEC/build.Lakka-S905.arm-8.0-devel/zlib-1.2.8/.x86_64-linux-gnu/CMakeFiles/CMakeError.log".
Makefile:12: recipe for target 'image' failed
make: *** [image] Error 1
`

Ntemis, 2017-03-24:
Weird I don't get any errors on clean build.

shantigilbert, 2017-03-24:
Don't know why I do, but if I use the package from Kzsaq's S905 build, and change it to use elf instead of gnu it compiles fine. 

if [ $TARGET_KERNEL_ARCH = "arm64" ] && [ $TARGET_ARCH == "arm" ]; then
PKG_DEPENDS_TARGET="$PKG_DEPENDS_TARGET gcc-linaro-aarch64-elf:host"
export PATH=$ROOT/$TOOLCHAIN/lib/gcc-linaro-aarch64-elf/bin/:$PATH
TARGET_PREFIX=aarch64-elf-
fi

I am using Ubuntu 16.04 if it helps.

Ntemis, 2017-03-24:
I can't accept issue, compiles fine on Ubuntu 16.04.2 here. Not Lakka related and am closing it.

closed 2017-03-24 by Ntemis