libretro/Lakka-LibreELEC #83: Orange Pi - Bluetooth pairings lost after reboot

hooby3dfx, 2017-03-16:
- Which version of Lakka are you using?
Lakka-H3.opilite.arm-8.0-devel-20170305054450-r25834-g76ebcbd
- What system hardware are you using?
Orange Pi Lite. Also reproducible on Orange Pi PC.
- What did you do?
Pair, Connect, and Trust bluetooth gamepad.
Controller works fine for that session.
After restarting the system the pairing is lost, and the controller will not connect.
```
[bluetooth]# info F0:65:DD:BF:48:2A
Device F0:65:DD:BF:48:2A
        Name: Razer Serval
        Alias: Razer Serval
        Class: 0x002508
        Icon: input-gaming
        Paired: no
        Trusted: yes
        Blocked: no
        Connected: no
        LegacyPairing: no
        UUID: Human Interface Device... (00001124-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v1532p0900d3A16
```

- What did you expect to happen?
The pairing should have been retained across reboots.
This functionality is working on the OpenELEC fork on other hardware (ODROID).

hooby3dfx, 2017-04-01:
@Kivutar this is still reproducible in RC4 and also on Raspberry Pi (reported by @travisghansen ).
Is there any other info I can provide?
Seems like a major issue.

Kivutar, 2017-04-02:
I just tested on RPi3 RC4 with an official DualShock 3. It autopairs fine then is able to reconnect after a reboot. I'm going to test on OPiPC now.

Ntemis, 2017-04-03:
Looks like Orange Pi Lite doesn't have onboard bluetooth @hooby3dfx

hooby3dfx, 2017-04-03:
Yes, I have been using a USB bluetooth adapter... (Which works fine on
odroid / stable Lakka)

Ntemis, 2017-04-03:
So issue is fixed?

hooby3dfx, 2017-04-03:
No, the problem is that in RC4 after reboot all pairings are lost.

Ntemis, 2017-04-03:
you face this issue only with orange pi soc?

Kivutar, 2017-04-03:
He also has this error on RPi.

hooby3dfx, 2017-04-05:
@travisghansen are you able to consistently reproduce this on your RPi3 & Ouya controller?

I have been trying to figure out exactly when this happens and it seems inconsistent.

It does seem to consistently occur with the Ouya controller though. I also have a pretty high quality Razer controller that after pairing tonight seems to be retaining its pairing after many reboots and power cycles.

travisghansen, 2017-04-05:
Out for a few days but will review when I get back early next week.

Kivutar, 2017-04-06:
I could reproduce the problem on RPi3 with my 8Bitdo NES30 Pro.

After receiving these two messages, I was able to reconnect using bluetoothctl:

```
[CHG] Device 69:01:3C:D3:67:00 LegacyPairing: yes
[CHG] Device 69:01:3C:D3:67:00 RSSI: -56
```

hooby3dfx, 2017-04-07:
My Razer controller pairing remains resilient.

I have another controller though ("GEN GAME S3") that is doing the same thing as the Ouya. 

One other thing to note is that for both of these controllers, not just the reboot results in not being able to reconnect - even if you just disconnect it from bluetoothctl, it will not successfully reconnect.

hooby3dfx, 2017-04-07:
Breakthrough/face-palm!
I reviewed the instructions here http://www.lakka.tv/doc/8Bitdo-Wireless-Controller/
and found that the pairings seem to stick much better when these commands are run pre-pairing (and possibly every boot):
```
agent on
default-agent
discoverable on
pairable on
```

I was only doing:
```
scan on
pair
connect
trust
```

Will update once confirmed.

R3d-Sky, 2017-04-07:
I followed the whole set of instructions, pairing still does not stick with my NVidia Shield (2017) Controller and RPi3. Pairing on the controller does stick with my Shield Android TV when paired to it, though.

Kivutar, 2017-04-07:
Same here, I was doing the agent stuff before paring, and I still have some controllers that don't reconnect.

hooby3dfx, 2017-04-09:
I have been messing around with this quite a bit and found that
1. the timing which pairing and connecting is done in for the first time seems to matter.
2. sometimes you will actually have a good pairing that sticks, but the controller will fail to reconnect on the first or second try - but then will connect.

There is some black magic. I think it would be good to look at the files that bluez is using internally to store pairings, or watch its debug logs to get a deeper understanding.

However, I have been able to now automate the pairing process pretty well in a script that results in sticky pairings on all of my controllers (Razer, Ouya, and Gen Game).
https://github.com/hooby3dfx/lakka-addons/blob/master/btautopair.sh

Kivutar, 2017-04-09:
Interesting script. I was thinking that it was impossible to automate pairing because of the agent stuff. I was not aware of *empty*. I need to study this and integrate it in RA.

hooby3dfx, 2017-04-09:
Great - it would be awesome to integrate via a GUI option so that you can choose "Pair Bluetooth controller" and then the GUI waits until one controller pairs, or the script times out (ex 30 seconds).

luksireiku, 2017-04-12:
I have the same bug with Raspberry Pi 3 and 8Bitdo Zero controller, I get some error messages like `Can't get HIDP connection info` with `systemctl status bluetooth`, I can post the exact messages when I go home later.

Kivutar, 2017-10-21:
We're working on a new Allwinner unified image, with mainline kernel. It may fix this issue if we success.

closed 2020-08-18 by Ntemis