libretro/Lakka-LibreELEC #85: Problems navigating menus after starting a ROM in Lakka 2.0 RC3 on Raspberry Pi 3

ronfar623, 2017-03-20:
- Which version of Lakka are you using?
      Lakka 2.0 RC3.  Problem occurred under RC2 as well.
- What system hardware are you using?
      Raspberry Pi 3
- What did you do?
      Connected an Xbox One controller over USB using the latest firmware as of 3/19/2017.
- What happened?
      Navigating Retroarch menus works as expected until a ROM/core is started.  Regardless of core, once the "Xbox" button is pressed to return to the Retroarch menus, you can no longer select any menu items with the B button.  You can only go back/up menu levels by pressing A.  If the controller is disconnected and a regular Xbox 360 pad connected instead, all inputs work as expected.  If the Xbox 360 pad is disconnected and the Xbox One pad is then reconnected again, the problem returns in the same way.

Kivutar, 2017-03-20:
Is it an XBox One S controller? I have one and I also get this issue. I tried to patch the kernel to support it but it didn't work well yet.

ronfar623, 2017-03-20:
I discovered that I actually have one of each, and in testing found a workaround of sorts.

On the original Xbox One pad, pressing the "Xbox" button does nothing, but the default L3+R3 return to menu shortcut works fine, and the menus work as expected.

I then plugged the Xbox One S pad in, and discovered that using the L3+R3 shortcut works there as well, with menus functioning as expected.

It seems that pressing the "Xbox" button on the Xbox One S pad specifically causes a menu glitch, but everything works fine if you just don't ever press it.  If you guys can't work around it due to kernel issues or whatever, it might be worth changing the default mapping for this controller to just disable the "Xbox" button.  Thanks!

Kivutar, 2017-04-04:
I just pushed a fix for this. Can you give a try to this image please? http://static.kivutar.me/Lakka-RPi2.arm-2.0.img.gz

Please note that this image is called 2.0 but is not the definitive 2.0 yet.

Kivutar, 2017-05-31:
No replies and I can't reproduce. I close the issue.

closed 2017-05-31 by Kivutar