libretro/Lakka-LibreELEC #86: Lakka 2.0 RC3 not starting on Intel NUC DN2820FYKH

aboutafter, 2017-03-26:
Lakka 2.0 RC3 doesn't start on the Intel NUC DN2820FYKH when booting from the USB drive.

Selecting the USB drive where Lakka is to boot, it simply jumps straight to the PC's grub menu because it doesn't seem to recognize any booting partitions on it.

This same USB drive works correctly on my desktop PC (Intel i5-4670K and Nvidia GTX 770) and on my Thinkpad x230 laptop (Intel i7-3520M).

gouchi, 2017-03-26:
Please try to provide some Lakka system and graphics [logs](http://www.lakka.tv/doc/Troubleshooting-Lakka).

Thank you.

Kind regards,

aboutafter, 2017-03-27:
I don't have any logs because when I try to boot from the USB drive it simply ignores it and it goes straight to grub (the one on the SSD of the PC). 

This doesn't happen with the stable version of Lakka, so I guess it has something to do with LibreELEC.

Kivutar, 2017-03-27:
It's because of the EFI boot of LibreELEC. You need to boot your live USB, plug an ethernet cable, and connect using ssh. You need to add ssh in the cmdline in the bootloader config file in your first partition in your USB.

aboutafter, 2017-03-28:
But I can't boot my live USB. That's the issue.

gouchi, 2017-03-30:
Did you try to boot your live USB with legacy boot enabled ?

aboutafter, 2017-03-31:
I'm using legacy boot on all PC's. It's only failing to boot on the NUC. The stable Lakka version (the OpenELEC one) boots fine.

Kivutar, 2017-03-31:
Can you check if this NUC supports booting of GPT partitions?

If not, the workaround would be to setup the stable Lakka-OE, then use the manual upgrade to Lakka-LE by droping the .tar in the .update folder and rebooting.

thimslugga, 2017-05-07:
@Kivutar Which tar are you referring to? I'm unable to find this as I'm attempting a manual upgrade from the old stable based on OE.

Kivutar, 2017-05-07:
This one http://le.builds.lakka.tv/Generic.x86_64/Lakka-Generic.x86_64-2.0.tar

gouchi, 2017-06-03:
Is it possible to make a test with[ Lakka 2.1 rc1](http://le.builds.lakka.tv/Generic.x86_64/Lakka-Generic.x86_64-2.1-rc1.img.gz) ? 

Thank you.

gouchi, 2017-08-06:
Do your reproduce the issue with [Lakka 2.1 rc4](http://le.builds.lakka.tv/Generic.x86_64/Lakka-Generic.x86_64-2.1-rc4.img.gz) ?

aboutafter, 2017-08-09:
It still doesn't boot, no.

Kivutar, 2017-08-09:
Try to setup this old release http://mirror.lakka.tv/nightly/Generic.x86_64/Lakka-Generic.x86_64-devel-20161221183017-r21729-g274b40f.img.gz

And then do a manual upgrade to 2.1-rc4 using the tar http://le.builds.lakka.tv/Generic.x86_64/Lakka-Generic.x86_64-2.1-rc4.tar

aboutafter, 2017-08-09:
Yes. With an older version and manually updating it does work.

gouchi, 2017-08-09:
It seems there was [issue with usb/uefi boot](https://communities.intel.com/thread/54407) with your device.

Kivutar, 2017-08-29:
Yes, you should also try to downgrade your BIOS like the guy in this thread did.

Assault, 2017-09-21:
I had the same Problem. Updating to the latest Bios Version helped.

https://www.intel.de/content/www/de/de/support/boards-and-kits/000005636.html

https://downloadcenter.intel.com/de/download/26898/NUCs-BIOS-Update-FYBYT10H-86A-?product=78953

aboutafter, 2017-09-21:
Not for me. I tried downgrading the BIOS and with the latest version too. It simply doesn't want to boot from the USB.

Assault, 2017-09-22:
And have you tried disabling Secure boot and than start in uefi mode ?
Press F12 on boot to directly choose the uefi boot method

aboutafter, 2017-09-23:
Huh. It turns out that if I turn on the UEFI option on the BIOS an option to boot USB with UEFI appears, and in that mode I can boot Lakka correctly. 

So, why do I need to select the boot with UEFI option on the NUC but on the rest of PCs on my house Lakka boots without an issue without that option?

Kivutar, 2017-10-11:
To be honest, I don't know. The way our syslinux is configured looks like an attempt to hybrid UEFI and BIOS boots. This is certainly why we have two bootloader config files in the boot partition. One for UEFI, one for BIOS.

With the update to OpenELEC 6 (and LibreELEC) one year ago, the number of old PCs able to run Lakka has been reduced.

And I think that the number of new PCs able to run Lakka has increased.

There was also a move from MBR to GPT at that time, that was directly linked to the UEFI stuff.

closed 2018-01-31 by ToKe79