libretro/Lakka-LibreELEC #88: RPi3/RC4 hatari core not working

killer101, 2017-03-28:
Hello,

I just gave the RC4 a little testing. The now available hatari core won't work for me on the RPi3. It is just doing fine using Lakka on the PC (x64).

Here's the log ...

RetroArch [INFO] :: This is RetroArch version 1.5.0 (Git 17235d7)
RetroArch [INFO] :: === Build =======================================
Capabilities: NEON VFPv3 VFPv4 
Built: Mar 23 2017
RetroArch [INFO] :: Version: 1.5.0
RetroArch [INFO] :: Git: 17235d7
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Loading default config.
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Redirecting save file to "/storage/savefiles/.srm".
RetroArch [INFO] :: Redirecting savestate to "/storage/savestates/.state".
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29970.03 Hz.
RetroArch [INFO] :: Found shader "/tmp/shaders/bilinear.glslp"
RetroArch [INFO] :: Found shader "/tmp/shaders/nearest.glslp"
RetroArch [INFO] :: Found shader "/tmp/shaders/stock.glsl"
RetroArch [INFO] :: Video @ fullscreen
RetroArch [INFO] :: Starting threaded video driver ...
RetroArch [INFO] :: [EGL]: EGL version: 1.4
RetroArch [INFO] :: [EGL]: Current context: 0x1.
RetroArch [INFO] :: Found GL context: videocore
RetroArch [INFO] :: Detecting screen resolution 1920x1080.
RetroArch [INFO] :: [EGL]: eglSwapInterval(1)
RetroArch [INFO] :: [EGL]: eglSwapInterval(1)
RetroArch [INFO] :: [GL]: Vendor: Broadcom, Renderer: VideoCore IV HW.
RetroArch [INFO] :: [GL]: Version: OpenGL ES 2.0.
RetroArch [INFO] :: GL: Using resolution 1920x1080
RetroArch [INFO] :: [GL]: Default shader backend found: glsl.
RetroArch [INFO] :: [Shader driver]: Using GLSL shader backend.
RetroArch [WARN] :: [GL]: Stock GLSL shaders will be used.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
[... 97 lines cut ...]
[libretro-test]: Got size: 640 x 480.
RetroArch [INFO] :: Content loading skipped. Implementation will load it on its own.
RetroArch [INFO] :: Environ SET_INPUT_DESCRIPTORS:
RetroArch [INFO] :: RetroPad, User 1, Button "Y (left)" => "Enter GUI"
RetroArch [INFO] :: RetroPad, User 1, Button "Select" => "Mouse mode toggle"
RetroArch [INFO] :: RetroPad, User 1, Button "Start" => "Keyboard overlay"
RetroArch [INFO] :: RetroPad, User 1, Button "D-Pad Up" => "Up"
RetroArch [INFO] :: RetroPad, User 1, Button "D-Pad Down" => "Down"
RetroArch [INFO] :: RetroPad, User 1, Button "D-Pad Left" => "Left"
RetroArch [INFO] :: RetroPad, User 1, Button "D-Pad Right" => "Right"
RetroArch [INFO] :: RetroPad, User 1, Button "A (right)" => "Fire"
RetroArch [INFO] :: RetroPad, User 1, Button "L" => "Joystick number"
RetroArch [INFO] :: RetroPad, User 1, Button "R" => "Mouse speed"
RetroArch [INFO] :: RetroPad, User 1, Button "L2" => "Toggle m/k status"
 0  hatari
 1  --disk-a
 2  /storage/roms/stick/atarist/International Karate + (1988)(System 3).stx
RetroConf:'/storage/system/hatari.cfg'
Hatari v1.8.0, compiled on:  Mar 26 2017, 03:47:09
Configured max Hatari resolution = 640x480, optimal for ST = 832x588
STX : STX_Insert_internal drive=0 file=/storage/roms/stick/atarist/International Karate + (1988)(System 3).stx buf=0x72027008 size=888272
Inserted disk '/storage/roms/stick/atarist/International Karate + (1988)(System 3).stx' to drive A:.
Building CPU table for configuration: 68000 (compatible mode)
NVRAM not found at '/storage/.hatari/hatari.nvram'
Bus error (core dumped)

Kivutar, 2017-03-28:
Thanks for the report. It may be due to '/storage/.hatari/hatari.nvram'. Can you try to create an empty file here?

killer101, 2017-03-28:
Nope, doesn't change anything. Except making the nvram error going away!

Here's the log ...

RetroArch [INFO] :: This is RetroArch version 1.5.0 (Git 17235d7)
RetroArch [INFO] :: === Build =======================================
Capabilities: NEON VFPv3 VFPv4 
Built: Mar 23 2017
RetroArch [INFO] :: Version: 1.5.0
RetroArch [INFO] :: Git: 17235d7
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Loading default config.
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Redirecting save file to "/storage/savefiles/.srm".
RetroArch [INFO] :: Redirecting savestate to "/storage/savestates/.state".
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29970.03 Hz.
RetroArch [INFO] :: Found shader "/tmp/shaders/bilinear.glslp"
RetroArch [INFO] :: Found shader "/tmp/shaders/nearest.glslp"
RetroArch [INFO] :: Found shader "/tmp/shaders/stock.glsl"
RetroArch [INFO] :: Video @ fullscreen
RetroArch [INFO] :: Starting threaded video driver ...
RetroArch [INFO] :: [EGL]: EGL version: 1.4
RetroArch [INFO] :: [EGL]: Current context: 0x1.
RetroArch [INFO] :: Found GL context: videocore
RetroArch [INFO] :: Detecting screen resolution 1920x1080.
RetroArch [INFO] :: [EGL]: eglSwapInterval(1)
RetroArch [INFO] :: [EGL]: eglSwapInterval(1)
RetroArch [INFO] :: [GL]: Vendor: Broadcom, Renderer: VideoCore IV HW.
RetroArch [INFO] :: [GL]: Version: OpenGL ES 2.0.
RetroArch [INFO] :: GL: Using resolution 1920x1080
RetroArch [INFO] :: [GL]: Default shader backend found: glsl.
RetroArch [INFO] :: [Shader driver]: Using GLSL shader backend.
RetroArch [WARN] :: [GL]: Stock GLSL shaders will be used.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
[... 94 lines cut ...]
RetroArch [INFO] :: 640x480
[libretro-test]: Got size: 640 x 480.
RetroArch [INFO] :: Content loading skipped. Implementation will load it on its own.
RetroArch [INFO] :: Environ SET_INPUT_DESCRIPTORS:
RetroArch [INFO] :: RetroPad, User 1, Button "Y (left)" => "Enter GUI"
RetroArch [INFO] :: RetroPad, User 1, Button "Select" => "Mouse mode toggle"
RetroArch [INFO] :: RetroPad, User 1, Button "Start" => "Keyboard overlay"
RetroArch [INFO] :: RetroPad, User 1, Button "D-Pad Up" => "Up"
RetroArch [INFO] :: RetroPad, User 1, Button "D-Pad Down" => "Down"
RetroArch [INFO] :: RetroPad, User 1, Button "D-Pad Left" => "Left"
RetroArch [INFO] :: RetroPad, User 1, Button "D-Pad Right" => "Right"
RetroArch [INFO] :: RetroPad, User 1, Button "A (right)" => "Fire"
RetroArch [INFO] :: RetroPad, User 1, Button "L" => "Joystick number"
RetroArch [INFO] :: RetroPad, User 1, Button "R" => "Mouse speed"
RetroArch [INFO] :: RetroPad, User 1, Button "L2" => "Toggle m/k status"
 0  hatari
 1  --disk-a
 2  /storage/roms/stick/atarist/International Karate + (1988)(System 3).stx
RetroConf:'/storage/system/hatari.cfg'
Hatari v1.8.0, compiled on:  Mar 26 2017, 03:47:09
Configured max Hatari resolution = 640x480, optimal for ST = 832x588
STX : STX_Insert_internal drive=0 file=/storage/roms/stick/atarist/International Karate + (1988)(System 3).stx buf=0x74322008 size=888272
Inserted disk '/storage/roms/stick/atarist/International Karate + (1988)(System 3).stx' to drive A:.
Building CPU table for configuration: 68000 (compatible mode)
Bus error (core dumped)

r-type, 2017-04-02:
i thinks it's libco related ,if rpi3 then arm so you should add an -marm in cflags if not the case  .

Kivutar, 2017-04-02:
Thanks, I try it now.

closed 2017-04-03 by Kivutar

Kivutar, 2017-04-03:
@r-type is there something we can do about the location of the tos.img?

commit 3964cef0 (2017-04-03): Fix hatari on ARM. Fixes #88