libretro/Lakka-LibreELEC #89: RPi3/RC4 puae core not working

killer101, 2017-03-28:
Hello again,

More RC4 testing. The puae core won't work for me on the RPi3. It is just doing fine using Lakka on the PC (x64).

Here's the log ...

RetroArch [INFO] :: This is RetroArch version 1.5.0 (Git 17235d7)
RetroArch [INFO] :: === Build =======================================
Capabilities: NEON VFPv3 VFPv4 
Built: Mar 23 2017
RetroArch [INFO] :: Version: 1.5.0
RetroArch [INFO] :: Git: 17235d7
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Loading default config.
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Redirecting save file to "/storage/savefiles/.srm".
RetroArch [INFO] :: Redirecting savestate to "/storage/savestates/.state".
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29970.03 Hz.
RetroArch [INFO] :: Found shader "/tmp/shaders/bilinear.glslp"
RetroArch [INFO] :: Found shader "/tmp/shaders/nearest.glslp"
RetroArch [INFO] :: Found shader "/tmp/shaders/stock.glsl"
RetroArch [INFO] :: Video @ fullscreen
RetroArch [INFO] :: Starting threaded video driver ...
RetroArch [INFO] :: [EGL]: EGL version: 1.4
RetroArch [INFO] :: [EGL]: Current context: 0x1.
RetroArch [INFO] :: Found GL context: videocore
RetroArch [INFO] :: Detecting screen resolution 1920x1080.
RetroArch [INFO] :: [EGL]: eglSwapInterval(1)
RetroArch [INFO] :: [EGL]: eglSwapInterval(1)
RetroArch [INFO] :: [GL]: Vendor: Broadcom, Renderer: VideoCore IV HW.
RetroArch [INFO] :: [GL]: Version: OpenGL ES 2.0.
RetroArch [INFO] :: GL: Using resolution 1920x1080
RetroArch [INFO] :: [GL]: Default shader backend found: glsl.
RetroArch [INFO] :: [Shader driver]: Using GLSL shader backend.
RetroArch [WARN] :: [GL]: Stock GLSL shaders will be used.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
[... 175 lines cut ...]
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_music_history.lpl].
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_video_history.lpl].
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_image_history.lpl].
------------------------------------------------------------------------------------
P-UAE 2.6.1
Git Commit: 0186c1b16f7181ffa02d73e6920d3180ce457c46
Build date: Mar 16 2017 05:56:35
------------------------------------------------------------------------------------
Copyright 1995-2002 Bernd Schmidt
          1999-2013 Toni Wilen
          2003-2007 Richard Drummond
          2006-2013 Mustafa 'GnoStiC' Tufan

See the source for a full list of contributors.
This is free software; see the file COPYING for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
------------------------------------------------------------------------------------
Enumerating display devices.. 
Sorting devices and modes..
done
load config '/storage/roms/stick/amiga/config/AAARGH.uae':0
Opening cfgfile '/storage/roms/stick/amiga/config/AAARGH.uae': OK
RTAREA located at 00F00000
*** Error in `retroarch': malloc(): memory corruption (fast): 0x01de67b8 ***
setup_soundAborted (core dumped)

Kivutar, 2017-03-28:
The backtrace:

```
*** Error in `/usr/bin/retroarch': malloc(): memory corruption (fast): 0x01a7d7b0 ***
setup_sound
Thread 1 "retroarch" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:58
58	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:58
#1  0x75914400 in __GI_abort () at abort.c:89
#2  0x7594c890 in __libc_message (do_abort=do_abort@entry=2, fmt=<optimized out>) at ../sysdeps/posix/libc_fatal.c:175
#3  0x759532f0 in malloc_printerr (action=<optimized out>, str=0x759feda4 "malloc(): memory corruption (fast)", ptr=<optimized out>, ar_ptr=<optimized out>) at malloc.c:5046
#4  0x75955a28 in _int_malloc (av=av@entry=0x75a1b7a4 <main_arena>, bytes=bytes@entry=13) at malloc.c:3421
#5  0x75957868 in __GI___libc_malloc (bytes=13) at malloc.c:2925
#6  0x7595c0a4 in __GI___strdup (s=0x71ac82ac "uae.resource") at strdup.c:42
#7  0x7160da10 in ds () from /storage/cores/puae_libretro.so
#8  0x71679838 in uaeres_install () from /storage/cores/puae_libretro.so
#9  0x715ab9c0 in virtualdevice_init () from /storage/cores/puae_libretro.so
#10 0x715c945c in memory_init () from /storage/cores/puae_libretro.so
#11 0x715ca9a8 in memory_reset () from /storage/cores/puae_libretro.so
#12 0x715abd28 in real_main () from /storage/cores/puae_libretro.so
#13 0x715ac514 in umain () from /storage/cores/puae_libretro.so
#14 0x71aad39c in retro_wrap_emulator () from /storage/cores/puae_libretro.so
#15 0x71aadc8c in retro_run () from /storage/cores/puae_libretro.so
#16 0x00000000 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

killer101, 2017-03-29:
And this means ...?

Kivutar, 2017-04-03:
@twinaphex any idea on this one? It only happens on ARM. I've not forgot to pass DARM -marm

r-type, 2017-04-04:
seem related to the uae.cfg file 
at some point it return true to uaecptr base = need_uae_boot_rom ();
because the log show RTAREA located at 00F00000
``
void rtarea_setup (void)
{
	uaecptr base = need_uae_boot_rom ();
	if (base) {
		write_log (_T("RTAREA located at %08X\n"), base);
		rtarea_base = base;
	}
}
``
can you post the uae cfg used ?
seem it failed if return other than 0 here
``
static uaecptr check_boot_rom (void)
{
	uaecptr b = RTAREA_DEFAULT;
	addrbank *ab;

	if (currprefs.cs_cdtvcd || currprefs.cs_cdtvscsi || currprefs.uae_hide > 1)
		b = RTAREA_BACKUP;
	if (currprefs.cs_mbdmac == 1)
		b = RTAREA_BACKUP;
	ab = &get_mem_bank (RTAREA_DEFAULT);
	if (ab) {
		if (valid_address (RTAREA_DEFAULT, 65536))
			b = RTAREA_BACKUP;
	}
	if (nr_directory_units (NULL))
		return b;
	if (nr_directory_units (&currprefs))
		return b;
	if (currprefs.socket_emu)
		return b;
	if (currprefs.uaeserial)
		return b;
	if (currprefs.scsi == 1)
		return b;
	if (currprefs.sana2)
		return b;
	if (currprefs.input_tablet > 0)
		return b;
	if (currprefs.rtgmem_size)
		return b;
	if (currprefs.chipmem_size > 2 * 1024 * 1024)
		return b;
	if (currprefs.z3chipmem_size)
		return b;
	return 0;
}
``
with a flat basic cfg like the rickdangerous.uae provide it works for me and no show up  RTAREA location  .
maybe usage of rtgmem or z3chipmem cause the issue. not sure this old implementation of puae support it either.

Kivutar, 2017-04-04:
Oh! Here is the .uae used:

```
kickstart_rom_file=/storage/roms/amiga/Kickstart v1.3 rev 34.5 (1987)(Commodore)(A500-A1000-A2000-CDTV)[!].rom
chipmem_size=1
bogomem_size=2
use_gui=no
nr_floppies=1
cpu_type=68000
cpu_speed=real
cpu_compatible=true
ntsc=true
chipset=ocs
immediate_blits=false
gfx_linemode=double
gfx_framerate=1
sound_output=normal
sound_frequency=44100
sound_channels=mixed
sound_interpol=none
show_leds=true
floppy_speed=800
gfx_center_vertical=smart
gfx_center_horizontal=smart
gfx_color_mode=16
floppy0=/storage/roms/amiga/James Pond 2 - Codename Robocod (1991)(Millennium)[cr SKR][a2].adf
```

killer101, 2017-04-04:
Here's the one I just tested, works fine on the x86/x64 build of RC4 ...

use_gui=no
show_leds=false
chipset=aga
cpu_type=68ec020
chipmem_size=4
fastmem_size=4
gfx_center_vertical=simple
gfx_center_horizontal=simple
#gfx_width=640
#gfx_height=420
sound_channels=mixed
kickstart_rom_file=/storage/system/kick31.rom
hardfile=rw,32,1,2,512,/storage/roms/stick/amiga/hdf/WHDLoad.hdf
hardfile=rw,32,1,2,512,/storage/roms/stick/amiga/hdf/International Karate Plus.hdf

r-type, 2017-04-04:
yes seem cfg is not the pb here !
I was thinking about that seeing  RTAREA located at 00F00000 in the above log
but it's explained by chipmem>2M

looking at your BT , it seem related to  uaeres_install () 

```
#ifdef FILESYS
	rtarea_init ();
	uaeres_install ();
	hardfile_install ();
#endif
```
and failed at ds ()
```
void uaeres_install (void)
{
	uae_u32 functable, datatable;
	uae_u32 initcode, getfunc;
	TCHAR tmp[100];

	_stprintf (tmp, _T("UAE resource %d.%d.%d"), UAEMAJOR, UAEMINOR, UAESUBREV);
	res_name = ds (_T("uae.resource"));
	res_id = ds (tmp);
```
not sure why!

on my own pi3 build (not running lakka) it load & work fine at least for adf .

Kivutar, 2017-04-05:
This is a part of my build log, maybe you can notice something wrong in my CFLAGS:

```
/home/kivutar/lakkadisk/kivutar/Lakka-LibreELEC/build.Lakka-RPi2.arm-2.0/toolchain/bin/armv7ve-libreelec-linux-gnueabi-gcc -fPIC -march=armv7ve -mabi=aapcs-linux -Wno-psabi -Wa,-
mno-warn-deprecated -mtune=cortex-a7 -mfloat-abi=hard -mfpu=neon-vfpv4 -fomit-frame-pointer -Wall -pipe -O2  -DARM -marm -O2 -DCPUEMU_0 -DCPUEMU_11 -DCPUEMU_12 -DCPUEMU_20 -DCPUE
MU_21 -DCPUEMU_22  -DFPUEMU -DUNALIGNED_PROFITABLE -DAMAX -DAGA -DAUTOCONFIG -DFILESYS -DSUPPORT_THREADS  -DFDI2RAW -DDEBUGGER -DSAVESTATE -DENFORCER -DACTION_REPLAY -DXARCADE -D
CPUEMU_31 -DCPUEMU_32  -DCPUEMU_33 -DMMUEMU -DFULLMMU -D__LIBRETRO__ -DRETRO=1 -DINLINE="inline" -std=gnu99    -I./sources/src -I./sources/src/include -I./libretro -I./libretro/l
ibco -I./sources/utils -c -o libretro/diskutils.o libretro/diskutils.c
```

r-type, 2017-04-05:
seems fine for me .
could you provide the core so i can test it on my pi3 without lakka to see if it also crash .
the only diff i see is gcc tume for march=armv7ve -mabi=aapcs-linux -Wno-psabi -Wa,-
mno-warn-deprecated -mtune=cortex-a7 -mfloat-abi=hard -mfpu=neon-vfpv4

r-type, 2017-04-05:
[puae_libretro.so.zip](https://github.com/lakkatv/Lakka-LibreELEC/files/900193/puae_libretro.so.zip)
can you test also mine to see if it crash too for you

killer101, 2017-04-06:
I've tried it. Works!

What I noticed ...

- your compiled core is more then a meg bigger than lakkas?! Don't know if this means anything.
- with my config from above, this thing is real slow. It is getting better using a standard config (A500), but still slowdowns here and there (IK+).
- don't know if speed is a RPi3 thing or a problem of this "foreign" core
- if it is an RPi3 thing, for me it makes no really sense emulating the amiga on the RPi3

As I read in the manual, the buildin dynamic recomiler works only x86/x64 machines. Maybe thats a problem!

r-type, 2017-04-06:
size increase is only because I haven't strip the core.

Yes dynamic rec is not available here for arm ,  for this you should move to uae4arm
I've working in the past on this core  https://github.com/r-type/uae4arm-libretro but it's a bit outdated
and I'm not sure where/how I leave it since I'm on PI very occasionally.

I know that @repojohnray working on a new fork and it's available here https://github.com/repojohnray/libretro-uae4arm but I'm not sure about the status of his core , you should ask him for detail.

killer101, 2017-04-06:
As I'm not related to Lakka in any way, I think I'm not the one to decide anything here!  ;-)

As RC4 saw the light of day, I was just setting up my DIY retro gaming machine. I wanted to test some cores and I decided to share my impressions to help to improve the whole thing! I'm just a user!

Kivutar, 2017-04-10:
@killer101 I generated a test image with an updated PUAE and also uae4arm. http://static.kivutar.me/Lakka-RPi2.arm-2.0.img.gz Can you test it please and tell me how both core behaves?

killer101, 2017-04-10:
@Kivutar 

Here are the test results:

- Amiga playlist symbols are gone (C64 symbols are there now!)
- Both cores are working
- I just tested with my config files (A1200) for the WHDLoad versions of the games.

PUAE. As expected nothing changed when it comes to speed. Getting around 35 FPS for IK+. For me still no option for the RaspPi.

UAE4ARM. Not bad so far. Running full 50 FPS for IK+ and R-Type. The only two games I have tested so far. When it comes to speed, this is the core to go.

Bad news are ... No support for an on-screen keyboard or the GUI, or I don't know how to get there. 
Maybe there is still some work to do on this core. 

Core Options:

Autorun - off (default) ... don't know what this is for!

Internal resolution - 320x240 (default) ... the default should be 640x256, the only setting which is scaled correctly to fit the screen. Any other setting works, but the scaling is fu**ed up!

Model - 6128 (default) ... don't know what this is!

Wrap Factor - 1 (default) ... don't know what this is, maybe some floppy speeder?!

**Bundling the UAE4ARM core is a good idea I think, even there is still some work to do!**

Kivutar, 2017-04-11:
OK. I'm going to close this bug report then. The issues of UAE4ARM should be filled in this tracker https://github.com/libretro/uae4arm-libretro/issues

Thanks for the test!

closed 2017-04-11 by Kivutar

r-type, 2017-04-11:
lol!
for uae4arm , as i said previously ,  it's a bit outdated and I'm not sure where/how I leave it .
and as it's a fork of my repo (which is a fork from chips-fr) , better to fill up an issue on my repo if you hope a will look at .

Et voila.

killer101, 2017-04-11:
@r-type 
I think opening new issues are not enabled at your repo.

r-type, 2017-04-11:
you are right !
issues are enable now

killer101, 2017-04-12:
@Kivutar 

Any idea why the Amiga playlist symbols are gone now (Playlist name is still "Commodore Amiga")?

By the way. The Hatari core is still not working in this special test build of Lakka. Don't know if the compiler flag changes already included?!

Kivutar, 2017-04-12:
I see that the icon is named "Commodore - Amiga" now.