libretro/Lakka-LibreELEC #90: x360 wireless Joystick doesnt work after update RC4

rick-pit, 2017-03-28:
- Which version of Lakka are you using?
Lakka 2.0 RC4

- What system hardware are you using?
Orange Pi PC

- What did you do?
I Just update

- What did you expect to happen?
Joystick was configured in Lakka, can handle menu and games

- What happened instead?
It doesnt show load message when is detected in botton left border. Anyway joystick green light work showing "player one" status and after that it blink 2 times. i use a keyboard to move tu input menu but its doesnt listed there. PS3 controller works. I think wireless receiver driver is broken or lost

rick-pit, 2017-03-28:
This is the terminal log for USB

Lakka:~ # lsusb
Bus 008 Device 003: ID 045e:0719 Microsoft Corp. Xbox 360 Wireless Adapter
Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 007 Device 002: ID 04d9:1702 Holtek Semiconductor, Inc. Keyboard LKS02
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Lakka:~ # dmesg
[    0.000000] Booting Linux on physical CPU 0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Linux version 3.4.113 (kivutar@Server01) (gcc version 6.2.0 (GCC) ) #1 SMP PREEMPT Sat Mar 25 05:39:34 CDT 2017
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine: sun8i
[    0.000000] cma: CMA: reserved 256 MiB at 50000000
[    0.000000] Memory policy: ECC disabled, Data cache writealloc
[    0.000000] On node 0 totalpages: 262144
[    0.000000] free_area_init_node: node 0, pgdat c0c1b0c0, node_mem_map c0d02000
[    0.000000]   Normal zone: 1710 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 192850 pages, LIFO batch:31
[    0.000000]   HighMem zone: 594 pages used for memmap
[    0.000000]   HighMem zone: 66990 pages, LIFO batch:15
[    0.000000] script_init enter!
[    0.000000] script_init exit!
[    0.000000] PERCPU: Embedded 8 pages/cpu @c1618000 s11584 r8192 d12992 u32768
[    0.000000] pcpu-alloc: s11584 r8192 d12992 u32768 alloc=8*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 259840
[    0.000000] Kernel command line: console=ttyS0,115200 boot=/dev/mmcblk0p1 disk=/dev/mmcblk0p2 consoleblank=0
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] allocated 2097152 bytes of page_cgroup
[    0.000000] please try 'cgroup_disable=memory' option if you don't want memory cgroups
[    0.000000] Memory: 1024MB = 1024MB total
[    0.000000] Memory: 760068k/760068k available, 288508k reserved, 270336K highmem
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
[    0.000000]     vmalloc : 0xf0000000 - 0xff000000   ( 240 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xef800000   ( 760 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0008000 - 0xc0794914   (7731 kB)
[    0.000000]       .init : 0xc0795000 - 0xc0baad40   (4184 kB)
[    0.000000]       .data : 0xc0bac000 - 0xc0c1f8b0   ( 463 kB)
[    0.000000]        .bss : 0xc0c20064 - 0xc0d0121c   ( 901 kB)
[    0.000000] SLUB: Genslabs=11, HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000]  Additional per-CPU info printed with stalls.
[    0.000000] NR_IRQS:544
[    0.000000] Architected local timer running at 24.00MHz.
[    0.000000] Switching to timer-based delay loop
[    0.000000] sched_clock: 32 bits at 24MHz, resolution 41ns, wraps every 178956ms
[    0.000000] Console: colour dummy device 80x30
[    0.000117] Calibrating delay loop (skipped), value calculated using timer frequency.. 4800.00 BogoMIPS (lpj=24000000)
[    0.000141] pid_max: default: 32768 minimum: 301
[    0.000538] Mount-cache hash table entries: 512
[    0.001524] Initializing cgroup subsys cpuacct
[    0.001540] Initializing cgroup subsys memory
[    0.001583] Initializing cgroup subsys devices
[    0.001595] Initializing cgroup subsys freezer
[... 352 lines cut ...]
[   22.370182] vmouse_input_dev_close
[   23.402140] gmac0: probed
[   23.402527] gmac0 gmac0: eth0: eth0: PHY ID 00441400 at 0 IRQ poll (gmac0-0:00)
[   23.429760] 8021q: 802.1Q VLAN Support v1.8
[   24.114477] get_item_store: input buf bt_para bt_used
[   24.114501] get_item_store: get "bt_para"->"bt_used"
[   24.114574] get_item_show(815): script_get_item return type [int] for bt_para->bt_used
[   25.461029] vmouse_input_dev_open
[   27.400378] PHY: gmac0-0:00 - Link is Up - 100/Full
[   27.401052] ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[  962.040033] vmouse_input_dev_close
[  965.238134] vmouse_input_dev_open
[ 1053.027718] ohci_irq: fullspeed or lowspeed device disconnect
[ 1053.027824] usb 8-1: USB disconnect, device number 2
[ 1053.031099] xpad 8-1:1.2: xpad_irq_in - usb_submit_urb failed with result -19
[ 1053.031144] xpad 8-1:1.4: xpad_irq_in - usb_submit_urb failed with result -19
[ 1053.031183] xpad 8-1:1.6: xpad_irq_in - usb_submit_urb failed with result -19
[ 1054.531325] ehci_irq: highspeed device connect
[ 1054.680061] ehci_irq: highspeed device disconnect
[ 1054.680087] ohci_irq: fullspeed or lowspeed device connect
[ 1055.020089] usb 8-1: new full-speed USB device number 3 using sunxi-ohci
[ 1055.253620] Registered led device: xpad0
[ 1055.255065] Registered led device: xpad1
[ 1055.256589] Registered led device: xpad2
[ 1055.257832] Registered led device: xpad3

Kivutar, 2017-03-28:
Thanks for the report. We indeed removed a patch for Xbox One S controllers. It may be the cause of this bug.

Ntemis, 2017-03-28:
Hi rick-pit,
this new image has  9fe0d7e commit reverted. Please test and report if issue is fixed.
https://www.androidfilehost.com/?fid=817550096634757988

closed 2017-03-29 by Ntemis