libretro/Lakka-LibreELEC #91: RPi3/RC4 vice core not working

killer101, 2017-03-28:
Hello,

me again. The vice core won't work for me on the RPi3. 

Here's the log ...

RetroArch [INFO] :: This is RetroArch version 1.5.0 (Git 17235d7)
RetroArch [INFO] :: === Build =======================================
Capabilities: NEON VFPv3 VFPv4 
Built: Mar 23 2017
RetroArch [INFO] :: Version: 1.5.0
RetroArch [INFO] :: Git: 17235d7
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Loading default config.
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Redirecting save file to "/storage/savefiles/.srm".
RetroArch [INFO] :: Redirecting savestate to "/storage/savestates/.state".
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29970.03 Hz.
RetroArch [INFO] :: Found shader "/tmp/shaders/bilinear.glslp"
RetroArch [INFO] :: Found shader "/tmp/shaders/nearest.glslp"
RetroArch [INFO] :: Found shader "/tmp/shaders/stock.glsl"
RetroArch [INFO] :: Video @ fullscreen
RetroArch [INFO] :: Starting threaded video driver ...
RetroArch [INFO] :: [EGL]: EGL version: 1.4
RetroArch [INFO] :: [EGL]: Current context: 0x1.
RetroArch [INFO] :: Found GL context: videocore
RetroArch [INFO] :: Detecting screen resolution 1920x1080.
RetroArch [INFO] :: [EGL]: eglSwapInterval(1)
RetroArch [INFO] :: [EGL]: eglSwapInterval(1)
RetroArch [INFO] :: [GL]: Vendor: Broadcom, Renderer: VideoCore IV HW.
RetroArch [INFO] :: [GL]: Version: OpenGL ES 2.0.
RetroArch [INFO] :: GL: Using resolution 1920x1080
RetroArch [INFO] :: [GL]: Default shader backend found: glsl.
RetroArch [INFO] :: [Shader driver]: Using GLSL shader backend.
RetroArch [WARN] :: [GL]: Stock GLSL shaders will be used.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
[... 205 lines cut ...]
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_image_history.lpl].
MAIN FIRST
RetroArch [INFO] :: Environ GET_VARIABLE vice_Statusbar:
RetroArch [INFO] :: disabled
RetroArch [INFO] :: Environ GET_VARIABLE vice_Drive8Type:
RetroArch [INFO] :: 1540
RetroArch [INFO] :: Environ GET_VARIABLE vice_DriveTrueEmulation:
RetroArch [INFO] :: disabled
RetroArch [INFO] :: Environ GET_VARIABLE vice_RetroJoy:
RetroArch [INFO] :: disabled
RetroArch [INFO] :: Environ GET_VARIABLE vice_Controller:
RetroArch [INFO] :: keyboard
 0  x64
 1  /storage/roms/stick/c64/International Karate Plus.d64
bootp:(/storage/system/data)
Error - Default keymap not found, this should be fixed. Going on anyway...
Error - Default keymap not found, this should be fixed. Going on anyway...
Error - Default keymap not found, this should be fixed. Going on anyway...
Error - Default keymap not found, this should be fixed. Going on anyway...
Error - Default keymap not found, this should be fixed. Going on anyway...
HW scale not available, forcing to disabled
Error - Default keymap not found, this should be fixed. Going on anyway...
Error - Default keymap not found, this should be fixed. Going on anyway...
Error - Default keymap not found, this should be fixed. Going on anyway...
HW scale not available, forcing to disabled

killer101, 2017-03-28:
This post got it up and running. But maybe this is not the "out of the box" solution!

https://forums.libretro.com/t/vice-setup/9559

Kivutar, 2017-03-28:
The core itself is working, but you have to change the core settings. There is an issue open about that in the core issue tracker: https://github.com/libretro/vice-libretro/issues/2

killer101, 2017-03-28:
That is the smaller problem I think. Having "normal" people to copy the vice data folder from github over to the system folder is maybe the bigger task!

Kivutar, 2017-03-28:
Yes. @Twinaphex said he will try to improve that. But I don't know how. We could at least have a .zip containing that data folder hosted on the buildbot.

r-type, 2017-04-02:
There is an embedded compile time option.
I will try a build with --enable-embedded

r-type, 2017-04-02:
so i generated the 94 embedded files (and add #define USE_EMBEDDED 1 to config.h ), vice load fine with.
but one big problem remain. it miss the keyboard default keymap (that are not embedded).

Keyboard: Error - Default keymap not found, this should be fixed. Going on anyway... 
then kbd input are not working until you provide sdl_sym.vkm in data\C64

Kivutar, 2017-04-02:
It's already a nice progress!

commit d381fe24 (2017-04-03): Update vice, better default options + embedded data. Fixes #91
closed 2017-04-03 by Kivutar