libretro/Lakka-LibreELEC #92: C64 (Vice) playlist symbols?

killer101, 2017-03-28:
Hello.

Do Lakka have some? I'm just getting a black squares.

If so, what's the correct name for the playlist?

Kivutar, 2017-03-28:
We don't have icons for the C64 yet. We don't have a database for the games either: https://github.com/libretro/libretro-database/tree/master/rdb

However, no-into does have 3 C64 databases. @RobLoach do you have free time to add them in libretro-database?

RobLoach, 2017-03-28:
The DATs exist:
https://github.com/libretro/libretro-database/tree/master/metadat/no-intro

Just need to build the RDBs. Added an issue over at https://github.com/libretro/libretro-database/issues/432 .

As for black squares, made https://github.com/libretro/retroarch-assets/issues/131 in retroarch-assets. Feel free to close this issue. Not related to Lakka.

Thanks @killer101 + @Kivutar! Feel free to carry on discussion in those issues.

Kivutar, 2017-04-02:
I updated the assets, we now have icons for C64.

closed 2017-04-02 by Kivutar