libretro/Lakka-LibreELEC #93: Wetek Play video output problem

NuisMons, 2017-03-30:
I`m using WeTekPlay OpenELEC edition with LibreELEC 8.0.1 on NAND
Lakka loading from sd card without choose whats to load (no need to press power button on restart/start wetek play)
After first install Lakka video output set 720p but my tv can 1080p and video shows as 
[image]
 - changing resolution on LibreELEC (nand) to 1920x1080 not helped
 - set` video_fullscreen_x=1920` and `video_fullscreen_y=1080` in retroarch.cfg not helped

to aviod this, i wrote `autostart.sh` with `fbset -fb /dev/fb0 -g 1280 720 1280 1440 32`
i think, need force change resolution to 1920x1080 (or founded on init)
This problem only for Lakka, on LibreELEC all works fine

Ntemis, 2017-03-30:
If I produce an image can you test?

NuisMons, 2017-03-30:
Yes, ofc

Kivutar, 2017-03-30:
I can't reproduce the issue. This is how it looks for me with RC4 freshly installed:

[image]

My android settings are like this:

[image]

NuisMons, 2017-03-30:
I think whis is reason of issue using WeTekPlay OpenELEC edition with **LibreELEC 8.0.1 on NAND**

NuisMons, 2017-03-30:
when i compile with oldest[ platform_init ](https://drive.google.com/open?id=0B8PwdsWEFa9eNHZXb0dEUTI5aFE) all fine

Ntemis, 2017-03-30:
Please test this image
https://www.androidfilehost.com/?w=files&flid=167147
And report back, thank you

NuisMons, 2017-03-30:
Nothing changes
some data:
`Lakka:~ # cat /proc/cmdline
console=ttyS0,115200n8 logo=osd1,0x84100000,720p,full androidboot.resolution=720p hlt vmalloc=256m mem=1024m a9_clk_max=1512000000 vdachwswitch=cvbs hdmitx=cec0 mac=e8:18:63:50:48:ed boot=/dev/mmcblk0p1 disk=/dev/mmcblk0p2 BOOT_IMAGE=kernel.img console=tty0 consoleblank=0 scaling_governor=hotplug scaling_min_freq=200000 scaling_max_freq=1512000 systemd.show_status=auto`
`Lakka:~ # tail -n1 /sys/class/amhdmitx/amhdmitx0/disp_cap | sed -e 's/^[[:space:]]*//' -e 's/[[:space:]]*$//'
1080p24hz
`
This console data from platform_int initializations,
From first command lakka changed resolution to 720p,
but from second, lakka sets framebuffer(?) to 1080p24Hz

NuisMons, 2017-03-30:
i can change resolution to FullHD in ssh:
```
echo 1080p60hz > /sys/class/display/mode
fbset -fb /dev/fb0 -g 1920 1080  1920  2160 32
systemctl stop retroarch
systemctl start retroarch
```

Ntemis, 2017-03-30:
hmmm looks like  doesnt support 1080p 60Hz.

NuisMons, 2017-03-30:
mmmm... no
[image]
Maybe i need to use android firmware on NAND...

Ntemis, 2017-03-30:
ok let me recompile new image
Can you test?

NuisMons, 2017-03-30:
Yes, sure, but in ~20hrs

Ntemis, 2017-03-30:
Sure np

Kivutar, 2017-04-02:
@NuisMons any news on this?

NuisMons, 2017-04-02:
I`m using autostart.sh with `fbset -fb /dev/fb0 -g 1280 720 1280 1440 32 ` to avoid this issue
another problem with dualshock - they auto paired, connected, but disconnect after few seconds
Don`t see new image, and new commits to compile by myself. Maybe, if bug very rare - just add my solution to wiki (autostart.sh)

NuisMons, 2017-04-06:
maybe i should use
`105 fbset -fb /dev/fb0 -g $xRes $yRes $xRes $(( $yRes * 2 )) $bpp;` in `platform_init`?,
not
`fbset -fb /dev/fb0 -g $xRes $yRes $vXRes $vYRes $bpp;`? This ignore max resolution to set in framebuffer.
Try this, or need to find another solution?

Kivutar, 2017-04-06:
So it would be a double buffering issue?

NuisMons, 2017-04-06:
Can You try this in ssh to reproduce my bug?
```
echo 720p > /sys/class/display/mode
fbset -fb /dev/fb0 -g 1280 720  1920  2160 32
systemctl stop retroarch
systemctl start retroarch
```

Kivutar, 2017-04-06:
Yes it reproduces the bug

NuisMons, 2017-04-06:
I think its need to change in platform_init to
`105 fbset -fb /dev/fb0 -g $xRes $yRes $xRes $(( $yRes * 2 )) $bpp;`
or add `echo $maxres > /sys/class/display/mode` before this line

Kivutar, 2017-04-06:
I think the problem comes from the fact that you don't have hdmimode=1080p in your cmdline, which is certainly set by Android in my case.

Can you try a cat /proc/cmdline ?

NuisMons, 2017-04-06:
yes, i haven`t andoid and i havent this line
```
Lakka:~ # cat /proc/cmdline
console=ttyS0,115200n8 logo=osd1,0x84100000,720p,full androidboot.resolution=720p hlt vmalloc=256m mem=1024m a9_clk_max=1512000000 vdachwswitch=cvbs hdmitx=cec0 mac=e8:18:63:50:48:ed boot=/dev/mmcblk0p1 disk=/dev/mmcblk0p2 BOOT_IMAGE=kernel.img console=tty0 consoleblank=0 scaling_governor=hotplug scaling_min_freq=200000 scaling_max_freq=1512000 systemd.show_status=auto

```

Kivutar, 2017-04-06:
I have this:

```
Lakka:~ # cat /proc/cmdline 
console=ttyS0,115200n8 logo=osd1,0x84100000,1080p,full androidboot.resolution=1080p hdmimode=1080p cvbsmode=576cvbs nohlt vmalloc=256m mem=1024m a9_clk_max=1512000000 vdachwswitch=cvbs hdmitx= mac=e8:18:63:50:07:ca boot=/dev/mmcblk0p1 disk=/dev/mmcblk0p2 BOOT_IMAGE=kernel.img console=tty0 consoleblank=0 scaling_governor=hotplug scaling_min_freq=200000 scaling_max_freq=1512000 systemd.show_status=auto
```

And platform init is looking for this `hdmimode=1080p`. It if doesn't find it, it defaults to `hdmimode=720p` which causes the bug here.

There may be a way to access the NAND to edit the cmdline and fix this.

NuisMons, 2017-04-06:
i think, if this line exists, need to use default 720p as in platform_init? or use maxRes

NuisMons, 2017-04-06:
or use androidboot.resolution=1080p (720p) to configure, but platform init really ignores 720p to config framebuffer

Kivutar, 2017-04-06:
720p is the default setting here https://github.com/lakkatv/Lakka-LibreELEC/blob/Lakka-V2.0/projects/WeTek_Play/initramfs/platform_init#L20

But if hdmi_mode is set in the cmdline https://github.com/lakkatv/Lakka-LibreELEC/blob/Lakka-V2.0/projects/WeTek_Play/initramfs/platform_init#L22-L29

It sets the display mode https://github.com/lakkatv/Lakka-LibreELEC/blob/Lakka-V2.0/projects/WeTek_Play/initramfs/platform_init#L31

And the maxRes https://github.com/lakkatv/Lakka-LibreELEC/blob/Lakka-V2.0/projects/WeTek_Play/initramfs/platform_init#L53

And the xRes and yRes https://github.com/lakkatv/Lakka-LibreELEC/blob/Lakka-V2.0/projects/WeTek_Play/initramfs/platform_init#L76-L102

With your cmdline, hdmi_mode is never set, and you end up with the default 720p.

NuisMons, 2017-04-06:
Lakka:~ # tail -n1 /sys/class/amhdmitx/amhdmitx0/disp_cap | sed -e 's/^[[:space:]]*//' -e 's/[[:space:]]*$//'
1080p24hz

```
if [ -z "$maxRes" ]; then
  maxRes="$hdmimode";
``` returns maxRes=1080p24hz, not 720p, or i`m wrong?

NuisMons, 2017-04-06:
I so sorry, but i have no more ideas...

Kivutar, 2017-04-06:
Maybe it's the 24hz that cause problems.

NuisMons, 2017-04-06:
24hz not using at any place, and, as i shows my tv works good at 1080p60hz, if im write in autostart.sh:
`echo 1080p60hz > /sys/class/display/mode
fbset -fb /dev/fb0 -g 1920 1080  1920  2160 32`
think `fbset -fb /dev/fb0 -g 1280 720 1920 2160 32;` is problem

NuisMons, 2017-04-06:
Can You just change resolution to 720p in android and reboot to lakka?
Resolution changed to 720p and not video bug?

Kivutar, 2017-04-06:
If I change the resolution to 720p in android, I get the bug in Lakka

NuisMons, 2017-04-07:
I suggest (not testet yet)
[platform_init.txt](https://github.com/lakkatv/Lakka-LibreELEC/files/904806/platform_init.txt)

Kivutar, 2017-04-07:
Here is an image with your PR merged, if you can test it http://static.kivutar.me/Lakka-WeTek_Play.arm-2.0.img.gz

NuisMons, 2017-04-07:
Your image have video bug, But i compile image by myself - all good, no more issue

Kivutar, 2017-04-07:
How come? I just merged your PR and recompiled

Kivutar, 2017-04-07:
Ok, I need to clean the kernel and initramfs

NuisMons, 2017-04-07:
I compile first time for ~7hrs, now collecting logs for official dualshocksfrom ps3  not worked on bluetooth(

Kivutar, 2017-04-07:
We should discuss the dualshock issue on another topic. But rememer to press the hidden reset button before pairing it.

Kivutar, 2017-04-07:
I uploaded a new image at the same address: http://static.kivutar.me/Lakka-WeTek_Play.arm-2.0.img.gz

Can you give it a try please?

NuisMons, 2017-04-07:
This image works good, no video bug for me. My lakka boots in 720p (default no android) and all cool

Kivutar, 2017-04-07:
Great, I will try it tomorrow on mine with 2 or 3 tvs

Kivutar, 2017-04-09:
Works for me too, thanks for the PR.

closed 2017-04-09 by Kivutar

drunk3nm0nk3y, 2017-08-04:
Just wanted to say that I had the exact same problem with odroid-c2, the latest stable release of Lakka and a 720p@50/60Hz TV. Setting the framebuffer with autostart.sh like described by @NuisMons did the trick for me