libretro/Lakka-LibreELEC #94: Add support for wifi usb with Atheros AR9271 802.11n on BananaPi image

n3wm4rk, 2017-04-02:
Hey, can you add support for USB WIFI TP-LINK TL-WN722N. It's based on Atheros AR9271. On my Bananapi M1 Lakka couldn't load firmware.

from dmesg:

[   50.353327] usb 3-1: ath9k_htc: Firmware htc_9271.fw requested
[   50.369626] usbcore: registered new interface driver ath9k_htc
[  111.224804] usb 3-1: ath9k_htc: Failed to get firmware htc_9271.fw
[  111.229679] usb 3-1: ath9k_htc: USB layer deinitialized

from lsusb:

Bus 003 Device 002: ID 0cf3:9271 Qualcomm Atheros Communications AR9271 802.11n

gouchi, 2017-04-02:
It is [enabled](https://github.com/lakkatv/Lakka-LibreELEC/blob/Lakka-V2.0/projects/Bananapi/options#L119). Can you check that the firmware is present by typing `ls -l /lib/firmware/htc_9271.fw`

n3wm4rk, 2017-04-03:
firmware is present:
lakka:~ # ls -l /lib/firmware/htc_9271.fw
-rw-rw-r--    1 root     root         50980 Mar 27 12:55 /lib/firmware/htc_9271.fw

but dmesg is telling me that:
[   29.054628] usb 1-1: ath9k_htc: Firmware htc_9271.fw requested
[   29.073249] usbcore: registered new interface driver ath9k_htc
[   89.133771] usb 1-1: ath9k_htc: Failed to get firmware htc_9271.fw
[   89.138682] usb 1-1: ath9k_htc: USB layer deinitialized

On my bananian linux installed on BananaPi wifi is working, firmware is loaded.

gouchi, 2017-04-03:
What is the output of ?

`modinfo ath9k_htc | grep firmware`

Thank you.

n3wm4rk, 2017-04-04:
```
lakka:~ # modinfo ath9k_htc|grep firmware
firmware:       htc_9271.fw
firmware:       htc_7010.fw

lakka:~ # lsmod
Module                  Size  Used by
8021q                  18691  0
garp                    5645  1 8021q
stp                     1890  1 garp
llc                     4764  2 stp,garp
ath9k_htc              53244  0
ath9k_common            2686  1 ath9k_htc
ath9k_hw              348863  2 ath9k_common,ath9k_htc
ath                    15869  3 ath9k_common,ath9k_htc,ath9k_hw
mac80211              240474  1 ath9k_htc
mali                  113629  2
ump                    56474  3 mali
```

gouchi, 2017-04-04:
After booting, if you unplug and plug your wifi usb adapter, does it work ?

n3wm4rk, 2017-04-06:
Unfortunately, it does not work when I disconnect and connect my wifi card. I get the same messages in dmesg :(

Kivutar, 2017-06-05:
What image are you using? Bananapi or H3.bpim2p?

Kivutar, 2017-07-12:
No replies, I close this.

closed 2017-07-12 by Kivutar