libretro/Lakka-LibreELEC #97: GPU Issue - black screen after booting (NVIDIA GeForce GT630)

lanciaux, 2017-04-03:
If you are sure that it's a bug in Lakka, please answer the following questions:

- Which version of Lakka are you using? All Lakka 1.x - 2.0RC4 since early November 2016 
- What system hardware are you using? AMD Athlon(tm) 64 X2 Dual Core Processor 4200+
- What did you do? Make USB stick with all releases since november + start USB in "live mode"
- What did you expect to happen? Reaching menu like on my laptops with AMD GPU or Intel On Board
- What happened instead? Black screen but session is running (usb stick blinking then reacting to keyboard actions).

As others, I suspect this is related to NOUVEAU driver and I tried all kind of configuration modifications as nouveau.modeset=0, etc. found in forums. Neither i was lucky to see final screen as I have other issues and/or I got the flower logo then usb light stayed on. It was running perfect until October 2016 and it has changed in November 2016 releases.

As Lakka 2.0 is based on LibreElec, I tested yesterday (in order to convince myself this could be also related to my dying hardware) for the very first time of my life LibreElec 8.0.1 - apart this is a beautiful media manager (it was stunned) ... usb stick has booted "live" right at first boot with such big speed I am sure my Nvidia GeGT630 got perfectly recognized and used.

Why then Lakka break if based on same LibreElec ?

Ludovic

PS : Amazing Lakka/Retroarch - I thank whole team for such great software I shared with my daugthers 30 years younger than some games they have played with.:-)

gouchi, 2017-04-03:
Hi Ludovic,

Your graphic card should be supported by nouveau driver.

Is it possible to get some [Lakka system and graphic logs](http://www.lakka.tv/doc/Troubleshooting-Lakka/) ? 

You can try to boot by adding those parameters to [cmdline](http://www.lakka.tv/doc/The-bootloader/#on-generic-pc) ?
`live nosplash debugging textmode`

Thank  you.

Kivutar, 2017-04-03:
> Why then Lakka break if based on same LibreElec ?

LibreELEC uses X11 while we use DRM/KMS to avoid overhead.

lanciaux, 2017-04-05:
Hello, once again, I made numerous tests to be able to boot whatever method is and for example "live nosplash debugging textmode" has no effect. Removing all cmdline parameters after append except boot uuid shows kernel loading, loading nouveau, then black screen again before i could read something useful. SSH not working (in RC4) - i suspect kernel doesn't have enough time to start it before crashing. I cannot (anymore) ask nomodeset or nouveau.modeset=0 as this would prevent KMS to start ... so every NVIDIA owner of GeForce 6x-8y in trouble ?

Who has a NVIDIA card GeForce managed to run Lakka 2.0 live until the menu screen ?

Does someone know a kernel cmdline command to output "boot log" to a file while booting ?

Thank you.

Ludovic

gouchi, 2017-04-05:
> for example "live nosplash debugging textmode" has no effect.

So you enter this cmdline, press enter then nothing appears on the screen ? Just black screen ? I am kind of surprised you don't see any debug messages ?

Can you try to a picture or video of your screen ?

My card is older than yours and it is working.

Thank you.

NuisMons, 2017-04-06:
Usually `nomodeset` in cmdline helps me to loads live linux with nvidia card normally

lanciaux, 2017-04-06:
with "live nomodeset" => I reach a point where it is written "*** error in mount_flash : mount_common : Could not mount UUID=2703-3417 ***" starting debugging shell ... sh : can't access tty ; job control turned off" 

Then if you write "exit" you get the flower, but you stop with can't stat '/flash/system'

gouchi, 2017-04-06:
Can  you try to use [Etcher](https://etcher.io) to burn the live usb image ?
Thank you.

lanciaux, 2017-04-07:
Apoligize for previous post reply. Mount errors were due to usb key issue with partition resizing (due to reboots). Once I burn again and let lakka resize on a pc with no gpu issue (amd firegl) ... so seeing menu ...  I switch back to my nvidia based pc then cmdline 'live' black screen no flower, cmdline 'live nomodeset' flower appears ssh up and from ssh console flower stays on screen and retroarch -v complaints 'no drm' (due to nomodeset ?). I will drop a real log tonight as I have to go now.

lanciaux, 2017-04-07:
Hello here are finally complete data - obtained with ssh in cmdline. Without nomodeset, I have indeed DRM loaded (checks logs). Nouveau is loaded but screen is always black. I have absolutely no idea how to go further. So I'm open to check your different proposal.

Ludovic

[GeForce930_retroarch_log.txt](https://github.com/lakkatv/Lakka-LibreELEC/files/906400/GeForce930_retroarch_log.txt)
[GeForce930_lsusb_log.txt](https://github.com/lakkatv/Lakka-LibreELEC/files/906397/GeForce930_lsusb_log.txt)
[GeForce930_lspci_log.txt](https://github.com/lakkatv/Lakka-LibreELEC/files/906399/GeForce930_lspci_log.txt)
[GeForce930_log.txt](https://github.com/lakkatv/Lakka-LibreELEC/files/906401/GeForce930_log.txt)
[GeForce930_aplay_log.txt](https://github.com/lakkatv/Lakka-LibreELEC/files/906398/GeForce930_aplay_log.txt)

gouchi, 2017-04-07:
Fixed by adding to cmdline `video=DVI-I-1:d video=VGA-1:d video=HDMI-A-1:e`

closed 2017-04-07 by gouchi