libretro/Lakka-LibreELEC #100: Does not remember bluetooth pairing with controller

R3d-Sky, 2017-04-07:
BEFORE you report a bug make sure you have tried the latest nightly version of Lakka at http://sources.lakka.tv/nightly/. Your bug might be already fixed.

If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
Lakka 2.0 RC4
- What system hardware are you using?
Raspberry Pi 3, Controller: 2017 NVidia Shield (as in Shield TV box)
- What did you do?
Paired controller with these instructions:
http://www.lakka.tv/doc/Wireless-Dualshock/
Played a while, powered off, and powered on again later.
- What did you expect to happen?
Expected it to remember controller pairing.
- What happened instead?
Have to pair controller via SSH again.

hooby3dfx, 2017-04-07:
Please check out https://github.com/lakkatv/Lakka-LibreELEC/issues/83 (your issue is possibly a duplicate of that)

R3d-Sky, 2017-04-07:
You're right. I derped.

closed 2017-04-07 by R3d-Sky