libretro/Lakka-LibreELEC #105: WeTekPlay Lakka-V2.0 RC4 shaders too slow

NuisMons, 2017-04-10:
- Which version of Lakka are you using?
Lakka V2.0 with latest patches
- What system hardware are you using?
Wetek Play
- What did you do?
Try using some shaders to games
- What did you expect to happen?
Play games with enchanged looks
- What happened instead?
Lakka extremelly slows ui, and games worked too slow (No hardware acceleration?)

In 2016, i played games on raspberry Pi 1 B  with using some shaders, and there is no problems with it. Now, I try shaders at wetek play and all goes very slow, No hardware acceleration in wetek play and its ok? In LibreELEC (https://github.com/LibreELEC/LibreELEC.tv/blob/master/projects/WeTek_Play/linux/linux.arm.conf) 
```
#
# Mali 400 UMP device driver
#
CONFIG_UMP=y
# CONFIG_UMP_DEBUG is not set

#
# Mali GPU OpenGL device driver
#
CONFIG_MALI400=m
# CONFIG_MALI400_DEBUG is not set
# CONFIG_MALI400_PROFILING is not set
CONFIG_MALI400_UMP=y
# CONFIG_MALI_DVFS is not set
CONFIG_MALI_DMA_BUF_MAP_ON_ATTACH=y
CONFIG_MALI_SHARED_INTERRUPTS=y
CONFIG_MESON6_GPU_EXTRA=y
# CONFIG_MALI_PMU_PARALLEL_POWER_UP is not set
# CONFIG_MALI_DT is not set
# CONFIG_MALI_QUIET is not set 
```
But Lakka have not this options. Its correct, or prefer to not use shaders on Lakka?

Kivutar, 2017-04-10:
I notice that the shaders were very slow on WeTek Play too.

I suggest you to set the scale to 2 when you try a shader. It prevents them to run at the resolution of your TV. It will run them at the resolution of the game * 2.

Also, I'm going to change how the shaders work in Lakka soon, by adding a simple mode with 10 to 20 popular shader presets scaled at 2.

It may fix this bug as a side effect. If not, I scare that it will be a "won't fix".

NuisMons, 2017-04-17:
Just try shaders and games again, and now I think gpu of wetek play really bad.
I think slow shaders is not bug (find some useful), but I want shaders work only in games, not in lakka ui, because sometimes need to poweroff wetekplay to unfreeze user interface.
Thanks, and sorry my bad english)

Kivutar, 2017-05-13:
I don't think we can fix this.

closed 2017-05-13 by Kivutar