libretro/Lakka-LibreELEC #107: WeTekPlay dualshocks 3 original not worked over bluetooth

NuisMons, 2017-04-11:
If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
Lakka v2.0 compiled 11.04.17
- What system hardware are you using?
WetekPlay
- What did you do?
Try to connect ds3 over bluetooth
- What did you expect to happen?
working joystics
- What happened instead?

DS3 auto paired
disconnect ds3 over usb
press PS button
Lakka show "playstation(R) 3 connected #0", when menu jumped up up up... (as i press "up" button many times)
3-5 seconds and lakka show "disconnected PLAYSTATION(R) 3 from #0"

dmesg:
```
[  676.008131@1] sony 0005:054C:0268.0009: unknown main item tag 0x0
[  676.008158@1] sony 0005:054C:0268.0009: unknown main item tag 0x0
[  676.010537@1] input: PLAYSTATION(R)3 Controller as /devices/platform/uart_0/tty/ttyS1/hci0/hci0:13/input10
[  676.014490@1] sony 0005:054C:0268.0009: input,hidraw0: BLUETOOTH HID v0.00 Joystick [PLAYSTATION(R)3 Controller] on 20:70:02:a0:00:00
[  686.085727@1] sony: probe of 0005:054C:0268.0009 failed with error -5

```

Then just connected by usb (worked):
```
[ 1115.247275@1] Indeed it is in host mode hprt0 = 00021501
[ 1115.437238@0] usb 2-1: new full-speed USB device number 4 using dwc_otg
[ 1115.445807@0] Indeed it is in host mode hprt0 = 00021501
[ 1115.661714@0] usb 2-1: New USB device found, idVendor=054c, idProduct=0268
[ 1115.661731@0] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1115.661754@0] usb 2-1: Product: PLAYSTATION(R)3 Controller
[ 1115.661764@0] usb 2-1: Manufacturer: Sony
[ 1115.682793@1] sony 0003:054C:0268.000C: Fixing up Sony Sixaxis report descriptor
[ 1115.700281@0] input: Sony PLAYSTATION(R)3 Controller as /devices/lm0/usb2/2-1/2-1:1.0/input/input13
[ 1115.708298@0] sony 0003:054C:0268.000C: input,hiddev0,hidraw0: USB HID v1.11 Joystick [Sony PLAYSTATION(R)3 Controller] on usb-lm0-1/input0
```

NuisMons, 2017-04-14:
I tryed external bluetooth dongle, and nothing changed
in /storage/.cache/bluetooth/XX:XX:XX:XX joystics presents
maybe i do something wrong? joys worker with scp server in win10 and with PS3

Kivutar, 2017-05-31:
I can reproduce the issue.

Kivutar, 2017-10-21:
Maybe related to https://github.com/libretro/Lakka-LibreELEC/issues/54

ToKe79, 2018-01-31:
Issue seems to be resolved by https://github.com/scobo/Lakka-LibreELEC/commit/ef35580a45ffddaaa8651d616569a4ea2c50cf9f. If the issue still exists with latest [release](http://www.lakka.tv/get) and/or [nightly](http://nightly.builds.lakka.tv), please feel free to re-open this issue. Thank you.

closed 2018-01-31 by ToKe79