libretro/Lakka-LibreELEC #108: lakka RC4 x64 don't start any core on LattePanda x64

kingknuts, 2017-04-13:
Hi!

I'm testing the latest release candidate (RC4) of Lakka on my Lattepanda and it cannot start any core. It just froze making it to press the reset button.

The stable version of Lakka works everything excepting the Wi-Fi (Broadcom 4334) and the touchscreen (No very important actually).
The Wi-Fi driver for windows is found here:
https://github.com/LattePandaTeam/LattePanda-Win10-Software/tree/master/Drivers/LattePanda_4G64GB_Drivers/MRD-64bit-Intel-Driver-Release/Drivers/WIFI

The linux driver: http://www.lattepanda.com/forum/viewtopic.php?f=6&t=58&start=90#p2327

Here is the support topic based on Ubuntu. The Wi-Fi is working on it. I don't know if a port of the linux drivers could be done:
http://www.lattepanda.com/forum/viewtopic.php?f=6&t=58&start=140

Thanks a lot for your hard work.

gouchi, 2017-04-13:
The [issue template](https://github.com/libretro/Lakka-LibreELEC/blob/Lakka-V2.0/ISSUE_TEMPLATE.md) is not working ?

Please provide some [Lakka system log](http://www.lakka.tv/doc/Troubleshooting-Lakka/).

Thank you.

kingknuts, 2017-04-13:
Clearly seems like not:

Which version of Lakka are you using?
Lakka 2.0 RC4

What system hardware are you using?
Lattepanda Enhanced:
Processor: Intel Cherry Trail Atom x5 Quad-Core Z8300
RAM: 4GB
Storage: 64 GB MMC
BIOS -> Forced UEFI (Cannot change to Legacy/CSM)
Controller: Mad Catz Call of Duty Gamepad Xbox 360 Wired Controller
Network: Wireless Broadcom 4334
Ethernet: Intel 10/100
Graphics: Touchscreen of 5 points Goodix Touch HID
Extras:
-MicroController Arduino Leonardo Embedded
-Dualbooted with Windows 10 Pro.

What did you do?
Run a Picture.
Run Game Boy Rom (Mario Six Golden Coins)
Run MrBoomb core
Connect to home network to play multiplayer.

What did you expect to happen?
When connecting to home network. Just connect to mount an NFS share and also 2 LAN players.
Open the Picture
Run the game
Start the core

What happened instead?
No networks detected.
Run a Picture. -> System Freeze
Run Game Boy Rom (Mario Six Golden Coins) -> System Freeze
Run MrBoomb core -> System Freeze

Cannot get the log as the system completely freezes and it not even connecting to the ssh. I will wipe the Hard disk and install Lakka alone. I will try to get that way the logs.

Thanks for your reply.

Kivutar, 2017-04-14:
Could it be a duplicate of this bug? https://github.com/libretro/Lakka-LibreELEC/issues/35

Maybe it's worth trying to stop connman and test again:

    systemctl stop connman

kingknuts, 2017-04-14:
Is not exactly as #35 as It cannot open. I  left there opening Mr.Boom for more than an hour and nothing...
the systemctl stop connman don't fix it. It just let me out of the SSH session as expected.
The 1st log attached is using just the normal start of the Mr.boom core.
[LOG.txt](https://github.com/libretro/Lakka-LibreELEC/files/922392/LOG.txt)

This is the dmesg log:
[DMESG.txt](https://github.com/libretro/Lakka-LibreELEC/files/922394/DMESG.txt)

This is the lspci log:
[LSPCI.txt](https://github.com/libretro/Lakka-LibreELEC/files/922397/LSPCI.txt)

If you need anything else let me know.

Thanks a lot

gouchi, 2017-04-15:
Can you try some settings in your BIOS ?

- [GOP Driver](http://www.intel.com/content/www/us/en/intelligent-systems/intel-embedded-graphics-drivers/faq-bios-firmware.html) is disabled.

- LPE Audio Support is set to LPE Audio PCI mode in Chipset > South Bridge > Audio Configuration

Thank you.

Kind regards,

kingknuts, 2017-04-15:
With GOP Driver disabled I don't have signal on the screen... With the LPE Audio Support nothing changes. It still freeze.

Thanks Again.

gouchi, 2017-04-15:
Can you try to provide some log ?

Another thing you can try is to disable `video_threaded` by setting it to false in the [configuration file](http://www.lakka.tv/doc/Configuring-Lakka).

Thank you for tests.

kingknuts, 2017-04-15:
Ok, with the video_threaded changed happens the same but the log is different
[LOG video_threaded.txt](https://github.com/libretro/Lakka-LibreELEC/files/923617/LOG.video_threaded.txt)

kingknuts, 2017-04-15:
After rebooting the whole system the log is this:
[LOG video_threaded-reboot.txt](https://github.com/libretro/Lakka-LibreELEC/files/923622/LOG.video_threaded-reboot.txt)

gouchi, 2017-04-15:
Can you try boot to Windows and reboot to boot to Lakka if it is changing something ?

[Source](https://github.com/libretro/Lakka/issues/398#issuecomment-225147406)

kingknuts, 2017-04-15:
It only changed my IP address...
[LOG WinReboot.txt](https://github.com/libretro/Lakka-LibreELEC/files/923646/LOG.WinReboot.txt)

kingknuts, 2017-04-15:
Now I tried with a GB rom and the log is the same...
[LOG Gambatte.txt](https://github.com/libretro/Lakka-LibreELEC/files/923654/LOG.Gambatte.txt)

kingknuts, 2017-04-18:
Updated to RC5... The Issue with the cores and the Wireless card still persists...
Should I close this issue and open a new one??

Kivutar, 2017-04-18:
We can keep this one. I rename it.

kingknuts, 2017-04-25:
Updated to the stable 2.0 release. Still not working, so I will switch back to the latest stable Lakka 1.0. I will keep the dualboot to test the new releases and see if it gets fixed.
Thank you

kingknuts, 2017-05-22:
One question:
Do you have any plans to make nightly builds as the previous Lakka versions? Also I've installed the 2.0 on my laptop and for some strange reason Pcsx-rearmed works but bettle-psx not...(I have lower or even no priority on this last thing)

Kivutar, 2017-05-23:
Releasing everyday would take too much disk space, each batch of images is more than 10Go now. I plan to do one update per month. I should publish 2.1-rc1 today or tomorrow.

gouchi, 2017-05-30:
Is it possible to make a test with [2.1-rc1](http://le.builds.lakka.tv/Generic.x86_64/Lakka-Generic.x86_64-2.1-rc1.img.gz) ?

Thank you.

kingknuts, 2017-05-31:
Right. So far the WiFi is still not working and I didn't bring with me the network cable so no logs right now until I reach my office.
The are improvements! Now the retroarch when I start a core the screen turns black and then the berry appears and then it freezes at it. Before that it was freezing just in the xmb.

gouchi, 2017-08-06:
Is it possible to make a test with [Lakka 2.1 rc4](http://le.builds.lakka.tv/Generic.x86_64/Lakka-Generic.x86_64-2.1-rc4.img.gz) ?

Thank you.

kingknuts, 2017-08-20:
Yep, downloading!!

ToKe79, 2018-01-31:
Issue closed as no response received. If the issue still exists with latest [release](http://www.lakka.tv/get) and/or [nightly](http://nightly.builds.lakka.tv), please feel free to re-open this issue. Thank you.

closed 2018-01-31 by ToKe79

kingknuts, 2018-01-31:
I apologize as my board is broken, I hope I will get the replacement this week. Then I will report back

raidensnake, 2020-04-06:
I apologise for having to bring up this old thread however I can confirm the same issue on the latest version as it requires the 4.12 lattepanda kernel and the firmware for both the wifi and bluetooth. Is there anyway of customising this in the source code?