libretro/Lakka-LibreELEC #109: Sound device and speakers RC4/x64

killer101, 2017-04-16:
Hi!

I've just tested some features using my PC. I have a HP Prodesk 600 G2 Mini-Desktop. I have a monitor and a TV connected via DisplayPort2HDMI cables. No problems using this with Windows so far.

The Prodesk has some kind of internal speaker. If I use Lakka on this machine, it always uses the internal speaker of the Prodesk. Using the monitor or the TV as output device makes no change. Is there any feature to change the output device?

By the way ... Having both the monitor and the TV connected to the Prodesk (even if turned off), Lakka always tries to send the picture to the monitor. On the TV I just see the berry. Disconecting the cables off the monitor solves the problem. I think LibreElec tries to expand the "desktop" to the second screen. Maybe there is a switch in LibreElec to set it to "clone" the screen by default to the second device. If so, maybe it is a good idea to use it!

gouchi, 2017-04-16:
Did you try to [change audio device ](http://www.lakka.tv/doc/Audio-settings/#switching-audio-device-from-the-gui)?

killer101, 2017-04-16:
Shame on me! It works as you said.

I looked at the audio device setting before but there wasn't anything set up on the right side. Just blank. I didn't came up with the idea to use the left und right keys. I pressed a button and the virtual keyboard popped up. Issue solved! ;-)

Any idea on the second screen?

closed 2017-04-16 by killer101