libretro/Lakka-LibreELEC #110: RPi3/RC5 hatari core not working

killer101, 2017-04-20:
Still not working ...

RetroArch [INFO] :: This is RetroArch version 1.5.0 (Git c045732)
RetroArch [INFO] :: === Build =======================================
Capabilities: NEON VFPv3 VFPv4 
Built: Apr 13 2017
RetroArch [INFO] :: Version: 1.5.0
RetroArch [INFO] :: Git: c045732
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Loading default config.
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Redirecting save file to "/storage/savefiles/.srm".
RetroArch [INFO] :: Redirecting savestate to "/storage/savestates/.state".
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29970.03 Hz.
RetroArch [INFO] :: Found shader "/tmp/shaders/bilinear.glslp"
RetroArch [INFO] :: Found shader "/tmp/shaders/nearest.glslp"
RetroArch [INFO] :: Found shader "/tmp/shaders/stock.glsl"
RetroArch [INFO] :: Video @ fullscreen
RetroArch [INFO] :: Starting threaded video driver ...
RetroArch [INFO] :: [EGL]: EGL version: 1.4
RetroArch [INFO] :: [EGL]: Current context: 0x1.
RetroArch [INFO] :: Found GL context: videocore
RetroArch [INFO] :: Detecting screen resolution 1920x1080.
RetroArch [INFO] :: [EGL]: eglSwapInterval(1)
RetroArch [INFO] :: [EGL]: eglSwapInterval(1)
RetroArch [INFO] :: [GL]: Vendor: Broadcom, Renderer: VideoCore IV HW.
RetroArch [INFO] :: [GL]: Version: OpenGL ES 2.0.
RetroArch [INFO] :: GL: Using resolution 1920x1080
RetroArch [INFO] :: [GL]: Default shader backend found: glsl.
RetroArch [INFO] :: [Shader driver]: Using GLSL shader backend.
RetroArch [WARN] :: [GL]: Stock GLSL shaders will be used.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
[... 92 lines cut ...]
RetroArch [INFO] :: 640x480
[libretro-test]: Got size: 640 x 480.
RetroArch [INFO] :: Content loading skipped. Implementation will load it on its own.
RetroArch [INFO] :: Environ SET_INPUT_DESCRIPTORS:
RetroArch [INFO] :: RetroPad, User 1, Button "Y (left)" => "Enter GUI"
RetroArch [INFO] :: RetroPad, User 1, Button "Select" => "Mouse mode toggle"
RetroArch [INFO] :: RetroPad, User 1, Button "Start" => "Keyboard overlay"
RetroArch [INFO] :: RetroPad, User 1, Button "D-Pad Up" => "Up"
RetroArch [INFO] :: RetroPad, User 1, Button "D-Pad Down" => "Down"
RetroArch [INFO] :: RetroPad, User 1, Button "D-Pad Left" => "Left"
RetroArch [INFO] :: RetroPad, User 1, Button "D-Pad Right" => "Right"
RetroArch [INFO] :: RetroPad, User 1, Button "A (right)" => "Fire"
RetroArch [INFO] :: RetroPad, User 1, Button "L" => "Joystick number"
RetroArch [INFO] :: RetroPad, User 1, Button "R" => "Mouse speed"
RetroArch [INFO] :: RetroPad, User 1, Button "L2" => "Toggle m/k status"
 0  hatari
 1  --disk-a
 2  /storage/roms/stick/atarist/International Karate Plus.stx
RetroConf:'/storage/system/hatari.cfg'
Hatari v1.8.0, compiled on:  Apr 13 2017, 12:21:38
Configured max Hatari resolution = 640x480, optimal for ST = 832x588
STX : STX_Insert_internal drive=0 file=/storage/roms/stick/atarist/International Karate Plus.stx buf=0x74314008 size=890124
Inserted disk '/storage/roms/stick/atarist/International Karate Plus.stx' to drive A:.
Building CPU table for configuration: 68000 (compatible mode)
Bus error (core dumped)

killer101, 2017-05-12:
Still exists in the release version!

Kivutar, 2017-06-05:
@killer101 where did you put the tos.img?

rgruenewald, 2017-06-19:
same issue here. I've the tos.img next to the retroarch binary.

killer101, 2017-07-01:
To the system folder!

Kivutar, 2017-07-01:
For me the tos.img needs to be in the same folder as the retroarch binary. It's a limitation of this emulator.

masid101, 2017-10-12:
Where exactly is the retroarch binary?

Kivutar, 2017-10-13:
It's in /usr/bin, but you can move it to somewhere else by copying it, I do this for my tests.

mstempell, 2017-12-07:
I got the same kind of cash with any TOS i may try (UK, US, 1.02, 1.62).

```
[INFO] Updating firmware status for: /tmp/cores/hatari_libretro.so on /storage/system
[INFO] Using content: /storage/roms/Llamatron.st.
[INFO] arg #0: retroarch
[INFO] arg #1: /storage/roms/Llamatron.st
[INFO] arg #2: -s
[INFO] arg #3: /storage/savefiles
[INFO] arg #4: -S
[INFO] arg #5: /storage/savestates
[INFO] arg #6: -c
[INFO] arg #7: /storage/.config/retroarch/retroarch.cfg
[INFO] arg #8: -L
[INFO] arg #9: /tmp/cores/hatari_libretro.so
[INFO] [Video]: Does not have enough samples for monitor refresh rate estimation. Requires to run for at least 4096 frames.
[INFO] Set config file to : /storage/.config/retroarch/retroarch.cfg
[INFO] RetroArch 1.6.9 (Git c056726)
[INFO] Redirecting save file to "/storage/savefiles/Llamatron.srm".
[INFO] Redirecting savestate to "/storage/savestates/Llamatron.state".
[INFO] === Build =======================================
Capabilities: NEON VFPv3 VFPv4
Built: Nov 21 2017
[INFO] Version: 1.6.9
[INFO] Git: c056726
[INFO] =================================================
[INFO] Loading dynamic libretro core from: "/tmp/cores/hatari_libretro.so"
[INFO] [overrides] no core-specific overrides found at /storage/.config/retroarch/config/Hatari/Hatari.cfg.
[INFO] [overrides] no game-specific overrides found at /storage/.config/retroarch/config/Hatari/Llamatron.cfg.
[INFO] Shaders: preset directory: /tmp/shaders/presets
[INFO] Shaders: no game-specific preset found at /tmp/shaders/presets/Hatari/Llamatron.cgp.
[INFO] Shaders: no game-specific preset found at /tmp/shaders/presets/Hatari/Llamatron.glslp.
[INFO] Shaders: no core-specific preset found at /tmp/shaders/presets/Hatari/Hatari.cgp.
[INFO] Shaders: no core-specific preset found at /tmp/shaders/presets/Hatari/Hatari.glslp.
[INFO] Environ SET_VARIABLES.
[INFO] Remaps: remap directory: /storage/remappings
[INFO] Remaps: no game-specific remap found at /storage/remappings/Hatari/Llamatron.rmp.
[INFO] Remaps: no core-specific remap found at /storage/remappings/Hatari/Hatari.rmp.
[INFO] Redirecting save file to "/storage/savefiles/Llamatron.srm".
[INFO] Redirecting savestate to "/storage/savestates/Llamatron.state".
[INFO] Environ SYSTEM_DIRECTORY: "/storage/system".
[INFO] Environ CORE_ASSETS_DIRECTORY: "/storage/downloads".
Retro SYSTEM_DIRECTORY /storage/system
Retro SAVE_DIRECTORY /storage/savefiles
Retro CONTENT_DIRECTORY /storage/downloads
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Environ SET_INPUT_DESCRIPTORS:
[INFO]  RetroPad, User 1, Button "B (bottom)" => "B"
[INFO]  RetroPad, User 1, Button "Y (left)" => "Y"
[INFO]  RetroPad, User 1, Button "Select" => "Select"
[INFO]  RetroPad, User 1, Button "Start" => "Start"
[INFO]  RetroPad, User 1, Button "D-Pad Up" => "Up"
[INFO]  RetroPad, User 1, Button "D-Pad Down" => "Down"
[INFO]  RetroPad, User 1, Button "D-Pad Left" => "Left"
[INFO]  RetroPad, User 1, Button "D-Pad Right" => "Right"
[INFO]  RetroPad, User 1, Button "A (right)" => "A"
[INFO]  RetroPad, User 1, Button "X (up)" => "X"
[INFO]  RetroPad, User 1, Button "L" => "L"
[INFO]  RetroPad, User 1, Button "R" => "R"
[INFO]  RetroPad, User 1, Button "L2" => "L2"
[INFO]  RetroPad, User 1, Button "R2" => "R2"
[INFO]  RetroPad, User 1, Button "L3" => "L3"
[INFO]  RetroPad, User 1, Button "R3" => "R3"
[INFO] Environ GET_VARIABLE Hatari_resolution:
[INFO]  640x480
[libretro-test]: Got size: 640 x 480.
[INFO] Content loading skipped. Implementation will load it on its own.
[INFO] Environ SET_INPUT_DESCRIPTORS:
[INFO]  RetroPad, User 1, Button "Y (left)" => "Enter GUI"
[INFO]  RetroPad, User 1, Button "Select" => "Mouse mode toggle"
[... 108 lines cut ...]
```
[Floppy]
bAutoInsertDiskB = FALSE
FastFloppy = TRUE 
nWriteProtection = 0
szDiskAFileName =
szDiskBFileName =
szDiskImageDirectory = /storage/roms/
szDiskAZipPath =
szDiskBZipPath =

[Memory]
nMemorySize = 1

[ROM]
szCartridgeImageFileName =
szTosImageFileName = /storage/system/tos.img

[Log]
sLogFileName = /storage/system/hatari.log
sTraceFileName = /storage/system/hatari.log
nTextLogLevel = 4
```

`hatari.log` file is also created, but it stays plain empty on crash...

roms2000, 2018-11-16:
Same here with a crash each time with version Lakka 2.1.1 on RPi 3

> Nov 16 23:40:24 Lakka retroarch[498]: [libretro-test]: Got size: 640 x 480.
> Nov 16 23:40:24 Lakka retroarch[498]: Configured max Hatari resolution = 640x480, optimal for ST = 832x588
> Nov 16 23:40:24 Lakka retroarch[498]: STX : STX_Insert_internal drive=0 file=/storage/roms/Atari_ST/Cabal.stx buf=0x73a1c008 size=930048
> Nov 16 23:40:24 Lakka kernel: Alignment trap: not handling instruction f44d0a1f at [<7588ee1c>]
> Nov 16 23:40:24 Lakka kernel: Unhandled fault: alignment exception (0x801) at 0x0054f9b4
> Nov 16 23:40:24 Lakka kernel: pgd = ae080000
> Nov 16 23:40:24 Lakka kernel: [0054f9b4] *pgd=2ebbd835, *pte=25a8355f, *ppte=25a83c7e
> Nov 16 23:40:25 Lakka systemd[1]: retroarch.service: Main process exited, code=dumped, status=7/BUS
> Nov 16 23:40:25 Lakka systemd[1]: retroarch.service: Unit entered failed state.
> Nov 16 23:40:25 Lakka systemd[1]: retroarch.service: Failed with result 'core-dump'.
> Nov 16 23:40:28 Lakka systemd[1]: retroarch.service: Service hold-off time over, scheduling restart.
> Nov 16 23:40:28 Lakka systemd[1]: Stopped Retroarch.
> Nov 16 23:40:28 Lakka systemd[1]: Starting Retroarch...
> Nov 16 23:40:28 Lakka systemd[1]: Started Retroarch.

> Nov 16 23:45:35 Lakka retroarch[554]: [libretro-test]: Got size: 640 x 480.
> Nov 16 23:45:35 Lakka retroarch[554]: Configured max Hatari resolution = 640x480, optimal for ST = 832x588
> Nov 16 23:45:35 Lakka retroarch[554]: STX : STX_Insert_internal drive=0 file=/storage/roms/Atari_ST/Cabal.stx buf=0x73a1c008 size=930048
> Nov 16 23:45:35 Lakka kernel: Alignment trap: not handling instruction f44d0a1f at [<75855e1c>]
> Nov 16 23:45:35 Lakka kernel: Unhandled fault: alignment exception (0x801) at 0x010d09b4
> Nov 16 23:45:35 Lakka kernel: pgd = a78f8000
> Nov 16 23:45:35 Lakka kernel: [010d09b4] *pgd=2785b835, *pte=0f06155f, *ppte=0f061c7e
> Nov 16 23:45:36 Lakka systemd[1]: retroarch.service: Main process exited, code=dumped, status=7/BUS
> Nov 16 23:45:36 Lakka systemd[1]: retroarch.service: Unit entered failed state.
> Nov 16 23:45:36 Lakka systemd[1]: retroarch.service: Failed with result 'core-dump'.
> Nov 16 23:45:38 Lakka systemd[1]: retroarch.service: Service hold-off time over, scheduling restart.
> Nov 16 23:45:38 Lakka systemd[1]: Stopped Retroarch.
> Nov 16 23:45:38 Lakka systemd[1]: Starting Retroarch...
> Nov 16 23:45:38 Lakka systemd[1]: Started Retroarch.

I have tried Lakka 2.0.0 with a RPi Zero WH and game / desktop can run.

closed 2020-08-18 by Ntemis