libretro/Lakka-LibreELEC #115: Several cores do not support ZIP

brianblakely, 2017-04-30:
So far, I have been able to reproduce this issue with both NES and SNES content:

- Which version of Lakka are you using?
2.0

- What system hardware are you using?
RPi3

- What did you do?
Tried to load zipped content.  Selected "load archive".

- What did you expect to happen?
An appropriate core should become available for selection.

- What happened instead?
Several cores become available, but none appropriate for the content.

Kivutar, 2017-04-30:
OK, what ROM was it?

brianblakely, 2017-04-30:
Mega Man 2 (US)

Super Mario Kart (US)

no-intro

Kivutar, 2017-04-30:
You need to do Load Content->Super Mario Kart (US).zip->Browse Archive->Snes9x2010

I checked on my RPi3 and it works as intended, 4 Snes cores appear when I hit Browse Archive

brianblakely, 2017-04-30:
Browse Archive shows "no items," although there is a single SFC file in the zip.

I've conducted some tests, and this only happens when files are served off of a Windows PC via a Samba mount.  When stored locally in the Lakka device's ROMs folder, it works as expected.  Is this an issue with Samba?

Kivutar, 2017-04-30:
OK, it's the known SAMBA bug then. Sombody found that it was due to mounting samba shares in read only mode. RetroArch tries to unzip the rom and fails. You need to pass the rw option like that:

    mount -t cifs -o user=guest,rw //{IP}/{SHARE} /storage/roms/samba

You're using a systemd unit to mount?

brianblakely, 2017-05-01:
Strange, that does not resolve the issue for me.  I am not using systemd.

brianblakely, 2017-05-01:
I just gave this guide a try: http://www.lakka.tv/doc/Serving-ROMs-from-a-NAS/

Sadly, same result.  Samba-mounted Super Mario Kart zip will not reveal its contents.

Kivutar, 2017-05-01:
OK, somebody on IRC suceeded one month ago. I'm trying to find who it was to ask for advice.

Kivutar, 2017-05-01:
Also, let check this page in case you haven't yet https://github.com/libretro/Lakka-LibreELEC/wiki/Serving-ROMs-from-a-NAS

brianblakely, 2017-05-01:
Those are the same instructions.  On a side note, these instructions do not work as-written.  The first line should read `mkdir /storage/nas` or the systemd unit should be rewritten to use `/storage/roms/nas`.

Kivutar, 2017-05-01:
Thanks I fixed the docs.

RobLoach, 2017-06-26:
http://wiki.openelec.tv/index.php/Mounting_network_shares

Use a `/storage/.config/system.d/storage-retroarch.mount` file.

Kivutar, 2017-06-26:
I think he did. The problem is that even when we mount rw, retroarch can't extract the content. I don't know why.

It's worth testing setting the temporary directory to /tmp

Haris1977, 2017-11-18:
@brianblakely did y find a solution 4 non - working mame roms? I have the same prob..

brianblakely, 2017-11-20:
Nope, need to unzip the ROMs.

Haris1977, 2017-11-20:
I dont have any problem loading snes/nes files. In fact i only have problem in my mame (mame-mame4all) roms. I dont use lakka but retropie (v4.2) but i guess they use the same cores (?). I think it is a permission matter (i use my win7 64bit pc as a server) and i cant load them. I have opened a thread [here](https://retropie.org.uk/forum/topic/14469/success-in-running-roms-from-nas-but-no-mame-roms) trying to get as much help. 

I know it is a different thread here but maybe there are some similarities.

Haris1977, 2017-11-21:
Well. After some fiddling i was able to play my mame roms through samba:) What did i do? I followed [this](https://github.com/RetroPie/RetroPie-Setup/wiki/Running-ROMs-from-a-Network-Share) - Option 2 - (Option 1 didnt work for me). So it was not a mame core problem but rather a "command" error.

ToKe79, 2018-01-31:
Editing/modifying /etc/fstab is not possible with Lakka, as the root-fs is read-only (squashfs).

ToKe79, 2018-09-17:
Is this still an issue? If so, feel free to re-open/post here.

closed 2018-09-17 by ToKe79