libretro/Lakka-LibreELEC #121: Slowly blinking flower on EeePC 701

zx6802, 2017-05-04:
I'm trying to use Generic.i386-2.0. on an [EeePC 701](https://www.asus.com/de/supportonly/Eee%20PC%20701SDLinux/HelpDesk_Manual/) with an XBOX compatible wired controller. Neither "live" nor "installer" gets me past the flower screen. For some time the flower disappears and reappears than the computer alternatively seems to shutdown or to hang. It doesn't make a difference whether I us a SDHC card or an usbstick. For more information see [here](https://github.com/libretro/Lakka/issues/137#issuecomment-299297985).

gouchi, 2017-05-07:
[Issue template](https://github.com/libretro/Lakka-LibreELEC/blob/Lakka-V2.0/ISSUE_TEMPLATE.md) is not working ?

Please provide some [Lakka system and graphic card logs](http://www.lakka.tv/doc/Troubleshooting-Lakka/).

Thank you.

zx6802, 2017-05-07:
Sorry but I'm not quite sure what I'm expected to do.

Referring to your link I don't get a chance to do what is recommanded under the headline "Introduction to Lakka logs", because I don't get a commandline prompt. The screen is either black or shows the flower.

gouchi, 2017-05-07:
"The information found in [Accessing the Lakka Command Line Interface](http://www.lakka.tv/doc/Accessing-Lakka-command-line-interface) can be very helpful during this process as well. The following procedures assume that the user has logged into Lakka via SSH."

gouchi, 2017-05-07:
So you succeeded to connect by `http://www.lakka.tv/doc/Accessing-Lakka-command-line-interface/#ssh` ? If so, you can enter the command to get the log.

zx6802, 2017-05-07:
I didn't even try to connect by ssh. It says "Then, you need to enable the SSH Service in Settings->Services." which I can't do for the reason mentioned. Should I nevertheless try to connect by ssh? Anyway I'll try "live ssh" ASAP.

zx6802, 2017-05-08:
Here is, what I got

`Lakka:~ # lspci -nnk | grep -A 3 VGA

00:02.0 VGA compatible controller [0300]: Intel Corporation Mobile 915GM/GMS/910GML Express Graphics Controller [8086:2592] (rev 04)
	Subsystem: ASUSTeK Computer Inc. Mobile 915GM/GMS/910GML Express Graphics Controller [1043:82d9]
	Kernel driver in use: i915
00:02.1 Display controller [0380]: Intel Corporation Mobile 915GM/GMS/910GML Express Graphics Controller [8086:2792] (rev 04)

Lakka:~ # aplay -L

null
    Discard all samples (playback) or generate zero samples (capture)
default:CARD=Intel
    HDA Intel, ALC662 rev1 Analog
    Default Audio Device
sysdefault:CARD=Intel
    HDA Intel, ALC662 rev1 Analog
    Default Audio Device
front:CARD=Intel,DEV=0
    HDA Intel, ALC662 rev1 Analog
    Front speakers
surround21:CARD=Intel,DEV=0
    HDA Intel, ALC662 rev1 Analog
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=Intel,DEV=0
    HDA Intel, ALC662 rev1 Analog
    4.0 Surround output to Front and Rear speakers
surround41:CARD=Intel,DEV=0
    HDA Intel, ALC662 rev1 Analog
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=Intel,DEV=0
    HDA Intel, ALC662 rev1 Analog
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=Intel,DEV=0
    HDA Intel, ALC662 rev1 Analog
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=Intel,DEV=0
    HDA Intel, ALC662 rev1 Analog
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
Lakka:~ # lsusb
Bus 001 Device 003: ID 0951:1606 Kingston Technology Eee PC 701 SD Card Reader [ENE UB6225]
Bus 001 Device 002: ID 090c:1000 Silicon Motion, Inc. - Taiwan (formerly Feiya Technology Corp.) Flash Drive
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 002: ID 045e:028e Microsoft Corp. Xbox360 Controller
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub`

And this is the log-file

`RetroArch [INFO] :: This is RetroArch version 1.5.0 (Git 2a13af5)
RetroArch [INFO] :: === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 
Built: Apr 24 2017
RetroArch [INFO] :: Version: 1.5.0
RetroArch [INFO] :: Git: 2a13af5
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Loading default config.
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Redirecting save file to "/storage/savefiles/.srm".
RetroArch [INFO] :: Redirecting savestate to "/storage/savestates/.state".
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29970.03 Hz.
RetroArch [INFO] :: Found shader "/tmp/shaders/bilinear.glslp"
RetroArch [INFO] :: Found shader "/tmp/shaders/nearest.glslp"
RetroArch [INFO] :: Found shader "/tmp/shaders/stock.glsl"
[... 49 lines cut ...]
RetroArch [INFO] :: Linker log: warning: Couldn't flatten if-statement.  This will likely result in software rasterization.

RetroArch [INFO] :: [GL]: Using 4 textures.
RetroArch [INFO] :: [GL]: Loaded 1 program(s).
RetroArch [INFO] :: [GL]: Using GL_RGB565 for texture uploads.
RetroArch [INFO] :: Using font rendering backend: freetype.
RetroArch [INFO] :: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
RetroArch [INFO] :: [udev] Adding device /dev/input/event5 as type ID_INPUT_KEYBOARD.
RetroArch [INFO] :: [udev] Adding device /dev/input/event6 as type ID_INPUT_TOUCHPAD.
RetroArch [INFO] :: [udev] Adding device /dev/input/mouse0 as type ID_INPUT_TOUCHPAD.
RetroArch [ERROR] :: [udev] Failed to open device: /dev/input/mouse0 (Inappropriate ioctl for device).
RetroArch [INFO] :: Found joypad driver: "udev".
RetroArch [INFO] :: ALSA: Using signed 16-bit format.
RetroArch [INFO] :: ALSA: Period size: 1024 frames
RetroArch [INFO] :: ALSA: Buffer size: 2048 frames
RetroArch [INFO] :: Found menu display driver: "menu_display_gl".
RetroArch [INFO] :: Using font rendering backend: freetype.
RetroArch [INFO] :: Using font rendering backend: freetype.
function is no-op
RetroArch [INFO] :: SRAM will not be saved.
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_history.lpl].
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_music_history.lpl].
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_video_history.lpl].
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_image_history.lpl].
intel_do_flush_locked failed: Invalid argument`

zx6802, 2017-05-09:
I also tried an [EeePC 900A](https://www.asus.com/support/Download/30/20/0/33/5/) 1,6 GHz Intel Atom N270, 1GB RAM. Sometimes the screen blinks, sometimes it freezes. In the latter case case the specs are

`Lakka:~ # lspci -nnk | grep -A 3 VGA
00:02.0 VGA compatible controller [0300]: Intel Corporation Mobile 945GSE Express Integrated Graphics Controller [8086:27ae] (rev 03)
	Subsystem: ASUSTeK Computer Inc. Mobile 945GSE Express Integrated Graphics Controller [1043:8340]
	Kernel driver in use: i915
00:02.1 Display controller [0380]: Intel Corporation Mobile 945GM/GMS/GME, 943/940GML Express Integrated Graphics Controller [8086:27a6] (rev 03)` 

`Lakka:~ # aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
default:CARD=Intel
    HDA Intel, ALC269 Analog
    Default Audio Device
sysdefault:CARD=Intel
    HDA Intel, ALC269 Analog
    Default Audio Device
front:CARD=Intel,DEV=0
    HDA Intel, ALC269 Analog
    Front speakers
surround21:CARD=Intel,DEV=0
    HDA Intel, ALC269 Analog
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=Intel,DEV=0
    HDA Intel, ALC269 Analog
    4.0 Surround output to Front and Rear speakers
surround41:CARD=Intel,DEV=0
    HDA Intel, ALC269 Analog
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=Intel,DEV=0
    HDA Intel, ALC269 Analog
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=Intel,DEV=0
    HDA Intel, ALC269 Analog
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=Intel,DEV=0
    HDA Intel, ALC269 Analog
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
`
`Lakka:~ # cat log.txt 
RetroArch [INFO] :: This is RetroArch version 1.5.0 (Git 2a13af5)
RetroArch [INFO] :: === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 
Built: Apr 24 2017
RetroArch [INFO] :: Version: 1.5.0
RetroArch [INFO] :: Git: 2a13af5
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Loading default config.
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Redirecting save file to "/storage/savefiles/.srm".
RetroArch [INFO] :: Redirecting savestate to "/storage/savestates/.state".
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29970.03 Hz.
RetroArch [INFO] :: Found shader "/tmp/shaders/bilinear.glslp"
RetroArch [INFO] :: Found shader "/tmp/shaders/nearest.glslp"
RetroArch [INFO] :: Found shader "/tmp/shaders/stock.glsl"
RetroArch [INFO] :: Video @ fullscreen
RetroArch [INFO] :: Starting threaded video driver ...
RetroArch [INFO] :: [DRM]: Found 2 connectors.
RetroArch [INFO] :: [DRM]: Connector 0 connected: yes
RetroArch [INFO] :: [DRM]: Connector 0 has 1 modes.
RetroArch [INFO] :: [DRM]: Connector 0 assigned to monitor index: #1.
RetroArch [INFO] :: [DRM]: Connector 1 connected: no
RetroArch [INFO] :: [DRM]: Connector 1 has 0 modes.
RetroArch [INFO] :: [DRM]: Mode 0: (1024x600) 1024 x 600, 60 Hz
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
RetroArch [INFO] :: Found GL context: kms
RetroArch [INFO] :: Detecting screen resolution 1024x600.
RetroArch [INFO] :: [EGL]: EGL version: 1.4
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
RetroArch [INFO] :: [EGL]: Current context: 0xb4507440.
RetroArch [INFO] :: [KMS]: New FB: 1024x600 (stride: 4096).
Segmentation fault (core dumped)
`
`Lakka:~ # lspci
00:00.0 Host bridge: Intel Corporation Mobile 945GSE Express Memory Controller Hub (rev 03)
00:02.0 VGA compatible controller: Intel Corporation Mobile 945GSE Express Integrated Graphics Controller (rev 03)
00:02.1 Display controller: Intel Corporation Mobile 945GM/GMS/GME, 943/940GML Express Integrated Graphics Controller (rev 03)
00:1b.0 Audio device: Intel Corporation NM10/ICH7 Family High Definition Audio Controller (rev 02)
00:1c.0 PCI bridge: Intel Corporation NM10/ICH7 Family PCI Express Port 1 (rev 02)
00:1c.1 PCI bridge: Intel Corporation NM10/ICH7 Family PCI Express Port 2 (rev 02)
00:1c.3 PCI bridge: Intel Corporation NM10/ICH7 Family PCI Express Port 4 (rev 02)
00:1d.0 USB controller: Intel Corporation NM10/ICH7 Family USB UHCI Controller (rev 02)
00:1d.1 USB controller: Intel Corporation NM10/ICH7 Family USB UHCI Controller (rev 02)
00:1d.2 USB controller: Intel Corporation NM10/ICH7 Family USB UHCI Controller (rev 02)
00:1d.3 USB controller: Intel Corporation NM10/ICH7 Family USB UHCI Controller (rev 02)
00:1d.7 USB controller: Intel Corporation NM10/ICH7 Family USB2 EHCI Controller (rev 02)
00:1e.0 PCI bridge: Intel Corporation 82801 Mobile PCI Bridge (rev e2)
00:1f.0 ISA bridge: Intel Corporation 82801GBM (ICH7-M) LPC Interface Bridge (rev 02)
00:1f.2 IDE interface: Intel Corporation 82801GBM/GHM (ICH7-M Family) SATA Controller [IDE mode] (rev 02)
01:00.0 Ethernet controller: Qualcomm Atheros AR8121/AR8113/AR8114 Gigabit or Fast Ethernet (rev b0)
02:00.0 Ethernet controller: Qualcomm Atheros AR242x / AR542x Wireless Network Adapter (PCI-Express) (rev 01)
`

gouchi, 2017-05-09:
For the EeePC 701, can you try to set in the [configuration file](http://www.lakka.tv/doc/Configuring-Lakka/) video_monitor_index = "1". Please check [nano editor shortcuts](https://gist.github.com/franz-josef-kaiser/43c045190356b0d5778c).

Unfortunately, for the second one you get crash of RA :(

zx6802, 2017-05-09:
Well I'm aiming for the 701 anyway. There with this .config/retroarch/retroarch.cfg 
`savefile_directory = "/storage/savefiles"
savestate_directory = "/storage/savestates"
input_remapping_directory = "/storage/remappings"
playlist_directory = "/storage/playlists"
libretro_directory = "/tmp/cores"
libretro_info_path = "/tmp/cores"
system_directory = "/storage/system"
rgui_browser_directory = "/storage/roms"
assets_directory = "/tmp/assets"
thumbnails_directory = "/storage/thumbnails"
rgui_show_start_screen = "false"
video_monitor_index = "1"
video_fullscreen = "true"
video_windowed_fullscreen = "false"
video_threaded = "true"
video_smooth = "false"
video_aspect_ratio_auto = "true"
video_shader_dir = "/tmp/shaders"
video_filter_dir = "/usr/share/video_filters"
video_font_path = "/usr/share/retroarch-assets/xmb/monochrome/font.ttf"
audio_driver = "alsathread"
audio_filter_dir = "/usr/share/audio_filters"
input_driver = "udev"
input_max_users = "5"
input_autodetect_enable = "true"
joypad_autoconfig_dir = "/tmp/joypads"
input_menu_toggle_gamepad_combo = "2"
all_users_control_menu = "true"
menu_driver = "xmb"
menu_mouse_enable = "false"
menu_core_enable = "false"
content_database_path = "/tmp/database/rdb"
cheat_database_path = "/tmp/database/cht"
screenshot_directory = "/storage/screenshots"
video_gpu_screenshot = "false"
core_assets_directory = "/storage/roms/downloads"
savestate_thumbnail_enable = "false"
menu_show_advanced_settings = "false"
menu_wallpaper_opacity = "1.0"
xmb_show_images = "false"
xmb_show_music = "false"
xmb_show_video = "false"
playlist_names = "Atari - 2600.lpl;Atari - 5200.lpl;Atari - 7800.lpl;Atari - Jaguar.lpl;Atari - Lynx.lpl;Atari - ST.lpl;Bandai - WonderSwan Color.lpl;Bandai - WonderSwan.lpl;Cave Story.lpl;Coleco - ColecoVision.lpl;Dinothawr.lpl;DOOM.lpl;DOS.lpl;FB Alpha - Arcade Games.lpl;GCE - Vectrex.lpl;Lutro.lpl;Magnavox - Odyssey2.lpl;MAME.lpl;Microsoft - MSX2.lpl;Microsoft - MSX.lpl;NEC - PC Engine SuperGrafx.lpl;NEC - PC Engine - TurboGrafx 16.lpl;Nintendo - Famicom Disk System.lpl;Nintendo - Game Boy Advance (e-Cards).lpl;Nintendo - Game Boy Advance.lpl;Nintendo - Game Boy Color.lpl;Nintendo - Game Boy.lpl;Nintendo - Nintendo 64.lpl;Nintendo - Nintendo DS Decrypted.lpl;Nintendo - Nintendo DS (Download Play) (BETA).lpl;Nintendo - Nintendo DSi Decrypted.lpl;Nintendo - Nintendo DSi (DLC).lpl;Nintendo - Nintendo Entertainment System.lpl;Nintendo - Satellaview.lpl;Nintendo - Super Nintendo Entertainment System.lpl;Nintendo - Virtual Boy.lpl;Quake1.lpl;ScummVM.lpl;Sega - 32X.lpl;Sega - Game Gear.lpl;Sega - Master System - Mark III.lpl;Sega - Mega Drive - Genesis.lpl;Sega - PICO.lpl;Sega - SG-1000.lpl;Sinclair - ZX Spectrum +3.lpl;SNK - Neo Geo Pocket Color.lpl;SNK - Neo Geo Pocket.lpl;Sony - PlayStation Portable.lpl;Sony - PlayStation.lpl"
playlist_cores = "/tmp/cores/stella_libretro.so;DETECT;/tmp/cores/prosystem_libretro.so;/tmp/cores/virtualjaguar_libretro.so;/tmp/cores/handy_libretro.so;/tmp/cores/hatari_libretro.so;/tmp/cores/mednafen_wswan_libretro.so;/tmp/cores/mednafen_wswan_libretro.so;/tmp/cores/nxengine_libretro.so;/tmp/cores/bluemsx_libretro.so;/tmp/cores/dinothawr_libretro.so;/tmp/cores/prboom_libretro.so;/tmp/cores/dosbox_libretro.so;/tmp/cores/fbalpha_libretro.so;/tmp/cores/vecx_libretro.so;/tmp/cores/lutro_libretro.so;/tmp/cores/o2em_libretro.so;/tmp/cores/mame_libretro.so;/tmp/cores/bluemsx_libretro.so;/tmp/cores/bluemsx_libretro.so;/tmp/cores/mednafen_supergrafx_libretro.so;/tmp/cores/mednafen_pce_fast_libretro.so;/tmp/cores/nestopia_libretro.so;/tmp/cores/mgba_libretro.so;/tmp/cores/mgba_libretro.so;/tmp/cores/gambatte_libretro.so;/tmp/cores/gambatte_libretro.so;/tmp/cores/parallel_n64_libretro.so;/tmp/cores/desmume_libretro.so;/tmp/cores/desmume_libretro.so;/tmp/cores/desmume_libretro.so;/tmp/cores/desmume_libretro.so;/tmp/cores/nestopia_libretro.so;/tmp/cores/snes9x_libretro.so;/tmp/cores/snes9x_libretro.so;/tmp/cores/mednafen_vb_libretro.so;/tmp/cores/tyrquake_libretro.so;/tmp/cores/scummvm_libretro.so;/tmp/cores/picodrive_libretro.so;/tmp/cores/genesis_plus_gx_libretro.so;/tmp/cores/genesis_plus_gx_libretro.so;/tmp/cores/genesis_plus_gx_libretro.so;/tmp/cores/picodrive_libretro.so;/tmp/cores/genesis_plus_gx_libretro.so;/tmp/cores/fuse_libretro.so;/tmp/cores/mednafen_ngp_libretro.so;/tmp/cores/mednafen_ngp_libretro.so;/tmp/cores/ppsspp_libretro.so;/tmp/cores/mednafen_psx_hw_libretro.so"
config_save_on_exit = "true"
`
the blinking berry persist and I get this 
`Lakka:~ # retroarch
intel_do_flush_locked failed: Invalid argument`

zx6802, 2017-05-09:
It seems that the contents of retroarch.cfg does not survive a reboot. Also if I change the file by mounting the usbdrive on another computer, I get the unchanged one after booting. Also it is not possible to soft-reboot Lakka with a shutdown -r now

zx6802, 2017-05-10:
Rechecked LAKKA_DISK/.config/retroarch/retroarch.cfg outside of a running Lakka and it now seems to permanentely contain the line video_monitor_index = "1".

Kivutar, 2017-05-11:
You need to stop retroarch if you want to edit the retroarch.cfg in SSH:

    systemctl stop retroarch
    nano .config/retroarch/retroarch.cfg
    systemctl start retroarch

zx6802, 2017-05-11:
Well, as far as I understand, there is no way to "stop retroarch" if I edit the file on another system, which was the first thing I tried. Anyway the above line seems no firmly planted in retroarch.cfg.

Kivutar, 2017-05-12:
I see that you get a segfault when launching retroarch. Can you try to get a backtrace?

You need to do like this:

    systemctl stop retroarch
    gdb retroarch

then type **r** and after the crash, type **bt**

Make a photo of the output and paste it here.

zx6802, 2017-05-12:
Here's what I get
`Lakka (community) Version: 2.0
Lakka:~ # systemctl stop retroarch
Lakka:~ # gdb retroarch
GNU gdb (GDB) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "i686-libreelec-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from retroarch...(no debugging symbols found)...done.
(gdb) r
Starting program: /usr/bin/retroarch 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
[New Thread 0xb5780b40 (LWP 826)]
[New Thread 0xb4ed8b40 (LWP 827)]
[New Thread 0xb1cfeb40 (LWP 828)]
intel_do_flush_locked failed: Invalid argument
[Thread 0xb1cfeb40 (LWP 828) exited]
[Thread 0xb5780b40 (LWP 826) exited]
[Thread 0xb58ae780 (LWP 822) exited]
[Inferior 1 (process 822) exited with code 01]
(gdb) bt
No stack.
(gdb) 
`

Kivutar, 2017-05-12:
Mhhh I don't see the segfault anymore!

zx6802, 2017-05-13:
This is what I get from the 900A
`Lakka:~ # systemctl stop retroarch
Lakka:~ # gdb retroarch
GNU gdb (GDB) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "i686-libreelec-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from retroarch...(no debugging symbols found)...done.
(gdb) r
Starting program: /usr/bin/retroarch 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
[New Thread 0xb56deb40 (LWP 559)]
[New Thread 0xb4e36b40 (LWP 560)]
[New Thread 0xb1cfeb40 (LWP 561)]

Thread 3 "retroarch" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0xb4e36b40 (LWP 560)]
0xb0500449 in ?? ()
(gdb) bt
#0  0xb0500449 in ?? ()
Backtrace stopped: Cannot access memory at address 0xec
(gdb)`

ltrvs, 2017-05-30:
I have an EeePC 900 version, different CPU (Intel(R) Celeron(R) M processor 900MHz) than in 900A, otherwise similar specs I assume.

Using Lakka 2.0 I ended up with the similar results as  zx6802's 900A: 
> Segmentation fault (core dumped)

Using Lakka-Generic.i386-2.1-rc1 I got the on my 900 the same as zx6802 got on the 701:
> intel_do_flush_locked failed: Invalid argument

Please let me know if I can help by providing logs or something else.

gouchi, 2017-05-30:
Thank you for the feedback.

Is it possible to [edit the configuration file](http://www.lakka.tv/doc/Configuring-Lakka/#using-the-command-line-interface) and put

`menu_shader_pipeline = "0"`

Thank you.

ltrvs, 2017-05-31:
Setting `menu_shader_pipeline = "0"` doesnt't help. But with it I can see XMB screen to flash for about a second, then it fails. Both outputs below are from Lakka-Generic.i386-2.1-rc1 with `menu_shader_pipeline = "0":

GDB attempt:

`Lakka:~ # gdb retroarch
GNU gdb (GDB) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "i686-libreelec-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from retroarch...(no debugging symbols found)...done.
(gdb) r
Starting program: /usr/bin/retroarch 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Redirecting save file to "/storage/savefiles/.srm".
RetroArch [INFO] :: Redirecting savestate to "/storage/savestates/.state".
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: [Audio]: Set audio input rate to: 29970.03 Hz.
RetroArch [INFO] :: Found shader "/tmp/shaders/slang-conversion-template.glsl"
RetroArch [INFO] :: Found shader "/tmp/shaders/stock.glsl"
RetroArch [INFO] :: [Video]: Video @ fullscreen
RetroArch [INFO] :: [Video]: Starting threaded video driver ...
[New Thread 0xb5732b40 (LWP 570)]
RetroArch [INFO] :: [DRM]: Found 3 connectors.
RetroArch [INFO] :: [DRM]: Connector 0 connected: yes
RetroArch [INFO] :: [DRM]: Connector 0 has 1 modes.
RetroArch [INFO] :: [DRM]: Connector 0 assigned to monitor index: #1.
RetroArch [INFO] :: [DRM]: Connector 1 connected: no
RetroArch [INFO] :: [DRM]: Connector 1 has 0 modes.
RetroArch [INFO] :: [DRM]: Connector 2 connected: no
RetroArch [INFO] :: [DRM]: Connector 2 has 0 modes.
RetroArch [INFO] :: [DRM]: Mode 0: (1024x600) 1024 x 600, 60 Hz
RetroArch [INFO] :: [GL]: Found GL context: kms
RetroArch [INFO] :: [GL]: Detecting screen resolution 1024x600.
RetroArch [INFO] :: [EGL] Found EGL_EXT_platform_base, trying eglGetPlatformDisplayEXT
RetroArch [INFO] :: [EGL]: EGL version: 1.4
RetroArch [INFO] :: [EGL]: Current context: 0xb4e07528.
RetroArch [INFO] :: [KMS]: New FB: 1024x600 (stride: 4096).
RetroArch [INFO] :: [GL]: Vendor: Intel Open Source Technology Center, Renderer: Mesa DRI Intel(R) 915GM x86/MMX/SSE2.
RetroArch [INFO] :: [GL]: Version: 2.1 Mesa 13.0.6.
RetroArch [INFO] :: [GL]: Using resolution 1024x600
RetroArch [INFO] :: [GL]: Default shader backend found: glsl.
RetroArch [INFO] :: [Shader driver]: Using GLSL shader backend.
RetroArch [INFO] :: [GLSL]: Checking GLSL shader support ...
RetroArch [WARN] :: [GL]: Stock GLSL shaders will be used.
RetroArch [INFO] :: [GLSL]: Found GLSL vertex shader.
RetroArch [INFO] :: [GLSL]: Found GLSL fragment shader.
RetroArch [INFO] :: [GLSL]: Linking GLSL program.
RetroArch [INFO] :: [GLSL]: Found GLSL vertex shader.
RetroArch [INFO] :: [GLSL]: Found GLSL fragment shader.
RetroArch [INFO] :: [GLSL]: Linking GLSL program.
RetroArch [INFO] :: [GLSL]: Found GLSL vertex shader.
RetroArch [INFO] :: [GLSL]: Found GLSL fragment shader.
RetroArch [INFO] :: [GLSL]: Linking GLSL program.
RetroArch [INFO] :: [GLSL]: Found GLSL vertex shader.
RetroArch [INFO] :: [GLSL]: Found GLSL fragment shader.
RetroArch [INFO] :: [GLSL]: Linking GLSL program.
RetroArch [INFO] :: [GLSL]: Found GLSL vertex shader.
[... 26 lines cut ...]
RetroArch [INFO] :: [Menu]: Found menu display driver: "menu_display_gl".
RetroArch [INFO] :: [Font]: Using font rendering backend: freetype.
RetroArch [INFO] :: [Font]: Using font rendering backend: freetype.
RetroArch [WARN] :: Input device ID 5 is unknown to this libretro implementation. Using RETRO_DEVICE_JOYPAD.
RetroArch [INFO] :: Disconnecting device from port 2.
RetroArch [INFO] :: SRAM will not be saved.
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_history.lpl].
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_music_history.lpl].
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_video_history.lpl].
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_image_history.lpl].
RetroArch [INFO] :: [KMS]: New FB: 1024x600 (stride: 4096).
RetroArch [INFO] :: [KMS]: New FB: 1024x600 (stride: 4096).
intel_do_flush_locked failed: No such file or directory
[Thread 0xb297db40 (LWP 571) exited]
[Thread 0xb57e0780 (LWP 566) exited]
[Inferior 1 (process 566) exited with code 01]
(gdb) bt
No stack.
(gdb) 
No stack.
(gdb) q`

Guessing some debug flags from https://www.mesa3d.org/envvars.html:
`INTEL_DEBUG=dri,bat MESA_DEBUG=1  LIBGL_DEBUG=verbose retroarch --menu --verbose > pipeline0.log.txt 2>&1`
[pipeline0.log.txt](https://github.com/libretro/Lakka-LibreELEC/files/1042110/pipeline0.log.txt)

ltrvs, 2017-06-01:
Got something with gdb (EeePc 900, Lakka-Generic.i386-2.1-rc1, menu_shader_pipeline = "2"), are there any clues in this?
[gdb.txt](https://github.com/libretro/Lakka-LibreELEC/files/1045368/gdb.txt)

gouchi, 2017-06-03:
There is no debugging symbol. You should try to compile RA on Linux distro with
./configure && DEBUG=1 make

Switch to console (CTRL+ALT+F1), launch RA and get some gdb bt.

Thank you.

ltrvs, 2017-06-03:
Ok, thanks for the advice. I've done a build env successfully (http://www.lakka.tv/doc/Compiling-Lakka/). I think it's easier for me to do a RA debug build there than set up RA for another distro.

Any tips for Lakka build newbie? Just clean RA and 'DEBUG=1 DISTRO=Lakka PROJECT=Generic ARCH=i386 make image'?

Thanks :-)

gouchi, 2017-06-03:
Oh I see cool, yes it should work. If  not you can force it in [RA package](http://le.builds.lakka.tv/Generic.x86_64/Lakka-Generic.x86_64-2.1-rc1.img.gz).

ltrvs, 2017-06-03:
How about this?

> Thread 2 "retroarch" received signal SIGSEGV, Segmentation fault.
> [Switching to Thread 0xb578bb40 (LWP 558)]
> 0xb4f77121 in ?? () from /usr/lib/libdrm_intel.so.1
> (gdb) bt
> *0  0xb4f77121 in ?? () from /usr/lib/libdrm_intel.so.1
> *1  0xb4f7bfcf in ?? () from /usr/lib/libdrm_intel.so.1
> *2  0xb4f7c0b2 in ?? () from /usr/lib/libdrm_intel.so.1
> *3  0xb48c5b74 in ?? () from /usr/lib/dri/i915_dri.so
> *4  0xb48d2f22 in ?? () from /usr/lib/dri/i915_dri.so
> *5  0xb72fc51b in ?? () from /usr/lib/libEGL.so.1
> *6  0xb72f8ddf in ?? () from /usr/lib/libEGL.so.1
> *7  0xb72fd6ce in eglSwapBuffers () from /usr/lib/libEGL.so.1
> *8  0x081b5313 in egl_swap_buffers (data=0xb4e01068) at gfx/common/egl_common.c:171
> *9  0x0819ff42 in gfx_ctx_drm_swap_buffers (data=0xb4e01068, data2=0xb578b024) at gfx/drivers_context/drm_ctx.c:242
> *10 0x081a4515 in gl_frame (data=0xb4e00470, frame=0xb578c008, frame_width=4, frame_height=4, frame_count=0, pitch=8, msg=0x0, video_info=0xb578b024) at gfx/drivers/gl.c:1304
> *11 0x08190f11 in video_thread_loop (data=0xa0079f0) at gfx/video_thread_wrapper.c:626
> *12 0x0818f8d4 in thread_wrap (data_=0xa007e28) at libretro-common/rthreads/rthreads.c:142
> *13 0xb756f606 in start_thread (arg=0xb578bb40) at pthread_create.c:333
> *14 0xb5cd2a9e in clone () at ../sysdeps/unix/sysv/linux/i386/clone.S:113
> (gdb)

Edited a bit (thanks to Markdown), full output here:
[gdb.txt](https://github.com/libretro/Lakka-LibreELEC/files/1049837/gdb.txt)

Damn, it wasn't so easy, all of my several attempts involving  `make image` led to missing symbols. This is how I finally got the symbols there, hope it's OK.

1. `DISTRO=Lakka PROJECT=Generic ARCH=i386 scripts/clean retroarch`
2. `DEBUG=1 DISTRO=Lakka PROJECT=Generic ARCH=i386 scripts/build retroarch`
3. Copied _Lakka-LibreELEC/build.Lakka-Generic.i386-8.0-devel/retroarch-882fe09/_ directory to the target's _/storage_
4. On Lakka target: `cd /storage/retroarch-882fe09`
5. On Lakka target: `gdb retroarch`

gouchi, 2017-06-04:
Is it possible to try with this configuration
```
menu_shader_pipeline = "0"
video_threaded = "false"
```
Thank you.

ltrvs, 2017-06-04:
Sure.

With `menu_shader_pipeline = "0"` + `video_threaded = "false"` I got RetroArch running: It displayed correctly for under a second, then distorded (yellow and black, texts unreadable) but kept running, then crashed when I pressed arrows on keyboard. Console output ended with
> Segmentation fault (core dumped)

A gdb session ends with SIGABORT, in case it makes a difference.
[gdb.txt](https://github.com/libretro/Lakka-LibreELEC/files/1050497/gdb.txt)

gouchi, 2017-06-25:
Thank you for the log, can you check if you reproduce with [Lakka 2.1 rc2](http://le.builds.lakka.tv/Generic.i386/Lakka-Generic.i386-2.1-rc2.img.gz) with  `menu_shader_pipeline = "0"` + `video_threaded = "false" `.

Thank you.

Oredukri, 2017-07-01:
I have the exact same issue with an [Advent AIO100 ](http://www.uktsupport.co.uk/advent/pc/AIO100.htm)  which is quite similar hardware-wise to an eepc .  

The problem is new to Lakka 2.0 since I've been using this same machine with previous versions of Lakka without any issue.  (I've also tested lakka 2.1 rc2 and rc3 without solving the problem). 

Installing a previous version might help you, zx6802. The issue might be finding an old .img, I lost mine and I can't find any online.

zx6802, 2017-07-02:
@Oredukri: Thanks. Time's a bit short here at the moment, so I don't know when I can come back to it, but it would be nice, to have a workaround.

Oredukri, 2017-07-02:
The only official old build I could find was here [nightly](http://sources.lakka.tv/nightly/Generic.i386/).
I've just tested it and my advent works again.  I tested nes, snes, gbc, gba, mega drive and master system.  

A heads up though. The first time you boot the system it will go slow, extremely slow, since our netbooks don't support a menu option that is enabled by default. You can fix it the tech savvy way or the lazy way, I'll explain the second one.  
Once you boot you'll find yourself on the menu, press and keep pressed right (it'll take a couple of seconds to change), until you end up at options tab; press and keep pressed down until you highlight "menu" press enter until you get menu options; get to "menu shader pipeline" and press left until it says "off".  All that process might take a minute but once menu shader pipeline goes from ribbon to off you'll have a fully responsive lakka distro. 

I hope it helps you as it helped me!

Rombusevil, 2017-07-02:
Thanks a lot @Oredukri !
I tested the image you linked and the instructions on how to make it fast and it works great on a thinkpad R60 and also on a Asus EEEPC 701. 
Haven't tested any games yet, but the base system works great.

ltrvs, 2017-07-04:
@gouchi : using [Lakka 2.1 rc3](http://le.builds.lakka.tv/Generic.i386/Lakka-Generic.i386-2.1-rc3.tar) and `menu_shader_pipeline = "0" + video_threaded = "false" + xmb_shadows_enable = "false"` the result is the same as earlier:

> I got RetroArch running: It displayed correctly for under a second, then distorded (yellow and black, texts unreadable) but kept running, then crashed when I pressed arrows on keyboard

Output from plain `retroarch` command: 
[2.1-rc3-ra.txt](https://github.com/libretro/Lakka-LibreELEC/files/1122874/2.1-rc3-ra.txt)

And for `LIBGL_DEBUG=verbose retroarch --menu --verbose` as it seems a bit different:
[2.1-rc3-ra-menu-verbose.txt](https://github.com/libretro/Lakka-LibreELEC/files/1122876/2.1-rc3-ra-menu-verbose.txt)

I extracted the KERNEL and SYSTEM files over the existing ones on my SD card. Please let me know if i should use RC2 and another upgrade method for some reason. I'd love to help you with more logs, but my resources timewise for this hobby are kind of limited... Feel free to ask for them, I'll deliver when possible :-)

@Oredukri : Do you happen to know which build is the first failing build after the working nightly you've found? Just wondering how many changes there might between them, and if the developers might spot a change that could be the reason for this issue.

**@All** Please note that you can use also recent builds if you change the menu driver to something else than XMB. If you're not comfortable editing the configuration file, I assume changing the menu driver setting with a working nightly and then upgrading e.g. to a recent RC3 does the trick too.

markwkidd, 2017-07-10:
Could this issue perhaps be more accurately titled: "Regression in Lakka 2.x support for EeePC 701 and EeePC 900A?"

I say this because I have just had the same experience with an EeePC 900A.

I can get into a slow, but functional XMB when I use the LakkaOE nightly labeled: Lakka-Generic.i386-devel-20161221183902-r21729-g274b40f-efi

When I use the Lakka 2.1 rc3 I make it to the Lakka logo but no farther. Every few sections the logo disappears and reappears. It looks a lot like how a screen responds when it is changing mode/resolution. (There are no evident changes in resolution though.) The image I'm using is Lakka-Generic.x86_64-2.1-rc3

I am having a difficult time accessing logs on this installation so far, but once I figure out how best to get to a commandline I will try to get SSH up and running which it doesn't seem to be at the moment.

Oredukri, 2017-07-13:
@ltrvs  Sorry for the delayed answer, it's been crazy here so I've been out of the computer for a while. 

I can't remember the exact build it was since I updated it a few times and then forgot about it for about 4 months, until last week I tried to update and it constantly restarted the download at 15% , so I ended up testing lakka 2.x, sorry. 

If I could access the isos of the last 1.9 versions I'd be willing to test if they work fine. 

I'm glad it worked for you too @Rombusevil

ltrvs, 2017-07-22:
FYI:
https://github.com/libretro/RetroArch/issues/5188
There's discussion about about not supported HW/driver GL versions.

Could this be the root cause also for this issue?

gouchi, 2017-12-16:
It seems related if your card provide OpenGL up to 1.4

You may try to make a test with [Lakka 2.1](http://le.builds.lakka.tv/Generic.i386/Lakka-Generic.i386-2.1.img.gz).

But I am afraid you will have to switch to rgui menu driver.

markwkidd, 2018-02-08:
@gouchi and @zx6802 Lakka 2.1 is working perfectly on the Asus EeePC 900A. XMB is functional as long as the ribbon is disabled.

gouchi, 2018-02-08:
@zx6802 : Try to edit your [configuration file](http://www.lakka.tv/doc/Configuring-Lakka) with `menu_shader_pipeline = "0"`

@markwkidd : Thank you for the update.

zx6802, 2018-02-08:
Thanks for the news and thanks to everyone who contributed. I'll look into it ASAP.

gouchi, 2018-08-01:
@zx6802 any news ?

zx6802, 2018-08-01:
Sorry, ASAP turned out to be much longer than expected. Thanks for your continuing interest. I'm not at my machine right now, but will come back to it and report ... uuhm ... ASAP.

markwkidd, 2018-11-10:
I've just installed successfully on an Asus Eee PC 1005HAB that has this same issue starting with the default live installer.

At the boot prompt I used `live textmode` to get to a console where I can then stop the retroarch service and use `nano` to add `menu_shader_pipeline = "0"` to `retroarch.cfg`. At that point things are working well.

@gouchi I'm pretty confident you can close this issue as solved.

closed 2018-11-10 by kivutar

rasb23, 2020-04-23:
Confirmed this fixed the same issue on Acer AspireONE