libretro/Lakka-LibreELEC #122: Neo Geo/FB Alpha bugs after switch from OpenElec to LibreElec

x2k13, 2017-05-06:
- Which version of Lakka are you using?
Lakka-RPi.arm-2.0
- What system hardware are you using?
Rpi.arm
- What did you do?
Run FB Alpha Neo Geo roms
- What did you expect to happen?
Fluid game performance like it used to be in Lakka-RPi.arm-devel-20161221201618-r21729-g274b40f
- What happened instead?
Running Neo Geo roms results in some parts of ingame sound/music not playing, e.g. Blazing Star (blazstar.zip) and stuttering (VSync disabled)

Kivutar, 2017-05-11:
OK, I was able to reproduce the bug. I'm going to check what happenned.

Kivutar, 2017-05-11:
I just tried the old fbalpha core from Lakka OpenELEC and I get the same result: under 40 FPS for blazstar.zip

How much do you get on Lakka OpenELEC?

x2k13, 2017-05-12:
Hey @Kivutar , thanks for looking into this.
I just checked the FPS on OpenELEC and I get ~ 60 (beginning of level, fewer sprites) to ~ 52 (level end-boss, lots of sprites). 

Using FBA_v0.2.97.39 reference set, btw.

Kivutar, 2017-05-12:
I booted Lakka-OE from december and got 40FPS with blazstar.zip

x2k13, 2017-05-12:
I forgot to mention that I overclocked the PI1:

arm_freq=1000
core_freq=500
sdram_freq=500
over_voltage=6

Maybe that explains the difference...

Kivutar, 2017-05-12:
That can clearly explain the difference. And you don't get the same result with Lakka 2.0 overclocked?

x2k13, 2017-05-13:
No, definateley not. I always overclock the PI before booting up anything new. I assume that without overclocking Lakka 2.0 would perform even worse (haven't tested that, though).

Kivutar, 2017-05-13:
I tried again and I notice no difference between the two versions, it's always around 40FPS with audio crackling for me.

x2k13, 2017-05-13:
Which RPI are you using? Mine is a model B (512 MB RAM). I get no crackling at all on OE, maybe just a little sound-lag on heavy load, LE version is missing parts of the sound completely (e.g. intro music) and notably slower, unfortunately.

x2k13, 2017-05-22:
@Kivutar
No pushing intended, but is there any progress on this issue? Thanks.

Kivutar, 2017-06-13:
There is progress now, I applied some patches to the kernel to use the performance governor and real time. It should increase the stability of the framerate and reduce input lag. Can you try the 2.1-rc2 build from the updater?

x2k13, 2017-06-26:
I've tried the new build (via updater and fresh flash), but no change at all for me, unfortunately. I assume framerate as such doesn't seem to be the problem. It would be interesting to know what OE did differently in the past from what LE does atm. Looks like the audio handling changed somehow because in OE I didn't get the "no-audio or stuttering-audio" effects (see initial post) like I experience now in LE.

Kivutar, 2017-06-26:
I think that nothing on the audio side changed much. We have always used libalsa. It has to be something else.

x2k13, 2017-06-27:
Hm, okay so that shouldn't be causing the issues. What do you think could cause e.g. the Intro music of blazstar not playing at all on LE (does so on OE) when the audio-libs used are basically still the same? That's some interesting bug then. Thanks for your kernel patching, I bet they'll come in handy at some point but as you said it has to be something else.

Kivutar, 2017-06-27:
I start thinking about a regression in either RetroArch or FBA.

RobLoach, 2017-07-21:
Been able to use FBA on RPi3 with the latest builds.

Kivutar, 2017-08-19:
Yeah it's always usable, but x2k13 reports that it's slower

x2k13, 2017-09-04:
That's right. It's weird, actually. Looks like something is being handled differently in LE compared to OE.

x2k13, 2019-01-20:
Hello @kivutar I don't know what or how you did it, but after a long time I tried Lakka again with a fresh start of 2.2.1 and now the issue is gone! Thanks, will close the issue now.
Keep up the good work.

closed 2019-01-20 by x2k13