libretro/Lakka-LibreELEC #123: No longer able to Connect to HDMI TV with Lakka-Generic.x86_64-2.0.img.gz

evbo, 2017-05-11:
I am not able to connect to my external HDMI monitor (and am getting stuck at flower screen on my laptop monitor). I did not have this issue with lakka <2.0.

Steps:

1. I installed Lakka 64-bit: `20161221184608-r21729-g274b40f , 21-Dec-2016`

    1. Lakka displayed fine on my laptop monitor, but it wouldn't display on my external HDMI connected monitor

    1. So I followed these steps to set the default monitor index to "2" instead of "0": https://github.com/libretro/Lakka/issues/465

    1. But, that didn't work on its own, so then I tried this and it worked: https://github.com/libretro/Lakka/issues/288

1. But, then I needed support for xbox one s controllers. So I started over (per [this](https://github.com/libretro/Lakka/issues/635)), installing 2.0 instead: http://le.builds.lakka.tv/Generic.x86_64/Lakka-Generic.x86_64-2.0.tar

    1. Same display issues, so I followed the same steps as above

But, this time, nothing seems to work. Since Lakka boots expecting monitor index 2, it just hangs and the external monitor keeps flickering but no solid connection is sustained. My laptop monitor just keeps showing the flower.

relevant info (I am connected to the external monitor with an HDMI cable):

```
Lakka:~ # for p in /sys/class/drm/*/status; do con=${p%/status}; echo -n "${con#*/card?-}: "; cat $p; done
HDMI-A-1: connected
LVDS-1: connected
VGA-1: disconnected
```

```
Lakka:~ # lspci -nn
00:00.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 12h Processor Root Complex [1022:1705]
00:01.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Sumo [Radeon HD 6480G] [1002:9648]
00:01.1 Audio device [0403]: Advanced Micro Devices, Inc. [AMD/ATI] BeaverCreek HDMI Audio [Radeon HD 6500D and 6400G-6600G series] [1002:1714]
00:04.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Family 12h Processor Root Port [1022:1709]
00:05.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Family 12h Processor Root Port [1022:170a]
00:06.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Family 12h Processor Root Port [1022:170b]
00:11.0 SATA controller [0106]: Advanced Micro Devices, Inc. [AMD] FCH SATA Controller [AHCI mode] [1022:7804]
00:12.0 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] FCH USB OHCI Controller [1022:7807] (rev 11)
00:12.2 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] FCH USB EHCI Controller [1022:7808] (rev 11)
00:13.0 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] FCH USB OHCI Controller [1022:7807] (rev 11)
00:13.2 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] FCH USB EHCI Controller [1022:7808] (rev 11)
00:14.0 SMBus [0c05]: Advanced Micro Devices, Inc. [AMD] FCH SMBus Controller [1022:780b] (rev 13)
00:14.2 Audio device [0403]: Advanced Micro Devices, Inc. [AMD] FCH Azalia Controller [1022:780d] (rev 01)
00:14.3 ISA bridge [0601]: Advanced Micro Devices, Inc. [AMD] FCH LPC Bridge [1022:780e] (rev 11)
00:14.4 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] FCH PCI Bridge [1022:780f] (rev 40)
00:16.0 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] FCH USB OHCI Controller [1022:7807] (rev 11)
00:16.2 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] FCH USB EHCI Controller [1022:7808] (rev 11)
00:18.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 12h/14h Processor Function 0 [1022:1700] (rev 43)
00:18.1 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 12h/14h Processor Function 1 [1022:1701]
00:18.2 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 12h/14h Processor Function 2 [1022:1702]
00:18.3 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 12h/14h Processor Function 3 [1022:1703]
00:18.4 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 12h/14h Processor Function 4 [1022:1704]
00:18.5 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 12h/14h Processor Function 6 [1022:1718]
00:18.6 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 12h/14h Processor Function 5 [1022:1716]
00:18.7 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 12h/14h Processor Function 7 [1022:1719]
01:00.0 Network controller [0280]: Realtek Semiconductor Co., Ltd. RTL8188CE 802.11b/g/n WiFi Adapter [10ec:8176] (rev 01)
02:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8101/2/6E PCI Express Fast/Gigabit Ethernet controller [10ec:8136] (rev 05)
03:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS5209 PCI Express Card Reader [10ec:5209] (rev 01)
03:00.1 SD Host controller [0805]: Realtek Semiconductor Co., Ltd. RTS5209 PCI Express Card Reader [10ec:5209] (rev 01)
```

startup logs:

```
Lakka:~ # systemctl stop retroarch.service ; retroarch -v
RetroArch [INFO] :: This is RetroArch version 1.5.0 (Git 2a13af5)
RetroArch [INFO] :: === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3
Built: Apr 24 2017
RetroArch [INFO] :: Version: 1.5.0
RetroArch [INFO] :: Git: 2a13af5
[... 80 lines cut ...]
RetroArch [INFO] :: [udev]: Pad #0 (/dev/input/event6) supports force feedback.
RetroArch [INFO] :: [udev]: Pad #0 (/dev/input/event6) supports 16 force feedback effects.
RetroArch [INFO] :: Autodetect: 179 profiles found.
RetroArch [INFO] :: Autodetect: selected configuration: /tmp/joypads/udev/Microsoft_X-Box_One_pad_(Firmware_2015).cfg
RetroArch [INFO] :: [udev]: Plugged pad: Microsoft X-Box One S pad (1118:746) on port #1.
RetroArch [INFO] :: [udev]: Pad #1 (/dev/input/event7) supports force feedback.
RetroArch [INFO] :: [udev]: Pad #1 (/dev/input/event7) supports 16 force feedback effects.
RetroArch [INFO] :: Autodetect: 179 profiles found.
RetroArch [INFO] :: Autodetect: selected configuration: /tmp/joypads/udev/Microsoft_X-Box_One_S_pad.cfg
RetroArch [INFO] :: Found joypad driver: "udev".
ALSA lib /home/kivutar/lakkadisk/kivutar/Lakka-LibreELEC/build.Lakka-Generic.x86_64-2.0/alsa-lib-1.1.2/src/pcm/pcm_dmix.c:1041:(snd_pcm_dmix_open) unable to open slave
RetroArch [ERROR] :: ALSA: Failed to initialize...
RetroArch [ERROR] :: Failed to initialize audio driver. Will continue without audio.
RetroArch [INFO] :: Found menu display driver: "menu_display_gl".
RetroArch [INFO] :: Using font rendering backend: freetype.
RetroArch [INFO] :: Using font rendering backend: freetype.
RetroArch [INFO] :: SRAM will not be saved.
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_history.lpl].
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_music_history.lpl].
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_video_history.lpl].
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_image_history.lpl].
RetroArch [ERROR] :: failed_to_start_audio_driver
RetroArch [INFO] :: [KMS]: New FB: 1600x900 (stride: 6400).
RetroArch [INFO] :: [KMS]: New FB: 1600x900 (stride: 6400).
```

gouchi, 2017-05-11:
Is it possible to [purge your configuration file](http://www.lakka.tv/doc/Upgrading-Lakka/#purging-your-retroarchcfg) ?

Then try to boot with (hit tab on the boot screen)
`live video=VGA-1:d LVDS-1:d video=HDMI-A-1:e`

If it still doesn't work, try to change to `video_monitor_index = "2"` in the [configuration file](http://www.lakka.tv/doc/Configuring-Lakka/#using-the-command-line-interface).

Thank you.

evbo, 2017-05-13:
Unfortunately I am still getting the same issue. In case it helps, here were the logs after the purge:

```
Lakka (community) Version: 2.0
Lakka:~ # systemctl stop retroarch
Lakka:~ # rm .config/retroarch/retroarch.cfg
Lakka:~ # retroarch
ALSA lib /home/kivutar/lakkadisk/kivutar/Lakka-LibreELEC/build.Lakka-Generic.x86_64-2.0/alsa-lib-1.1.2/src/pcm/pcm_dmix.c:1041:(snd_pcm_dmix_open) unable to open slave
```

(The command prompt never returned)

gouchi, 2017-05-13:
Thank you for your tests.

If  you boot with 
`live textmode`

And start manually do you get any output after put back `video_monitor_index="0"` ?

Another thing you can try that may help is to put `video_threaded= "false"` in [the configuration file](http://www.lakka.tv/doc/Configuring-Lakka/). 

Also `aplay -L` returns something ?

Thank you.

evbo, 2017-05-14:
I did not see any output after starting with `live textmode` (and index set back to 0 prior to booting)

Output of `aplay -L`:

```
Lakka (community) Version: 2.0
Lakka:~ # aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
hdmi:CARD=Generic,DEV=0
    HD-Audio Generic, HDMI 0
    HDMI Audio Output
default:CARD=Generic_1
    HD-Audio Generic, 92HD81B1X5 Analog
    Default Audio Device
sysdefault:CARD=Generic_1
    HD-Audio Generic, 92HD81B1X5 Analog
    Default Audio Device
front:CARD=Generic_1,DEV=0
    HD-Audio Generic, 92HD81B1X5 Analog
    Front speakers
surround21:CARD=Generic_1,DEV=0
    HD-Audio Generic, 92HD81B1X5 Analog
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=Generic_1,DEV=0
    HD-Audio Generic, 92HD81B1X5 Analog
    4.0 Surround output to Front and Rear speakers
surround41:CARD=Generic_1,DEV=0
    HD-Audio Generic, 92HD81B1X5 Analog
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=Generic_1,DEV=0
    HD-Audio Generic, 92HD81B1X5 Analog
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=Generic_1,DEV=0
    HD-Audio Generic, 92HD81B1X5 Analog
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=Generic_1,DEV=0
    HD-Audio Generic, 92HD81B1X5 Analog
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
```

After setting video_threaded= "false" and switching to video_monitor_index 2, here's what happens _sometimes_:

1. For maybe 60 seconds there is nothing displayed on the external monitor and my laptop displays the flower

1. There is then _sometimes_ a brief (at most ~10 seconds) moment of black screen on the external monitor.

1. After the <10 second black screen, I _sometimes_ get the manufacturer warning that there is "Not Support!". Then a few seconds later it always displays "No Signal!".

1. Every minute or so this cycle repeats.

While switching between video_monitor_index values, I noted these logs for my displays:

```
RetroArch [INFO] :: Video @ fullscreen
RetroArch [INFO] :: [DRM]: Found 3 connectors.
RetroArch [INFO] :: [DRM]: Connector 0 connected: no
RetroArch [INFO] :: [DRM]: Connector 0 has 0 modes.
RetroArch [INFO] :: [DRM]: Connector 1 connected: yes
RetroArch [INFO] :: [DRM]: Connector 1 has 11 modes.
RetroArch [INFO] :: [DRM]: Connector 1 assigned to monitor index: #1.
RetroArch [INFO] :: [DRM]: Connector 2 connected: yes
RetroArch [INFO] :: [DRM]: Connector 2 has 23 modes.
RetroArch [INFO] :: [DRM]: Connector 2 assigned to monitor index: #2.
RetroArch [INFO] :: [DRM]: Mode 0: (1360x768) 1360 x 768, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 1: (1920x1080) 1920 x 1080, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 2: (1920x1080) 1920 x 1080, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 3: (1920x1080i) 1920 x 1080, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 4: (1920x1080i) 1920 x 1080, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 5: (1920x1080) 1920 x 1080, 50 Hz
RetroArch [INFO] :: [DRM]: Mode 6: (1920x1080i) 1920 x 1080, 50 Hz
RetroArch [INFO] :: [DRM]: Mode 7: (1280x1024) 1280 x 1024, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 8: (1280x720) 1280 x 720, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 9: (1280x720) 1280 x 720, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 10: (1280x720) 1280 x 720, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 11: (1280x720) 1280 x 720, 50 Hz
RetroArch [INFO] :: [DRM]: Mode 12: (1280x720) 1280 x 720, 50 Hz
RetroArch [INFO] :: [DRM]: Mode 13: (1024x768) 1024 x 768, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 14: (800x600) 800 x 600, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 15: (800x600) 800 x 600, 72 Hz
RetroArch [INFO] :: [DRM]: Mode 16: (800x600) 800 x 600, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 17: (720x576) 720 x 576, 50 Hz
RetroArch [INFO] :: [DRM]: Mode 18: (720x480) 720 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 19: (720x480) 720 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 20: (640x480) 640 x 480, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 21: (640x480) 640 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 22: (640x480) 640 x 480, 60 Hz
```

Then, when my external monitor is active I get: `RetroArch [INFO] :: [KMS]: New FB: 1360x768 (stride: 5632).`

But when my laptop monitor is active I get: `RetroArch [INFO] :: [KMS]: New FB: 1600x900 (stride: 6400).`

gouchi, 2017-05-14:
You booted with `live textmode` and without `video=VGA-1:d LVDS-1:d video=HDMI-A-1:e` with `video_monitor_index = "0"` and when you start manually retroarch you get no output on your PC screen ?

We you can try to get more information about drm if you boot with 

`live ssh drm.debug=0x1e log_buf_len=1M`

then in the terminal
`journalctl | grep drm `

The log will be long...

Thank you.

evbo, 2017-05-14:
Yeah, when I booted using only `live textmode` with `video_monitor_index="0"` it immediately displayed the flower on my laptop screen and then the command prompt.

Attached are those logs after booting with `live ssh drm.debug=0x1e log_buf_len=1M` (too long to post)

[journalctl | grep drm.txt](https://github.com/libretro/Lakka-LibreELEC/files/999413/journalctl.grep.drm.txt)

evbo, 2017-05-14:
[dmesg.txt](https://github.com/libretro/Lakka-LibreELEC/files/999529/dmesg.txt)

Note: this time I booted with that command while the TV was left on (in case it makes a difference)

gouchi, 2017-05-14:
I don't understand in the dmesg output you don't have the boot sequence ?

evbo, 2017-05-14:
I ran `dmesg` from a separate computer that ssh'd onto the machine after it had booted. Would that make any difference?

Could it be that logging needs to be more verbose?

gouchi, 2017-05-14:
It should be fine. You should get something like [this output](https://github.com/libretro/Lakka-LibreELEC/files/922394/DMESG.txt).

evbo, 2017-05-14:
I see what was happening now. The `log_buf_len` was truncating the output. I ran it again this time with: `log_buf_len=100M` and got the output starting at `0`:
[dmesg.txt](https://github.com/libretro/Lakka-LibreELEC/files/999590/dmesg.txt)

My laptop is old and takes a while to boot :)

gouchi, 2017-05-14:
Can you try to blacklist nouveau and reboot  if it helps ?

`echo "nouveau" >> /storage/.config/modprobe.d/blacklist.conf`

Thank you.

gouchi, 2017-05-14:
I don't know if those lines are harmful as the modelines are looking good.
```
[drm:drm_add_edid_modes] HDMI: DVI dual 0, max TMDS clock 0 kHz
[drm:drm_edid_to_eld] HDMI: latency present 0 0, video latency 0 0, audio latency 0 0
```
Maybe you can try to force a resolution with something like ?

`video=VGA-1:d LVDS-1:d video=HDMI-A-1:640x480@60 `

You don't have another hdmi cable or screen to test ?

Thank you.

evbo, 2017-05-14:
No, I don't have another screen, but this TV works when booting Windows 7 and also worked when booting an older version of Lakka 64-bit: `20161221184608-r21729-g274b40f , 21-Dec-2016`

I will update this comment with results from your suggestion in a few hours, thanks!

evbo, 2017-05-15:
The external monitor is connected via a brand new HDMI cable. It is a PC, and when the lakka live doesn't boot, it boots windows 7 and the monitor works great. I previously had the monitor working with Lakka 64-bit: `20161221184608-r21729-g274b40f , 21-Dec-2016`.

It is now _partially_ working (displaying the flower!). Here is what I've tried:

1. Disabling nouveau (didn't fix it):

   ```
   LABEL live
   KERNEL /KERNEL
   APPEND boot=UUID=2404-4447 disk=UUID=ac471f6b-554f-4380-8bb5-4d4f3821014a portable quiet vt.global_cursor_default=0 loglevel=2 nouveau.blacklist=1
   ```

1. Disabling nouveau and fixing resolution (flower then displayed on ext monitor! but nothing else...):

   ```
   LABEL live
   KERNEL /KERNEL
   APPEND boot=UUID=2404-4447 disk=UUID=ac471f6b-554f-4380-8bb5-4d4f3821014a portable quiet vt.global_cursor_default=0 loglevel=2 nouveau.blacklist=1 video=HDMI-A-1:640x480@60
   ```
1. Disabling other monitors and fixing resolution to several different values on each attempt:
   ```
   LABEL live
   KERNEL /KERNEL
   APPEND boot=UUID=2404-4447 disk=UUID=ac471f6b-554f-4380-8bb5-4d4f3821014a portable quiet vt.global_cursor_default=0 loglevel=2 video=VGA-1:d LVDS-1:d video=HDMI-A-1:1280x720@60
   ```

In each case, the ext monitor is correctly being set to the right resolution and flower is displayed (except in one case), but the Lakka menu will not show after setting monitor index to "2". Here are all the modes I tested:
640x480
1024x768
1280x720
1280x1024
1360x768 (fails to load flower screen)

Here are the modes I didn't yet try (should I?):
720x480
800x600
1920x1080
1920x1080i

gouchi, 2017-05-15:
When you set the right resolution and flower is displayed, can you provide [Lakka system log ](http://www.lakka.tv/doc/Troubleshooting-Lakka/)?

Maybe the video_monitor_index has changed ? Like we have seen when you [purged the configuration file](https://github.com/libretro/Lakka-LibreELEC/issues/123#issuecomment-301255752).

You can try the others modes according to the [detected modelines](https://github.com/libretro/Lakka-LibreELEC/issues/123#issuecomment-301326986).

Thank you.

evbo, 2017-05-16:
Here are the logs with `video_monitor_index` set to 0 and the flower is displayed on the external monitor:
[retroarch -v.txt](https://github.com/libretro/Lakka-LibreELEC/files/1002882/retroarch.-v.txt)

I tried all other resolution settings. Upon reboot, I tried switching the monitor to "2" each time but the external display was only ever able to display the flower (with index at "0" or "1"). Before each reboot I reset monitor index to "0" each time.

gouchi, 2017-05-16:
According to your [log](https://github.com/libretro/Lakka-LibreELEC/files/1002882/retroarch.-v.txt), you need to set video_monitor_index="1".

```
RetroArch [INFO] :: [DRM]: Connector 1 has 11 modes.
RetroArch [INFO] :: [DRM]: Connector 1 assigned to monitor index: #1.
```
```
[RetroArch [INFO] :: [DRM]: Mode 0: (1600x900) 1600 x 900, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 1: (1440x900) 1440 x 900, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 2: (1280x854) 1280 x 854, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 3: (1280x800) 1280 x 800, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 4: (1280x720) 1280 x 720, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 5: (1152x768) 1152 x 768, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 6: (1024x768) 1024 x 768, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 7: (800x600) 800 x 600, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 8: (848x480) 848 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 9: (720x480) 720 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 10: (640x480) 640 x 480, 59 Hz](url)

evbo, 2017-05-16:
Aren't those the resolution modes of my laptop? Setting it to "1" continues to display on the laptop. It's when I set it to "2" that it adjusts the resolution and refresh rate on my external monitor but then no video gets displayed.

gouchi, 2017-05-16:
Strange when you specify video hdmi output RA tries to output to your laptop ?

Yes it will be interesting to see the log from the Dec 21st  2016 release.

Thank you.

evbo, 2017-05-17:
Ok, here were my precise steps using `20161221184608-r21729-g274b40f , 21-Dec-2016`:

1. Fresh new install on a flash drive, then boot from live

   1. Immediately, without any configuration, the external hard drive is displaying boot logs and then the flower

1. I then ssh onto the machine and restart `retroarch -v`, resulting in logs: 
[retoarch -v.txt](https://github.com/libretro/Lakka-LibreELEC/files/1006186/retoarch.-v.txt)
 
1. I set the `video_monitor_index` from "0" to "2" (http://www.lakka.tv/doc/Configuring-Lakka/#using-the-command-line-interface)
   
   1.  Immediately the external display works upon restarting with `retroarch -v`, with logs: 
[retoarch -v.txt](https://github.com/libretro/Lakka-LibreELEC/files/1006192/retoarch.-v.txt)

Please let me know if any further logs will help and thank you for your dedication to resolving this bug so far!

gouchi, 2017-05-18:
Is it possible to try to those images ([updated mesa](http://gouchi.free.fr/Lakka/Lakka-Generic.x86_64-8.0-devel-20170517145722-r26145-gf8bc02f.img.gz) and [updated mesa + RA](http://gouchi.free.fr/Lakka/Lakka-Generic.x86_64-8.0-devel-20170517160626-r26145-gf8bc02f.img.gz)), if you reproduce the issue.
Thank you for tests !

evbo, 2017-05-19:
Hi, same issue occurred with both of those images:

1. `Lakka-Generic.x86_64-8.0-devel-20170517160626-r26145-gf8bc02f.img`:

   1. `retroarch -v`: 
[retroarch -v silver.txt](https://github.com/libretro/Lakka-LibreELEC/files/1012806/retroarch.-v.silver.txt)
    
   1. setting `video_monitor_index`="2" then `retroarch -v`:
[retroarch -v ind2 silver.txt](https://github.com/libretro/Lakka-LibreELEC/files/1012808/retroarch.-v.ind2.silver.txt)
 

1. `Lakka-Generic.x86_64-8.0-devel-20170517145722-r26145-gf8bc02f.img`:

   1. `retroarch -v`: 
[retroarch -v purple.txt](https://github.com/libretro/Lakka-LibreELEC/files/1012812/retroarch.-v.purple.txt)

   1. setting `video_monitor_index`="2" then `retroarch -v`:
[retroarch -v ind2 purple.txt](https://github.com/libretro/Lakka-LibreELEC/files/1012813/retroarch.-v.ind2.purple.txt)

gouchi, 2017-05-21:
I can't find anything suspicious.

Is it possible to test with [this image](http://gouchi.free.fr/Lakka/Lakka-Generic.x86_64-8.0-devel-20170521122536-r26152-g01c8330.img.gz) which is using old RA from latest Lakka.

Kivutar, 2017-05-22:
His card is an evergreen, IIRC we had issue with it before.

evbo, 2017-05-24:
Here are the results with that latest image you requested:

[retroarch -v.txt](https://github.com/libretro/Lakka-LibreELEC/files/1024225/retroarch.-v.txt)

[video monitor index 2 and then retroarch -v.txt](https://github.com/libretro/Lakka-LibreELEC/files/1024226/ind.2.retroarch.-v.txt)

[dmesg with monitor index 0.txt](https://github.com/libretro/Lakka-LibreELEC/files/1024227/dmesg.txt)

evbo, 2017-05-26:
I have found a version of Lakka that both supports my TV and my XBox One "S" Controller:

http://le.builds.lakka.tv/Generic.x86_64/Lakka-Generic.x86_64-2.0-rc5.img.gz

This also means the issue has been narrowed down to: `Lakka-Generic.x86_64-2.0.img.gz`

So, for all intents and purposes, my problem is solved :)  but if you need someone to test potential bug fixes I'd be happy to.

gouchi, 2017-05-28:
With Lakka 2.0 rc5 did you try to [change audio device](http://www.lakka.tv/doc/Audio-settings/#switching-audio-device-from-the-gui) to get sound ?

evbo, 2017-05-28:
yes, and I tried different drivers as well. I could only get video working for the xmb and rgui menus. As soon as I tried to load a game or edit any video settings the video display would stop working until restart.

gouchi, 2017-05-28:
Can you make a test with [Lakka 2.1 rc1](http://le.builds.lakka.tv/Generic.x86_64/Lakka-Generic.x86_64-2.1-rc1.img.gz) ?

Thank you.

evbo, 2017-05-31:
thanks, this is an improvement, but it still doesn't totally work.

I am able to load the xmb menu, download the snes preview from content downloader, but when I go to play it I get the first few seconds of sound and then the display is killed and lakka does not recover (same thing happens after `systemctl stop retroarch` and then `retroarch -v`).

logs:
[dmseg](https://github.com/libretro/Lakka-LibreELEC/files/1040047/dmseg.txt)
[dmseg boot with lvds-1:d](https://github.com/libretro/Lakka-LibreELEC/files/1040048/dmseg.after.lvds.d.txt)
[retroarch -v](https://github.com/libretro/Lakka-LibreELEC/files/1040055/retroarch.-v.txt)

Kivutar, 2017-06-13:
@gouchi can it fail to launch a rom because of alsa?

gouchi, 2017-06-14:
@Kivutar  Hope not, I don't remember if it was breaker when we fixed this alsa issue.

@evbo same result with [Lakka 2.1 rc2](http://le.builds.lakka.tv/Generic.x86_64/Lakka-Generic.x86_64-2.1-rc2.img.gz) ?

evbo, 2017-06-14:
@gouchi slightly different result. I posted all relevant logs here: https://github.com/libretro/Lakka-LibreELEC/issues/123#issuecomment-305056332

Let me know if more would be helpful

evbo, 2017-06-17:
@gouchi @kivutar this is probably a separate issue, but for some reason `2.1 rc2` will not connect to my wifi internet. As such, I am unable to test running any roms/content  :(

When I try to connect to wifi on this build, every time I get a network that has a name consisting of randomly generated ascii text and after entering the password I get:
```
Error 'Connect': Invalid Argument
```

I have tried booting with `live ssh` using adhoc network with the external monitor powered off (to avoid interference) and have restarted numerous times over the past several days but it consistently fails to identify any networks to connect to. I have occasionally run into this issue before with older Lakka builds, but after enough restarts it eventually works.

I just switched back to my `Dec 21 2016 274b40f` Lakka build and it connected to the wifi perfectly. So I think there may be networking incompatibility issues with `2.1 rc2`

Despite all of that, the rgui menu loaded great with `2.1 rc2` if I manually disabled LVDS from `syslinux.cfg` when connecting the Lakka flashdrive to my laptop.

Kivutar, 2017-06-18:
What is your wifi card?

evbo, 2017-06-18:
Here is my laptop and all of its specs: https://support.hp.com/us-en/product/hp-pavilion-g7-1200-notebook-pc-series/5145712/document/c03024323

The network card is listed as: 10/100 Ethernet LAN
The wireless connectivity is listed as: 802.11b/g/n WLAN

Other than a RAM upgrade, this laptop has stock parts from HP.

gouchi, 2017-06-18:
Please provide the output of `lspci -nnk | grep -A3 802`

Thank you.

evbo, 2017-06-18:
I am unable to ssh onto `rc2` build to run that command, but here is the output of that command on the `Dec 21 2016 274b40f` build:

```
Lakka (unofficial) Version: devel-20161221183017-r21729-g274b40f
Lakka git: 274b40f7a5f7a880309a4fe8765e69ac8d8153e1
Lakka:~ # lspci -nnk | grep -A3 802
01:00.0 Network controller [0280]: Realtek Semiconductor Co., Ltd. RTL8188CE 802.11b/g/n WiFi Adapter [10ec:8176] (rev 01)
	DeviceName: Realtek RTL8188CE 802.11b/g/n WiFi Adapter
	Subsystem: Hewlett-Packard Company RTL8188CE 802.11b/g/n WiFi Adapter [103c:1629]
	Kernel driver in use: rtl8192ce
	Kernel modules: rtl8192ce
02:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8101/2/6E PCI Express Fast/Gigabit Ethernet controller [10ec:8136] (rev 05)
```

gouchi, 2017-08-06:
Do you reproduce your issue with [Lakka 2.1 RC4](http://le.builds.lakka.tv/Generic.x86_64/Lakka-Generic.x86_64-2.1-rc4.img.gz) ?

evbo, 2017-08-09:
@gouchi is there any other way to connect to lakka besides ssh? For some reason, with these newer versions, every time I try to connect it times out.

evbo, 2017-08-09:
Unfortunately I am still having difficulty with wireless ssh on these newer versions of Lakka and was not able to ssh onto the box. For some reason, only the very first time I booted this version of Lakka was I able to connect to wifi (and I promptly used to updater to download the "Boss Gaiden" SNES freeware sample). I will try looking into a direct ethernet connection (assuming its issues with the wifi).

However, in the meantime I did try:

- boot with `live ssh video=LVDS-1:d`
   As with RC3, the rgui menu displayed marvelously. Upon starting Boss Gaiden.sfc, the video cut out and game audio started coming out of my laptop (not external monitor HDMI sound). My external monitor then began displaying "Not Support!". If I tried to quit the game (pressing L3 + R3) the game audio stopped but the video did not recover.

gouchi, 2017-08-09:
Thank you for tests.

Did you try to change monitor index via Settings Tab > Video > Monitor Index when you use rgui menu ? Also for the sound, check in Settings Tab > Audio > Audio device is set to hdmi:CARD=Generic,DEV=0 device.

We did not try to change the option in the configuration file if you put back the xmb menu ?
menu_shader_pipeline = "2" to menu_shader_pipeline = "0".

If ssh doesn't work well you can try to enter the command directly to the terminal
`live textmode `

Thank you.

evbo, 2017-08-09:
I did not need to change the monitor index, after disabling LVDS video it correctly displayed on the external monitor.

Anytime I adjust the video settings (switching index) the external display disconnects.

I can display both xmb and rgui menus. But the video disconnects as soon as I start to play a game or change a video setting. I tried using `menu_shader_pipeline = "0"` (basic red theme with no background animation) and this time when I started a game the external monitor went black and then displayed a resolution adjustment message to "127 X 127". As before, game audio was coming through the laptop but no video was displayed while gaming.

evbo, 2017-08-11:
@gouchi if it's helpful, I would also be happy to host a google hangout sometime where we can interactively debug this issue.

Just thought to suggest that as you might get a better idea seeing the problem first hand.

gouchi, 2017-08-19:
It will be better if we had access to your PC by ssh.

evbo, 2017-11-14:
It appears my router supports remote access for port 80. I'll just need to set it up temporarily for access to you in a way that allows me to disable access when you're done. How can we coordinate this?

gouchi, 2017-11-17:
Please come to [our irc channel](http://webchat.freenode.net/?channels=%23lakkatv).

evbo, 2017-11-25:
Hi, I've joined the irc channel a few times, but not finding anyone. Is there a particular time UTC I should enter?

Also, for what it's worth I booted the latest RC6 version of lakka, but still it does not communicate with the TV.

I tried even booting with: `live video=VGA-1:d LVDS-1:d video=HDMI-A-1:e` but no luck.

Kivutar, 2017-11-25:
I saw you but I didn't have the time to reply. Try to connect during the evening in the european time, to catch gouchi.

evbo, 2018-01-21:
ok, thanks guys. I've purchased a static ip to allow you to access the lakka machine through ssh anytime you please. I've left Lakka 2.1 RC6 running so you can debug anytime that works for you. Please private message me to get the IP address to use for ssh'ing.

evbo, 2018-01-22:
Installed: http://nightly.builds.lakka.tv/2018-01-21/Generic.x86_64/Lakka-Generic.x86_64-2.1-devel-r28138-g6576593.img.gz

boot config'd with:
```
SAY Wait for live session to start or press <TAB> for more options (installer)
DEFAULT live
TIMEOUT 50
PROMPT 1

LABEL installer
  KERNEL /KERNEL
  APPEND boot=UUID=2101-4437 installer quiet tty vga=current vt.global_cursor_default=0 loglevel=2

LABEL live
  KERNEL /KERNEL
  APPEND boot=UUID=2101-4437 disk=UUID=580895f5-5236-4fbd-995b-a00c1725e7f0 portable quiet vt.global_cursor_default=0 loglevel=2 video=LVDS-1:d video=HDMI-A-1:D

```

HDMI displayed the menu, but as soon as Boss Gaiden was loaded it disconnected HDMI. This machine will be left connected via ethernet to the network tomorrow starting around 1 PM so that @gouchi should still have access the rest of the day.

gouchi, 2018-01-22:
```
 retroarch -v  -L /tmp/cores/snes9x_libretro.so roms/downloads/240pSuite.sfc 
[INFO] RetroArch 1.7.0 (Git fd799bccef)
[INFO] Redirecting save file to "roms/downloads/240pSuite.srm".
[INFO] Redirecting savestate to "roms/downloads/240pSuite.state".
[INFO] === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 
Built: Jan 21 2018
[INFO] Version: 1.7.0
[INFO] Git: fd799bccef
[INFO] =================================================
[INFO] [Config]: Loading default config.
[INFO] [Config]: loading config from: (null).
[INFO] Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
[INFO] Loading dynamic libretro core from: "/tmp/cores/snes9x_libretro.so"
[INFO] [overrides] no core-specific overrides found at /storage/.config/retroarch/config/Snes9x/Snes9x.cfg.
[INFO] [overrides] no game-specific overrides found at /storage/.config/retroarch/config/Snes9x/240pSuite.cfg.
[INFO] Shaders: preset directory: /tmp/shaders/presets
[INFO] Shaders: no game-specific preset found at /tmp/shaders/presets/Snes9x/240pSuite.cgp.
[INFO] Shaders: no game-specific preset found at /tmp/shaders/presets/Snes9x/240pSuite.glslp.
[INFO] Shaders: no core-specific preset found at /tmp/shaders/presets/Snes9x/Snes9x.cgp.
[INFO] Shaders: no core-specific preset found at /tmp/shaders/presets/Snes9x/Snes9x.glslp.
[INFO] Environ SET_VARIABLES.
[INFO] Environ SET_CONTROLLER_INFO.
[INFO] Controller port: 1
[INFO]    SNES Joypad (ID: 1)
[INFO]    SNES Mouse (ID: 2)
[INFO]    Multitap (ID: 257)
[INFO] Controller port: 2
[INFO]    SNES Joypad (ID: 1)
[INFO]    SNES Mouse (ID: 2)
[INFO]    Multitap (ID: 257)
[INFO]    SuperScope (ID: 260)
[INFO]    Justifier (ID: 516)
[INFO] Remaps: remap directory: /storage/remappings
[INFO] Remaps: no game-specific remap found at /storage/remappings/Snes9x/240pSuite.rmp.
[INFO] Remaps: no core-specific remap found at /storage/remappings/Snes9x/Snes9x.rmp.
[INFO] Redirecting save file to "/storage/savefiles/240pSuite.srm".
[INFO] Redirecting savestate to "/storage/savestates/240pSuite.state".
[INFO] Environ GET_LOG_INTERFACE.
[INFO] Environ SET_SUPPORT_ACHIEVEMENTS: yes.
Sound buffer size: 128160 (32040 samples)
[INFO] Environ PERFORMANCE_LEVEL: 12.
[INFO] Loading content file: roms/downloads/240pSuite.sfc.
[INFO] Did not find a valid content patch.
[INFO] CRC32: 0xfe53a088 .
[INFO] Environ SET_INPUT_DESCRIPTORS:
[INFO] 	RetroPad, User 1, Button "B (bottom)" => "B"
[INFO] 	RetroPad, User 1, Button "Y (left)" => "Y"
[INFO] 	RetroPad, User 1, Button "Select" => "Select"
[INFO] 	RetroPad, User 1, Button "Start" => "Start"
[INFO] 	RetroPad, User 1, Button "D-Pad Up" => "D-Pad Up"
[INFO] 	RetroPad, User 1, Button "D-Pad Down" => "D-Pad Down"
[INFO] 	RetroPad, User 1, Button "D-Pad Left" => "D-Pad Left"
[INFO] 	RetroPad, User 1, Button "D-Pad Right" => "D-Pad Right"
[INFO] 	RetroPad, User 1, Button "A (right)" => "A"
[INFO] 	RetroPad, User 1, Button "X (up)" => "X"
[INFO] 	RetroPad, User 1, Button "L" => "L"
[INFO] 	RetroPad, User 1, Button "R" => "R"
[INFO] 	RetroPad, User 2, Button "B (bottom)" => "B"
[INFO] 	RetroPad, User 2, Button "Y (left)" => "Y"
[INFO] 	RetroPad, User 2, Button "Select" => "Select"
[INFO] 	RetroPad, User 2, Button "Start" => "Start"
[INFO] 	RetroPad, User 2, Button "D-Pad Up" => "D-Pad Up"
[INFO] 	RetroPad, User 2, Button "D-Pad Down" => "D-Pad Down"
[INFO] 	RetroPad, User 2, Button "D-Pad Left" => "D-Pad Left"
[INFO] 	RetroPad, User 2, Button "D-Pad Right" => "D-Pad Right"
[INFO] 	RetroPad, User 2, Button "A (right)" => "A"
[INFO] 	RetroPad, User 2, Button "X (up)" => "X"
[INFO] 	RetroPad, User 2, Button "L" => "L"
[... 185 lines cut ...]
[INFO] [Autoconf]: 203 profiles found.
[INFO] [udev]: Plugged pad: Microsoft X-Box One pad (Firmware 2015) (1118:733) on port #1.
[INFO] [udev]: Pad #1 (/dev/input/event8) supports force feedback.
[INFO] [udev]: Pad #1 (/dev/input/event8) supports 16 force feedback effects.
[INFO] [autoconf]: selected configuration: /tmp/joypads/udev/Microsoft_X-Box_One_pad.cfg
[INFO] [Autoconf]: 203 profiles found.
[INFO] [Joypad]: Found joypad driver: "udev".
[INFO] [Video]: Found display server: null
[INFO] ALSA: Using signed 16-bit format.
[INFO] ALSA: Period size: 384 frames
[INFO] ALSA: Buffer size: 1536 frames
[INFO] [autoconf]: selected configuration: /tmp/joypads/udev/Microsoft_X-Box_One_pad_(Firmware_2015).cfg
[INFO] [Menu]: Found menu display driver: "menu_display_gl".
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [LED]: LED driver = 'null' 0x70c1e0
[INFO] Loading history file: [/storage/.config/retroarch/content_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_favorites.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_music_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_video_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_image_history.lpl].
[INFO] [KMS]: New FB: 1360x768 (stride: 5632).
[INFO] [KMS]: New FB: 1360x768 (stride: 5632).

```

evbo, 2018-01-23:
are the logs not capturing the error? When I came home the HDMI tv was blue screen with "no signal", so it seems likely you experienced the same issue as me. I'll restart the machine again tomorrow in case you'd like to run another test.

gouchi, 2018-01-30:
Hi,

If I am correct OpenElec is using Xorg and we are using only drm/kms on Generic platform.

Can you try to reproduce on another Linux distro ? Boot then go to terminal (Ctrl+Alt+F1) then launch RetroArch in command line. KMS mode needs to [be enable](https://docs.libretro.com/guides/kms-mode).

Thank you.

evbo, 2018-01-30:
@gouchi ok, I'll try this installation using a Lubuntu live usb: http://retroarch.com/index.php?page=linux-instructions

evbo, 2018-02-18:
Using Lubuntu 16.04, the video ran perfectly on the HDMI TV. I haven't figured out my audio drivers, but hopefully this is enough evidence that the HDMI TV does work with Retroarch on other distros. 

See attached text file for verbos logs:
[lubuntu-retroarch.txt](https://github.com/libretro/Lakka-LibreELEC/files/1734847/lubuntu-retroarch.txt)

evbo, 2018-02-18:
Thanks @gouchi, but it seems that KMS is not supported by all driver technologies (or am I reading outdated documentation?).

This is exactly the video card I have, so maybe the issue is that I need to follow [this advice:](https://wiki.archlinux.org/index.php/kernel_mode_setting#Late_KMS_start)

>The proprietary AMD Catalyst driver does not support KMS. In order to use KMS you have to replace it with the open-source ATI driver.

My efforts will be slow going, but I suppose I'll need to investigate installing alternative video drivers for KMS to work, and at that point it's conceivable that Lakka might work too.

gouchi, 2018-02-19:
If you are using k/l/x/ubuntu, by default it will use open source driver so you can use directly [KMS](https://en.wikipedia.org/wiki/Mode_setting).

Please confirm that the RetroArch binary you are using has KMS enabled with

`retroarch --features | grep -A1 -i kms`

evbo, 2018-02-20:
Thanks for clarifying, it appears so:

```
retroarch --features | grep -A1 -i kms
  KMS:
		Video context driver: yes
```

[47 more replies]

closed 2020-08-18 by Ntemis