libretro/Lakka-LibreELEC #125: Feature request - allow SD card file transfer

hooby3dfx, 2017-05-13:
I don't know if this is feasible or not, but right now with a working Lakka system (for example Raspberry Pi) if you access the SD card on a PC (like via Windows) the only partition that appears is the small boot partition. 

It is not possible to copy roms, etc directly to the SD card easily (must be done through network).

Alternatively (as a workaround), are there any good cross platform tools to access the partition with roms?

Kivutar, 2017-05-13:
Yes, they are listed in the documentation: https://github.com/libretro/Lakka-LibreELEC/wiki/Accessing-Lakka-filesystem#direct-drive-access

Mr-Jake, 2017-05-13:
@Kivutar

For Windows users that want to access the SD card to manage roms, would it would work to use partition software to shrink the Lakka ext4 partition and then create a new FAT or NTFS partition to use for the folders roms / savefiles / savestates / screenshots / system / playlists ....... etc?

Kivutar, 2017-05-13:
I'm not sure, you can try.

Ntemis, 2017-05-13:
Its because the other partition is ext4 you need linux os or a special app
on windows and maybe macos

Kivutar, 2017-05-22:
I close this, we can't deviate from LibreELEC that much.

closed 2017-05-22 by Kivutar