libretro/Lakka-LibreELEC #126: XU4 setenv hdmi settings ignored

externo6, 2017-05-13:
This issue tracker is NOT for help or support. Please use the Lakka forum (https://forums.libretro.com/c/libretro/lakka-tv-general) or IRC @ #lakka on freenode.net for those kinds of posts.

BEFORE you report a bug make sure you have tried the latest nightly version of Lakka at http://sources.lakka.tv/nightly/. Your bug might be already fixed.

If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using? 2.0
- What system hardware are you using? Odroid XU4
- What did you do? Changed flag in boot.ini to setenv video_output "video=HDMI-A-1:1920x1080M@60"
- What did you expect to happen? force 60hz at 1080P
- What happened instead? but stays at 50hz (I also tried forcing 720P but also ignored)

These settings worked with openelec versions of Lakka, I've also had this issue with other OS's and it was something todo with the kernels.

Kivutar, 2017-05-14:
Hello, it's true that I don't find any platform_init in the XU3 project, so this argument may be ignored.

I'm going to check in the project of zalaare https://github.com/zalaare/Project-Odroid-XU3/issues/2

externo6, 2017-05-21:
Is there a way to achieve this? IE force 60hz within Lakka?

Kivutar, 2017-05-22:
It should be fixed on 2.1. Or you can apply the change of gouchi on your own installation.

closed 2017-05-22 by Kivutar