libretro/Lakka-LibreELEC #127: OdroidXU3 , mupen64plus   failt to  build   (parallel-n64 ppsspp to)

Lino77, 2017-05-14:
custom/glsm/glsm.cpp:241:13: warning: 'void on_gl_error(GLenum, GLenum, GLuint, GLenum, GLsizei, const char*, void*)' defined but not used [-Wunused-function]
static void on_gl_error(GLenum source, GLenum type, GLuint id, GLenum severity, GLsizei length, const char* message, void *userParam)
^~~~~~~~~~~
make[1]: *** [Makefile:317: custom/glsm/glsm.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/lino/LibreELEC.tv/lakka-V2.0/build.Lakka-OdroidXU3.arm-8.0-devel/mupen64plus-eda9d16'
Makefile:12: die Regel für Ziel „image“ scheiterte
make: *** [image] Fehler 2
lino@ubuntu:~/LibreELEC.tv/lakka-V2.0$ git pull
Bereits aktuell.
lino@ubuntu:~/LibreELEC.tv/lakka-V2.0$

Kivutar, 2017-05-14:
On what branch are you?

Lino77, 2017-05-15:
Lakka-V2.0

Lino77, 2017-05-15:
Lakka-V2.0

Lino77, 2017-05-15:
Build in ubuntu 16.04 ,17.04 , AMD Linux Mint 18.1

Lino77, 2017-05-16:
okay  with ubuntu x86 it works

closed 2017-05-16 by Lino77