libretro/Lakka-LibreELEC #132: Invalid Mode is selected by default and no other modes are available

furatail, 2017-05-22:
I've got this rather unknown All-in-one PC bought from a School auction. Hatch brand, model numbers turn up absolutely nothing, it's a rare thing.

The problem I am having is with Lakka core itself, the detected modes for the built-in monitor has just one detected mode. I cannot force 1920x1080 because no suitable video mode for 1920 x 1080 is found.
  
`RetroArch [INFO] :: Video @ 1920x1080
RetroArch [INFO] :: [DRM]: Found 3 connectors.
RetroArch [INFO] :: [DRM]: Connector 0 connected: yes
RetroArch [INFO] :: [DRM]: Connector 0 has 1 modes.
RetroArch [INFO] :: [DRM]: Connector 0 assigned to monitor index: #1.`
...
`RetroArch [INFO] :: [DRM]: Mode 0: (1680x1050) 1680 x 1050, 60 Hz`

The mode detected is 1680x1050, 60hz and not supported by my 1920 x 1080 monitor and results in a kind of doubled image displayed and a bit jagged, no built in scaling. In fact, the Lakka installer itself presented the same problems, but it was readable enough to install.

I do not think this is a GPU issue since I can attach an external monitor and be able to run at my external monitors native resolution. If I attached an external Dell Monitor (Native 1600x1200), then all modes are properly detected and I can either set my own resolutions in config or stick with the default.

There needs to be some way to manually edit which modes my display can use. I'll follow with some logs for better insight of my environment:

`RetroArch [INFO] :: This is RetroArch version 1.5.0 (Git 2a13af5)
RetroArch [INFO] :: === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 
Built: Apr 24 2017
RetroArch [INFO] :: Version: 1.5.0
RetroArch [INFO] :: Git: 2a13af5
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Loading default config.
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Redirecting save file to "/storage/savefiles/.srm".
RetroArch [INFO] :: Redirecting savestate to "/storage/savestates/.state".
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29970.03 Hz.
RetroArch [INFO] :: Found shader "/tmp/shaders/bilinear.glslp"
RetroArch [INFO] :: Found shader "/tmp/shaders/nearest.glslp"
RetroArch [INFO] :: Found shader "/tmp/shaders/stock.glsl"
RetroArch [INFO] :: Video @ 1920x1080
RetroArch [INFO] :: [DRM]: Found 3 connectors.
RetroArch [INFO] :: [DRM]: Connector 0 connected: yes
RetroArch [INFO] :: [DRM]: Connector 0 has 1 modes.
RetroArch [INFO] :: [DRM]: Connector 0 assigned to monitor index: #1.
RetroArch [INFO] :: [DRM]: Connector 1 connected: no
RetroArch [INFO] :: [DRM]: Connector 1 has 0 modes.
RetroArch [INFO] :: [DRM]: Connector 2 connected: no
RetroArch [INFO] :: [DRM]: Connector 2 has 0 modes.
RetroArch [INFO] :: [DRM]: Mode 0: (1680x1050) 1680 x 1050, 60 Hz
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
RetroArch [INFO] :: Found GL context: kms
RetroArch [INFO] :: Detecting screen resolution 1680x1050.
RetroArch [ERROR] :: [KMS/EGL]: Did not find suitable video mode for 1920 x 1080.
RetroArch [ERROR] :: Cannot open video driver ... Exiting ...
RetroArch [ERROR] :: Fatal error received in: "init_video()"
RetroArch [ERROR] :: This core requires a content file.`

`00:00.0 Host bridge [0600]: Intel Corporation 4 Series Chipset DRAM Controller [8086:2e20] (rev 03)
00:02.0 VGA compatible controller [0300]: Intel Corporation 4 Series Chipset Integrated Graphics Controller [8086:2e22] (rev 03)
00:02.1 Display controller [0380]: Intel Corporation 4 Series Chipset Integrated Graphics Controller [8086:2e23] (rev 03)
00:1a.0 USB controller [0c03]: Intel Corporation 82801JI (ICH10 Family) USB UHCI Controller [8086:3a37]
00:1a.1 USB controller [0c03]: Intel Corporation 82801JI (ICH10 Family) USB UHCI Controller [8086:3a38]
00:1a.2 USB controller [0c03]: Intel Corporation 82801JI (ICH10 Family) USB UHCI Controller [8086:3a39]
00:1a.7 USB controller [0c03]: Intel Corporation 82801JI (ICH10 Family) USB2 EHCI Controller [8086:3a3c]
00:1b.0 Audio device [0403]: Intel Corporation 82801JI (ICH10 Family) HD Audio Controller [8086:3a3e]
00:1c.0 PCI bridge [0604]: Intel Corporation 82801JI (ICH10 Family) PCI Express Root Port 1 [8086:3a40]
00:1c.1 PCI bridge [0604]: Intel Corporation 82801JI (ICH10 Family) PCI Express Port 2 [8086:3a42]
00:1c.2 PCI bridge [0604]: Intel Corporation 82801JI (ICH10 Family) PCI Express Root Port 3 [8086:3a44]
00:1d.0 USB controller [0c03]: Intel Corporation 82801JI (ICH10 Family) USB UHCI Controller [8086:3a34]
00:1d.1 USB controller [0c03]: Intel Corporation 82801JI (ICH10 Family) USB UHCI Controller [8086:3a35]
00:1d.2 USB controller [0c03]: Intel Corporation 82801JI (ICH10 Family) USB UHCI Controller [8086:3a36]
00:1d.7 USB controller [0c03]: Intel Corporation 82801JI (ICH10 Family) USB2 EHCI Controller [8086:3a3a]
00:1e.0 PCI bridge [0604]: Intel Corporation 82801 PCI Bridge [8086:244e] (rev 90)
00:1f.0 ISA bridge [0601]: Intel Corporation 82801JIB (ICH10) LPC Interface Controller [8086:3a18]
00:1f.2 SATA controller [0106]: Intel Corporation 82801JI (ICH10 Family) SATA AHCI Controller [8086:3a22]
00:1f.3 SMBus [0c05]: Intel Corporation 82801JI (ICH10 Family) SMBus Controller [8086:3a30]
01:00.0 Memory controller [0580]: Intel Corporation Turbo Memory Controller [8086:444e] (rev 11)
02:00.0 Network controller [0280]: Intel Corporation WiFi Link 5100 [8086:4232]
03:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller [10ec:8168] (rev 02)`

gouchi, 2017-05-22:
Which output are using ? Please try to [list the available outputs](http://www.lakka.tv/doc/PC).

Then you can try to add to the [cmdline](http://www.lakka.tv/doc/The-bootloader/#on-generic-pc), the mode you want. For example, this one is using dvi output with resolution 1024x768 at 60 Hz

`video=DVI-I-1:1024x768@60 `

furatail, 2017-05-23:
The available outputs are:
`HDMI-A-1: disconnected`
`LVDS-1: connected`
`VGA-1: disconnected`

LVDS-1: is the built in monitor. 

I tried to add the "video=..." option per the instructions,
`Lakka:~ # mount -o remount,rw /flash`
`Lakka:~ # nano /flash/syslinux.cfg`
 but the file doesn't exit. I just end up creating a new file. I added the entry:
`video=LVDS-1:1920x1080@60` and saved the results, rebooted but saw no difference. 1680x1050 was still selected. I tried using live mode, but pushing tab during boot only gives me the option to live or install.

Kivutar, 2017-05-23:
Creating a new file will not help. Try to find the file with `ls`. I think the one you're looking for is in `/flash/EFI/`

gouchi, 2017-05-23:
For the live mode, please boot with
`live video=LVDS-1:1920x1080@60`

Thank you.

furatail, 2017-05-24:
I have tried the above comments: 
At boot I pressed TAB and ran the command, `live video=LVDS-1:1920x1080@60` this did not change the display output. 1680x1050, 60hz was still chosen.
Also, I figured out that I must actually insert the flash drive before mounting. After doing this I was able to edit both /flash/syslinux.cfg and /flash/EFI/BOOT/syslinux.cfg. I then booted from that flash drive into live mode. This also did not change output to supported resolution. I had the same results booting into installer as well.

The contents of /flash/syslinux.cfg is the following:
```
SAY Wait for live session to start or press <TAB> for more options (installer)
DEFAULT installer
TIMEOUT 50
PROMPT 1

LABEL installer
  KERNEL /KERNEL
  APPEND boot=UUID=2404-4447 installer quiet tty vga=current vt.global_cursor_default=0 loglevel=2
video=LVDS-1:1920x1080@60

LABEL live
  KERNEL /KERNEL
  APPEND boot=UUID=2404-4447 disk=UUID=ac471f6b-554f-4380-8bb5-4d4f3821014a portable quiet
vt.global_cursor_default=0 loglevel=2 video=LVDS-1:1920x1080@60

```
The contents of /flash/EFI/BOOT/syslinux.cfg is as followed:
```
SAY Wait for live session to start or press <TAB> for more options (installer)
DEFAULT installer
TIMEOUT 50
PROMPT 1

LABEL installer
  KERNEL /KERNEL
  APPEND boot=UUID=2404-4447 installer quiet tty vga=current vt.global_cursor_default=0 loglevel=2 video=LVDS-1:1920x1080@60

LABEL live
  KERNEL /KERNEL
$ loglevel=2 video=LVDS-1:1920x1080@60
```

The results of anything I have tried still results in the following output:
[Display Output Photograph](http://imgur.com/a/w96Mm)

Perhaps EDID information isn't properly read from the built in display, yet can be read from the DVI port?

gouchi, 2017-05-24:
You may try [this documentation]([url](http://wiki.openelec.tv/index.php/Configuring_a_Custom_xorg.conf#tab=Intel)) about custom EDID.

furatail, 2017-05-29:
I appreciate the help, I do understand what must be done but after several days I have had little luck. Turns out the EDID information read is a bit garbage. I have tried creating custom EDIDs with other linux distributions using https://github.com/akatrevorjay/edid-generator but I'm having trouble since nothing is able to read this "H@tch" brand display EDID information. No distro can display properly on this AIO machine.  I do believe the solution to this problem is to create a custom EDID which I have so far had no luck with but I am continuing to seek a solution. I may see what timing information I can gather using Windows image that did work. Once I figure out those timings I do believe once I get a working EDID and I can use the above documentation to set it. This is looking more to be a very specific hardware issue rather than Lakka-LibreElec related and there does seem to be enough guidance shared to get me going just as soon as I can build a proper EDID binary.

furatail, 2017-06-03:
Okay. Finally got my display to work properly. I followed the above mentioned documentation but rather than trying to build a proper EDID like I intended, I instead attached a VGA monitor to the AIO machine, grabbed its EDID information from card0-VGA-1 instead and packaged that into edid.cpio to be used with the internal LVDS-1 display. All is working now. Thanks for the help. I would not have figured this out on my own.

closed 2017-06-03 by furatail

gouchi, 2017-06-04:
Glad it finally works !