libretro/Lakka-LibreELEC #133: OdroidXU3 fails to build

externo6, 2017-05-24:
Building OdroidXU3.
DISTRO=Lakka PROJECT=OdroidXU3 ARCH=arm make image
Branch Lakka-V2.0
OS = Ubuntu 14.04, 16.04, CentOS 7
Output of termination:
`/usr/include/features.h:374:25: fatal error: sys/cdefs.h: No such file or directory
 #  include <sys/cdefs.h>
                         ^
compilation terminated.
In file included from host/buildvm_asm.c:6:0:
host/buildvm.h:9:23: fatal error: sys/types.h: No such file or directory
 #include <sys/types.h>
                       ^
compilation terminated.
In file included from host/buildvm_peobj.c:9:0:
host/buildvm.h:9:23: fatal error: sys/types.h: No such file or directory
 #include <sys/types.h>
                       ^
compilation terminated.
In file included from host/buildvm_lib.c:6:0:
host/buildvm.h:9:23: fatal error: sys/types.h: No such file or directory
 #include <sys/types.h>
                       ^
compilation terminated.
make[2]: *** [Makefile:661: host/minilua.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [Makefile:661: host/buildvm_asm.o] Error 1
make[2]: *** [Makefile:661: host/buildvm_peobj.o] Error 1
make[2]: *** [Makefile:661: host/buildvm_lib.o] Error 1
make[2]: Leaving directory '/root/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm-8.0-devel/lutro-687a19c/deps/luajit/src'
make[1]: *** [Makefile:260: deps/luajit/src/libluajit.a] Error 2
make[1]: Leaving directory '/root/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm-8.0-devel/lutro-687a19c'
make: *** [image] Error 2`

Hope im not doing something stupid. First time I've built the LE builds, didnt have an issue building OE builds.

externo6, 2017-05-24:
Tried building RPi also got the same:

```/usr/include/features.h:374:25: fatal error: sys/cdefs.h: No such file or directory
 #  include <sys/cdefs.h>
                         ^
compilation terminated.
HOSTCC    host/buildvm_asm.o
/root/Lakka-LibreELEC/build.Lakka-RPi.arm-8.0-devel/toolchain/bin/host-gcc -m32  -O2 -fomit-frame-pointer -Wall  -O3 -fPIC -I. -DLUAJIT_TARGET=LUAJIT_ARCH_arm -DLJ_ARCH_HASFPU=1 -DLJ_ABI_SOFTFP=0  -c -o host/buildvm_asm.o host/buildvm_asm.c
HOSTCC    host/buildvm_peobj.o
/root/Lakka-LibreELEC/build.Lakka-RPi.arm-8.0-devel/toolchain/bin/host-gcc -m32  -O2 -fomit-frame-pointer -Wall  -O3 -fPIC -I. -DLUAJIT_TARGET=LUAJIT_ARCH_arm -DLJ_ARCH_HASFPU=1 -DLJ_ABI_SOFTFP=0  -c -o host/buildvm_peobj.o host/buildvm_peobj.c
In file included from host/buildvm_asm.c:6:0:
host/buildvm.h:9:23: fatal error: sys/types.h: No such file or directory
 #include <sys/types.h>
                       ^
compilation terminated.
make[2]: *** [Makefile:662: host/minilua.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from host/buildvm_peobj.c:9:0:
host/buildvm.h:9:23: fatal error: sys/types.h: No such file or directory
 #include <sys/types.h>
                       ^
compilation terminated.
make[2]: *** [Makefile:662: host/buildvm_asm.o] Error 1
make[2]: *** [Makefile:662: host/buildvm_peobj.o] Error 1
make[2]: Leaving directory '/root/Lakka-LibreELEC/build.Lakka-RPi.arm-8.0-devel/lutro-687a19c/deps/luajit/src'
make[1]: *** [Makefile:260: deps/luajit/src/libluajit.a] Error 2
make[1]: Leaving directory '/root/Lakka-LibreELEC/build.Lakka-RPi.arm-8.0-devel/lutro-687a19c'
make: *** [image] Error 2
```

Kivutar, 2017-05-24:
I can't reproduce this issue on my Ubuntu 16.04.2 LTS

Kivutar, 2017-05-24:
Try to install gcc-multilib and libc6-dev-i386

externo6, 2017-05-24:
I searched one of the errors can came across this : https://github.com/libretro/Lakka/issues/303
Installed: apt-get install gcc-multilib seems to be progressing.

externo6, 2017-05-24:
Closing, gcc-multilib fixed it.

closed 2017-05-24 by externo6