libretro/Lakka-LibreELEC #134: Lakka won't launch Installer or Live

N00byKing, 2017-05-28:
I used Lakka before Version 2.0, without Problems.
However, when trying to install 2.0, neither the Installer or the Live System start, and I am left with a blinking Cursor, doing nothing (at least visibly).

I already tried textmode, ssh, and nomodeset, none make any difference, and connecting using SSH doesn't work (Checking my Router the PC wasn't even Connecting to the Local Network)

When choosing Live, the Display goes blank for a Secound, then back to the Blinking, when choosing Installer it doesn't go Blank and blinks forever, without Change.

I also tried using the 32Bit Image, where I, no matter what  I do, get a Kernel Panic.

- Which version of Lakka are you using? Latest from Lakka.tv, x64
- What system hardware are you using? MSI Nvidia GT9800, AMD Athlon 64 X2 Dual Core 3800+
- What did you do? Several Attempts at Installing or just trying to get into the Live System
- What did you expect to happen? Getting to the Installation Screen or the Live System
- What happened instead? Infinite Blinking Cursor, the PC never exits the Bootloader

gouchi, 2017-05-28:
You should be able to boot into terminal with

`live textmode ssh`

It it possible you try to burn your image with [etcher](https://etcher.io) ?

Thank you.

N00byKing, 2017-05-28:
Booting using "live textmode ssh" now crashes the System after switching to Etcher, after rebooting I get a Message telling me a "Hyper Transport Flood Error has Ocurred" (to be honest I think this has more to do with the Age of my System)

Generally Booting live doesn't work.

BUT, after switching to Etcher the Installer boots and Installs Successfully, and Lakka worked flawlessly after Rebooting

closed 2017-05-28 by N00byKing