libretro/Lakka-LibreELEC #135: 3840x240 hdmi timings on RPi cause segmentation fault

kevinfishburne, 2017-05-29:
Lakka 2.0, RPi 3 Model B.
hdmi_timings=3840 1 109 343 343 240 1 5 3 14 0 0 0 60 0 72925291 1

Using these timings Retroarch segfaults, showing only the Lakka logo. I can still ssh in to it to see what's happening, so it seems like LibreELEC is still working fine.

```
kevinfishburne@codex ~ $ ssh root@192.168.0.56
root@192.168.0.56's password: 
###########################################
# Lakka - The DIY retro emulation console #
# ...... visit http://www.lakka.tv/ ...... #
###########################################

Lakka (community) Version: 2.0
Lakka:~ # systemctl stop retroarch.service ; retroarch -v
RetroArch [INFO] :: This is RetroArch version 1.5.0 (Git 2a13af5)
RetroArch [INFO] :: === Build =======================================
Capabilities: NEON VFPv3 VFPv4 
Built: Apr 24 2017
RetroArch [INFO] :: Version: 1.5.0
RetroArch [INFO] :: Git: 2a13af5
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Loading default config.
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Redirecting save file to "/storage/savefiles/.srm".
RetroArch [INFO] :: Redirecting savestate to "/storage/savestates/.state".
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29970.03 Hz.
RetroArch [INFO] :: Found shader "/tmp/shaders/bilinear.glslp"
RetroArch [INFO] :: Found shader "/tmp/shaders/nearest.glslp"
RetroArch [INFO] :: Found shader "/tmp/shaders/stock.glsl"
RetroArch [INFO] :: Video @ fullscreen
RetroArch [INFO] :: [EGL]: EGL version: 1.4
RetroArch [INFO] :: Found GL context: null
RetroArch [INFO] :: Detecting screen resolution 320x240.
RetroArch [INFO] :: [GL]: Vendor: (null), Renderer: (null).
RetroArch [INFO] :: [GL]: Version: (null).
Segmentation fault
Lakka:~ #
```
a 3840x240 mode is the holy grail of integer-scaled retro emulation, as the vast majority of home console and arcade boards' horizontal resolutions divide evenly into 3840. This crash has also been confirmed by Mike Chi.

Kivutar, 2017-05-29:
I'm able to reproduce the issue. When you say that it doesn't happen on LibreELEC, you're talking about KODI or RetroArch on LibreELEC?

kevinfishburne, 2017-05-30:
I'm using Lakka 2.0, which I think uses LibreELEC instead of OpenELEC. So while RetroArch crashes, the underlying OS is still functioning allowing me to ssh in and poke around. I haven't tried to reproduce the crash using other distros, just Lakka. I also filed a bug report with libretro/RetroArch.

Kivutar, 2017-05-30:
But if I use a 1920 hdmi_timings, it works. It's certainly a size limit of the framebuffer or something in the display driver, but the backtrace it not very helpful.

Kivutar, 2017-05-31:
For now I use this line

    hdmi_timings=1920 1 152 247 280 240 1 3 7 12 0 0 0 60 0 40860000 1

mikechi2, 2017-06-06:
@Kivutar Thanks -- that solution works well, but it would be great if it were possible to get higher resolutions since we often need weird pixel clocks for various systems. For example, I need 2730 wide buffer to generate the correct timing for a NTSC chroma carrier but it appears that widths above 2048 cause the crash.

kevinfishburne, 2017-06-06:
FYI I also filed this with libretro/RetroArch, though not much activity so far (not even assigned or confirmed as a bug): https://github.com/libretro/RetroArch/issues/4978

ronfar623, 2017-07-04:
Apologies if this is the wrong forum, but until this is resolved, would you know how to generate an hdmi_timings string for 1280x240?  It's a common multiple of 256 and 320 that's below the 2048 limit, and would at least cover many of the 80s and 90s era home consoles.

Just received my RetroTink adapter, and have been frustrated trying to find a good all-around video mode that's the correct aspect ratio without the incorrect scaling shimmering.  Thank you!

mikechi2, 2017-07-04:
Hey did you check out the latest Lakka image? It runs @ 1920 x 240 and as
far as I can tell, the non-integer scaling artifacts aren't visible.

ronfar623, 2017-07-05:
@mikechi2 Hey!  Thanks for your work on RetroTink.  This has been a hell of a ride starting with the RPi composite 240p firmware earlier this year.  Now here I am reliving childhood memories via a flawless component video connection to one of the 5 CRTs (and one PVM) I've obtained this year as a result of all of this.  Needless to say, I don't think the wife appreciates you guy's work as much as I do. ;)

You're right; during normal use, non-integer scaling artifacts at 1920 are so subtle they may as well not exist, but shimmering is still apparent when running the scrolling test in the 240p Test Suite vs true integer scaling.  Really not a big deal, but the obsessive purist part of my brain always wants better.

I think I actually prefer the 1600 width timings string in your config.txt, at least for 80s-90s era consoles.  I'm sure it's due to some subtlety of the timings that I don't understand, but integer scaling fits the boundaries of my TV almost perfectly by default for both Genesis and SNES.

starquake, 2017-07-06:
A 1280x240 one sounds good to me too! Would love to hear if someone could work one out.

kevinfishburne, 2017-07-08:
@ronfar623 You can use the hdmi timing calculator to create your own timings: http://eightvirtues.com/lakka/hat/ It's an OpenOffice/LibreOffice file (ods).

closed 2020-08-18 by Ntemis