libretro/Lakka-LibreELEC #136: Black Icons Lakka 2.0 with Wetek Core

batserra, 2017-05-29:
Hi i have download latest (fresh install) versión 2.0 for my wetek core but after install all icons are black, how i can fix it? thanks for all

[image]

Kivutar, 2017-05-29:
Can you enable the advanced settings and check the assets directory in the directory settings?

Usually, a

    systemctl stop retroarch
    rm -rf .cache .config
    systemctl start retroarch

will fix it

kingknuts, 2017-05-29:
That happened to me on the lattepanda as well but I found out that was the storage partition not being mounted...
(I changed the partitiom format to fat32 trying to access it from a Windows OS and wasn't working)

batserra, 2017-05-29:
@Kivutar thanks but not working... and @kingknuts hi ,did you fix it? how?

Thanks for all....

batserra, 2017-05-31:
Hi ,i tryed Lakka 2.1 Rc1 ,but also the same
I have copied to /tmp/asset files from image ,though WinScp ,and works ok ,also y copied cores files.
But when i try to run a rom, does not work ,just nothing happends...

Its a bug? Really i dont know how to solve it ,any ideas?

Thanks

Kivutar, 2017-05-31:
Maybe there is a bug in hte overlayFS? What is the output of the `mount` command?

batserra, 2017-05-31:
@Kivutar ,hi ,thanks but i don't know how to do it sorry ,i have to enter with putty ,and after what? ,just write mount? ,is there a log for it?

Kivutar, 2017-05-31:
Yes you use putty, you type mount, and it will output some text, then you paste it there.

batserra, 2017-05-31:
hi @Kivutar  heres is mount....

Lakka:~ # mount
rootfs on / type rootfs (rw)
devtmpfs on /dev type devtmpfs (rw,relatime,size=761908k,nr_inodes=66075,mode=755)
proc on /proc type proc (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
/dev/mmcblk0p1 on /flash type vfat (ro,noatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p2 on /storage type ext4 (rw,noatime,journal_checksum,journal_async_commit,noauto_da_alloc,data=ordered)
/dev/loop0 on / type squashfs (ro,relatime)
securityfs on /sys/kernel/security type securityfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620)
tmpfs on /run type tmpfs (rw,nosuid,nodev,mode=755)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,release_agent=/usr/lib/systemd/systemd-cgroups-agent,name=systemd)
tmpfs on /var type tmpfs (rw,relatime)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev)
mqueue on /dev/mqueue type mqueue (rw,relatime)

Kivutar, 2017-05-31:
The overlayFS mounts are missing. I'm going to take a closer look at this.

batserra, 2017-05-31:
thanks for all @Kivutar i will wait for it,

kingknuts, 2017-05-31:
Sorry for being late...
My fix was that the partition wasn't mounted because in the eMMC of my lattepanda I've already had a storage partition. So to fix that I just changed the grub parameters to boot using the UUID Of the partitions instead of using the name of them. To get the UUID of the partitions you need to type blkid

Kivutar, 2017-05-31:
I will give you a fixed image in about 3h.

Kivutar, 2017-05-31:
http://static.kivutar.me/Lakka-WeTek_Core.arm-2.1-rc2.img.gz

batserra, 2017-05-31:
Thanks @Kivutar it's working !!! ,i have donwloaded 2.1-rc2 ,done a fresh install ,and now no black thubnails, cores updated and roms working ,

thanks again for all your work

Kivutar, 2017-05-31:
No problem, thanks for the bug report!

closed 2017-05-31 by Kivutar