libretro/Lakka-LibreELEC #139: Wetek Core and DualShock 4 ,usb and Bluetooh not working ,in menu or games

batserra, 2017-06-03:
- What system hardware are you using? 2.1 Rc2 Wetek Core
- What did you do? 
Try to connect DualShock Usb or Bluetooh
- What did you expect to happen?
Connect to play ,but no icon move  or game not working

Hi ,y tryed to connect my DualSock 4 following this page:

http://www.lakka.tv/doc/Wireless-Dualshock/

also y tryed with:

https://wiki.gentoo.org/wiki/Sony_DualShock#DualShock_4

But no one worked ,i thinks that is paired ,but it's not working...

what i have done?

Lakka (community) Version: 2.1-rc2
Lakka:~ # bluetoothctl
[NEW] Controller 43:30:B1:00:00:00 BlueZ 5.43 [default]
[NEW] Device 30:0E:D5:8F:72:A0 Wireless Controller

[bluetooth]# agent on
Agent registered
[bluetooth]# default-agent
Default agent request successful
[bluetooth]# power on
Changing power on succeeded
[bluetooth]# discoverable on
Changing discoverable on succeeded
[CHG] Controller 43:30:B1:00:00:00 Discoverable: yes
[bluetooth]# pairable on
Changing pairable on succeeded
[bluetooth]# scan on
Discovery started
[CHG] Controller 43:30:B1:00:00:00 Discovering: yes
[NEW] Device 74:E5:43:55:9C:B4 BATSPC
[CHG] Device 30:0E:D5:8F:72:A0 RSSI: -72
[bluetooth]# devices
Device 30:0E:D5:8F:72:A0 Wireless Controller
Device 74:E5:43:55:9C:B4 BATSPC
[bluetooth]# pair 30:0E:D5:8F:72:A0
Attempting to pair with 30:0E:D5:8F:72:A0
Failed to pair: org.bluez.Error.AlreadyExists
[bluetooth]# trust 30:0E:D5:8F:72:A0
Changing 30:0E:D5:8F:72:A0 trust succeeded
[bluetooth]# quit
Agent unregistered
[DEL] Controller 43:30:B1:00:00:00 BlueZ 5.43 [default]
Lakka:~ #

In information->System information ,there is
Port #1 device name: Wireless Controller (#1)

------------------

In in menu i can not move with DualShock and also in games it's not working...

So ,does anybody can help me? ,if there is a bug or something that i am doing wrong?

Thanks for all

Ntemis, 2017-06-03:
I don't know if Wetek Core kernel supports DualShock 4, i have to ask the R&D department.

batserra, 2017-06-04:
So its a bug ,thanks @Kivutar i will wait for a fix

ToKe79, 2018-01-31:
Hello, can you test with latest [release](http://www.lakka.tv/get) and/or [nightly](http://nightly.builds.lakka.tv)?

Ntemis, 2018-02-14:
no answer closing this down

closed 2018-02-14 by Ntemis